lvgl/lvgl #6832: LV_INDEV_MODE_EVENT is not compatible with scroll momentum

niklasf, 2024-09-11:
### LVGL version

v9.2.0

### What happened?

Try to throw a scrollable list with the input device created by `lv_sdl_mouse_create()`. Scrolling stops immediately, there is no scroll momentum.

This was working in LVGL 8. Bisected to 2b0587197c038b960b69920930c43877fb6b91c8 as the first bad commit.

If I understand correctly, this happens due to the decay in:

https://github.com/lvgl/lvgl/blob/559618d9d2c66069f61ba673b7bce6bf3e7eb036/src/indev/lv_indev.c#L1254-L1255

This assumes that `indev_proc_press()` is called at a roughly fixed interval. But if an input driver with `LV_INDEV_MODE_EVENT` delivers events much more quickly (like SDL does), the scroll vector decays to 0 almost immediately.

### How to reproduce?

Using the same example again, with the SDL input driver.

```c
static void demo()
{
    lv_obj_t * screen = lv_obj_create(NULL);
    lv_obj_set_size(screen, 512, 600);
    lv_obj_set_flex_flow(screen, LV_FLEX_FLOW_COLUMN);

    for (int i = 0; i < 100; i++)
    {
        lv_obj_t * btn = lv_button_create(screen);
        lv_obj_set_size(btn, lv_pct(100), LV_SIZE_CONTENT);
    }

    lv_disp_load_scr(screen);
}

int main()
{
    lv_init();

    lv_sdl_window_create(512, 600);
    lv_sdl_mouse_create();

    demo();

    while(true) {
        uint32_t delay = lv_timer_handler();
        usleep(delay * 1000);
    }

    return 0;
}
```

kisvegabor, 2024-09-12:
I can confirm it. You identified the root problem correctly. However I wonder what should be a good solution. Ideally the vectors should be speed vectors (pixels/ms) instead of simple distances. 

Now SDL events are processed in every 5ms. However what if SDL adds 10 mouse position events in 5 ms?
1. We will see only a very little movement? (usually 0, therefore speed will be zero too) 
2. There is no timestamp. We just see that in the exact same millisecond we read 5 data. That is the time diff is zero, meaning speed is infinite.

Do you have any ideas how to a mange it?

niklasf, 2024-09-14:
Perhaps we could keep a small buffer of vectors with timestamps? The issue exists also for input devices with timer that use `continue_reading`, so if it's not too expensive, this could be useful in general.

kisvegabor, 2024-09-17:
Okay, let's say we save the last 8 data like this:
```
0. x=0; y=0; t=1000
1. x=0; y=0; t=1000
2. x=0; y=0; t=1000
3. x=0; y=0; t=1000
4. x=0; y=5; t=1010
5. x=0; y=10; t=1010
6. x=0; y=15; t=1010
7. x=0; y=20; t=1010
```
What can we conclude from it?
``` 
time_diff = t[7]-t[0]
x_diff = max(x) - min(x)  
y_diff = max(y) - min(y)

x_speed = x_diff / time_diff
y_speed = y_diff / time_diff
```

What do you think? To be honest I have no idea how it will behave in the practice. Probably we should fill the the array with the first pressed coordinate to have valid data. 

Besides the array size can be configurable for each indev.

kisvegabor, 2024-10-02:
Not stale, let's keep it open a little longer to continue the discussion.

niklasf, 2024-10-19:
No updates, but I am still interested in having this fixed and will get back to it eventually. I am also happy if someone else beats me to it.

niklasf, 2024-11-06:
Not stale

kisvegabor, 2024-11-27:
Not stale

niklasf, 2024-12-15:
Not stale. Still no great solution

kisvegabor, 2024-12-18:
The idea [here](https://github.com/lvgl/lvgl/issues/6832#issuecomment-2355577604) seems like a good way to start, if someone has time to try it out.

kisvegabor, 2025-01-09:
I marked it as important as we should fix it sooner or later.

niklasf, 2025-02-18:
Not stale. And finally here's an attempted fix: #7794.

PR #7794 (2025-06-25): fix(indev): decouple scroll momentum decay from indev read loop (#6832)
Fixes #6832.

As suggested in the issue, this computes the scroll throw vector based on a short history.

Initially I computed the speed over the entire history up to a maximum time, but then it depends a lot on individual history entries making the cut or not and can feel a bit inconsistent.

Next, I applied exponential decay to history entries, `x * exp(-t / 33.0)`, which felt much smoother.

To avoid floating point arithmetic, I tried some series expansions, but ultimately simple linear interpolation seems good enough.
 [... cut]
closed 2025-06-25 by kisvegabor