lvgl/lvgl #6833: The placeholder text in textarea control cannot be centered！

anheiqq, 2024-09-12:
### LVGL version

9.1.0

### What happened?

The placeholder text in textarea control cannot be centered！

The cause of the problem has been found：
[image]

lv_area_move：
void lv_area_move(lv_area_t * area, int32_t x_ofs, int32_t y_ofs)
{
    area->x1 += x_ofs;
    area->x2 += x_ofs;
    area->y1 += y_ofs;
    area->y2 += y_ofs;
}

It should be replaced with:

void lv_area_decrease(lv_area_t * area, int32_t x_ofs, int32_t y_ofs)
{
    area->x1 += x_ofs;
    area->x2 += x_ofs;
    area->y1 -= y_ofs;
    area->y2 -= y_ofs;
}

liamHowatt, 2024-09-18:
It looks like centering the placeholder is not currently intended to be possible unless there are some significant changes.

What are you doing to try to center the placeholder? Can you send an example where your suggested fix causes the placeholder to be centered successfully?

kisvegabor, 2024-09-18:
Send a Pull request would be useful too the better see the suggested changes. :pray:

liamHowatt, 2024-09-23:
I understand now. Thanks!

closed 2024-09-26 by anheiqq