lvgl/lvgl #6837: Parent `height` set to `LV_SIZE_CONTENT` and child `y` position set using `LV_PCT` macro

mvladic, 2024-09-12:
### LVGL version

v9.1 v8.4

### What happened?

When there is a parent widget which `height` is set to `LV_SIZE_CONTENT` and a child widget which `y` position is set using `LV_PCT` macro then actual `y` position in px of the child will not be calculated correctly - it looks to me that height of the parent used in this calculation is not actual height calculated from the content but something else. And for some values given to the LV_PCT macro the LVGL code will freeze, looping forever.

### How to reproduce?

This code will make LVGL to freeze:

```
    // parent obj
    lv_obj_t *parent_obj = lv_img_create(lv_scr_act());
    lv_obj_set_pos(parent_obj, 0, 0);
    lv_obj_set_size(parent_obj, LV_SIZE_CONTENT, LV_SIZE_CONTENT);               // parent height set to LV_SIZE_CONTENT
    lv_img_set_src(parent_obj, &img_img_astra_top_view);
    lv_image_set_inner_align(parent_obj, LV_IMAGE_ALIGN_DEFAULT);

        // child object
        lv_obj_t *child_obj = lv_img_create(parent_obj );
        lv_obj_set_pos(child_obj , 0, LV_PCT(100));                    // child y position set to LV_PCT(100)
        lv_obj_set_size(child_obj , LV_SIZE_CONTENT, LV_SIZE_CONTENT);
        lv_img_set_src(child_obj , &img_img_brake_lights_on);
        lv_image_set_inner_align(child_obj , LV_IMAGE_ALIGN_DEFAULT);
```

Both images, `img_img_astra_top_view` and `img_img_brake_lights_on`, are of size 186x172.

kisvegabor, 2024-09-12:
Hi,

That's expected, although not elegant to freeze. :sweat_smile:

This is what happens:
1. When the parent has content size it will be as large as the children or it's content (image source in the case)
2. If you place the child to `y=100%` it means make the child start at the very bottom of the  parent
3. But then the parent shall be resized
4. But the then the child needs to be moved again to the bottom
5. But then the parent shall be resized ...
6. And infinite loop  

Ideally it shouldn't be possible to select such a combination.

mvladic, 2024-09-12:
Yes, it makes sense. As far as I'm concerned, this issue can be closed, unless you still want to prevent freezing in this case.

kisvegabor, 2024-09-30:
@kissa96 will join our team soon. I keep for him as a getting started issue. :slightly_smiling_face:

liamHowatt, 2024-11-07:
One day this may be done: #5938

PR #7041 (2024-11-12): fix(obj): fix crash with LV_SIZE_CONTENT parent and % positioned child
[Related issue](https://github.com/lvgl/lvgl/issues/6837)
Fixes #6837

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.co [... cut]
closed 2024-11-12 by FASTSHIFT