lvgl/lvgl #6842: How can I vertically center the content displayed in a dropdown control after adjusting its height?

anheiqq, 2024-09-13:
### LVGL version

v9.1.0

### What happened?

I noticed that after adjusting the height of the dropdown, the displayed content can only be shown at the top and I couldn't find a setting to vertically center it.

anheiqq, 2024-09-13:
Currently, I can only change the displayed height by setting the pad_top property, but I'm not sure if there's a better way to vertically center the content.

kisvegabor, 2024-09-19:
Hi,

There was no simple way to center the text so I've opened https://github.com/lvgl/lvgl/pull/6881 to automatically center the content. I see no use case to align the text to the top or bottom, so it's always center aligned.

anheiqq, 2024-09-19:
[image]
I solved it this way, but a conflict occurred during the pull request. I will verify  https://github.com/lvgl/lvgl/pull/6881 when I have time.

anheiqq, 2024-09-19:
I have verified https://github.com/lvgl/lvgl/pull/6881 ， It works well.

closed 2024-09-20 by anheiqq

h-milz, 2024-12-10:
Has this been addressed so far? in my recent clone of 9.1, I still see

```
        if(symbol_type == LV_IMAGE_SRC_SYMBOL) {
            symbol_area.y1 = obj->coords.y1 + top;
            symbol_area.y2 = symbol_area.y1 + symbol_h - 1;
            symbol_dsc.text = dropdown->symbol;
            lv_draw_label(layer, &symbol_dsc, &symbol_area);
        }

...

    lv_area_t txt_area;
    txt_area.y1 = obj->coords.y1 + top;
    txt_area.y2 = txt_area.y1 + size.y;
    /*Center align the text if no symbol*/
    if(dropdown->symbol == NULL) {
        txt_area.x1 = obj->coords.x1 + (lv_obj_get_width(obj) - size.x) / 2;
        txt_area.x2 = txt_area.x1 + size.x;
    }
```
and the button text is not centered vertically but top-mid.

h-milz, 2024-12-10:
I pulled the lvgl-micropython repo today which pulls lvgl:master, and it appears this problem has not been resolved. Do we see a regression here?

liamHowatt, 2024-12-11:
There's a CI test to ensure it works. Could you please send a code snippet to reproduce the bad behavior you're seeing? It can be Micropython code.

h-milz, 2024-12-11:
Oh, it's completely default. Only width and height are set to non-default values. 

            dd = lv.dropdown(parent)
            dd.set_options(dd_list)    # dd_list contains the options
            dd.set_symbol(None)      # None or not makes no difference. 
            dd.set_style_text_font(lv.font_montserrat_16, lv.PART.MAIN)
            dd.set_width(80)
            dd.set_height(60)
            # plus a couple of other things like color and borders, which are unrelated to the code snippet above. 
            
and it looks like on the attached image. I'd expect the label to be centered in x and y direction. The point is that in my recent checkout, the patch above is just not contained, and I cannot see how the label would be centered otherwise. 

[image]

liamHowatt, 2024-12-12:
I failed to come up with a way to center it too :slightly_frowning_face: hopefully v9.2.x Micropython can be released soon.

h-milz, 2024-12-13:
@liamHowatt OK, it appears to work correctly with 9.2.2.