lvgl/lvgl #6846: It gets suck in draw_buf_flush function just like #6391

githubljt, 2024-09-13:
### Introduce the problem

porting LVGL 9.1 to stm32u5 with FreeRTOS (poring by myself , not using the lv_freertos.c in lvgl). 
tried LV_OS_NONE and LV_OS_CUSTOM , neither is successfully.
Currently, I'm encountering a problem with draw_buf_flush. During debugging, I found that the issue occurs within the lv_task_handler function. It is stuck in the following code segment:

while (layer->draw_task_head) {
    lv_draw_dispatch_wait_for_request();
    lv_draw_dispatch();
}

githubljt, 2024-09-13:
not use LV_USE_DRAW_SW ,because I use dma2d in stm32u5 (dma2d's driver is implemented by myself), so I think there is no need to use software rendering

Viatorus, 2024-09-16:
So here are further information.

Stack Trace:
* lv_draw_buf_flush, infinity loop
* refr_area_part
* refr_area
* refr_invalid_areas
* lv_display_refr_timer
* lv_timer_exec
* lv_timer_handler

layer content:

[image]

draw_task_head:

[image]

Inside `lv_draw_dispatch_wait_for_request`:

```c
 // I don't understand what this function does or should do if there is an OS or no OS...
lv_draw_dispatch_wait_for_request() {
   while(!_draw_info.dispatch_req);
   _draw_info.dispatch_req = 0;
}
```

Inside `lv_draw_dispatch`:
* task_dispatched stays always false; 

```c
        if(!task_dispatched) {
            lv_draw_wait_for_finish();  // -> empty function, no OS
            lv_draw_dispatch_request();
        }
```

```c
void lv_draw_dispatch_request(void) {
    _draw_info.dispatch_req = 1;
}
```

Viatorus, 2024-09-17:
So here is my conclusion:

* I use a style with the `lv_style_set_opa_layered`, which requires the drawing via an separate drawing layer
* The maximal buffer size of such a drawing layer can be adjusted with LV_DRAW_LAYER_SIMPLE_BUF_SIZE, default is 24 kB
* In some cases, my UI requires multiple drawing layers so the actually free memory is getting less
* The mem_info in case of an error is not up to date anyomre
* Before drawing the free_biggest_size is 52 kB, 32 % memory used
* During drawing the free_biggest_size is 20 kB, 75 % memory used

[image]

However, there is still a bug/missing error handling in LVGL that causes the while loop to never exit if there is no more free memory. The code could determine that there is no progress and this could lead to an LV_ASSERT.

kisvegabor, 2024-09-17:
@Viatorus Can you resolve the issue by increasing LV_MEM_SIZE?

> However, there is still a bug/missing error handling in LVGL that causes the while loop to never exit if there is no more free memory. The code could determine that there is no progress and this could lead to an LV_ASSERT.

Yeah, it should be detected somehow for sure, however I think we can also limit the number of simple layers to avoid out of memory issues like this. But there is problem. Imagine that we limit number of simple layer to 2 and we these widgets
- main_container (opa_layered 50%)
   - big_button (opa layered 50%)
      - image
      - small_container (opa layered 50%)
         - label
               
This will happen:
1. LVGL allocates  the first simple layer for the `main_container`, however to fully render that it needs `big_button` too. 
2. However `big_button` also needs a layer, so the second will be allocated.
3. Similarly to finish `big_button` `small_container` is needed for which an other layer is required. But this would be the third layer.

If you are unlucky it can happen that all layers are allocated for `main_container` and there won't be more layers (or more memory) to start `big_button`.

I'm puzzled with this problem for quite long time, and I can't see a good solution yet. So ideas are very welcome :slightly_smiling_face:

Viatorus, 2024-09-27:
Yes, increasing LV_MEM_SIZE does also help, just like reducing LV_DRAW_LAYER_SIMPLE_BUF_SIZE to something like 8 kB.

Thank you for your deeper explanation why this can not be easily avoided.

kisvegabor, 2024-10-02:
Thank you for the feedback. In lack of a good idea I close this issue now. :slightly_frowning_face:

closed 2024-10-02 by kisvegabor