lvgl/lvgl #6849: Behavior about col_act/row_act of pointer/button devices in lv_table.c

YangGangUEFI, 2024-09-14:
### LVGL version

v9.2.0

### What happened?

Hi, with the following descriptions, i have the question: the reason of assign` LV_TABLE_CELL_NONE` to `col_act/row_act` ?

1. Test with `lv_demo_keypad_encoder()` table
2. Follow https://github.com/lvgl/lvgl/issues/5954#issuecomment-2032990896 this to get the outlines when the widgets were clicked
3. After 2. step, table obj get the outlines but click cell only get flicker (Video 1 show that)
4. If remove following code in lv_table.c, when click cell in table, it can show the cell with bg_color (Video 2)
    https://github.com/lvgl/lvgl/blob/7f57f37560c29ae6007b828fb93a61764f483e1f/src/widgets/table/lv_table.c#L599-L603

https://github.com/user-attachments/assets/d5965d62-8c84-48d5-80cd-a9d943337b63

https://github.com/user-attachments/assets/c806b7a5-be12-4b0c-bbda-37bb04bef4d8

Thank you for reading this detailed, please let me know if anything is unclear.

kisvegabor, 2024-09-19:
The goal here was to indicate the pressed cell with a mouse. However with a encoder and keypad we need to keep the cell highlighted during the navigation. 

Do you think we should keep the cell highlighted with mouse too? If so, how to remove the selection after a cell is clicked?

YangGangUEFI, 2024-09-19:
Hi Gabor,

I understand this goal.

For these two questions:

1. I just thought if we could keep the same behavior between each devices when pressing the cell. If so, i think keep the cell highlighted maybe better.
2. How about keep the selection(highlighted)? As if it's also keep the selection when clicked by keyboard.

Thanks for watching these questions.

kisvegabor, 2024-09-30:
I've just checked it again and I can't reproduce it now. In the `lv_demo_keypad_encoder` I an use the keyboard, but when I click a cell it just gets darker. (which is the pressed state any it's correct)

How does it work on your end with the latest master?

YangGangUEFI, 2024-10-09:
I've tested with the latest master. The behavior is same with before(tag v9.2.0).

The keyboard behavior you describe is the same as the behavior in my environment.

Question is that if mouse can have same behave with keyboard? 😅   and with previous comments, step 2. and 4. changes, does it lead to something bad?

Thanks again for checking and reproducing.

> 1. Test with `lv_demo_keypad_encoder()` table
> 2. Follow [LVGL 9 Group missing function `lv_group_set_click_focus` #5954 (comment)](https://github.com/lvgl/lvgl/issues/5954#issuecomment-2032990896) this to get the outlines when the widgets were clicked
> 3. After 2. step, table obj get the outlines but click cell only get flicker (Video 1 show that)
> 4. If remove following code in lv_table.c, when click cell in table, it can show the cell with bg_color (Video 2)
>    https://github.com/lvgl/lvgl/blob/7f57f37560c29ae6007b828fb93a61764f483e1f/src/widgets/table/lv_table.c#L599-L603

YangGangUEFI, 2024-10-16:
I test another things(e.g. `lv_list`) recently, and i think it's acceptable to keep mouse and keyboard have some different behavior.

So, i close this as it's current behavior of LVGL.

Thanks again!

closed 2024-10-16 by YangGangUEFI