lvgl/lvgl #6860: An optimization for drop-down scrolling menus

FASTSHIFT, 2024-09-15:
### Introduce the problem

The need for a drop-down menu is common during application development and is implemented using lvgl scroll, which redraws the entire obj tag when the object is scrolled. This is fine in most cases, but a transparent obj is created to take up space in such a drop-down menu scene, which leads to a lot of unnecessary full-screen redrawing and poor performance.

I found that it is possible to manually intercept the update time of the dirty area of the display and make a secondary clipping of the dirty area generated by scoll for targeted optimization, which can greatly reduce the redrawn area. Is there any other good method? Is it possible to implement it in a framework?

### Proposal

To view this, turn on the `LV_USE_REFR_DEBUG` option.

```c
#define SCROLL_OPTIMIZE 1

#if SCROLL_OPTIMIZE

typedef struct {
    lv_area_t area;
    lv_area_t last_area;
} scroll_ctx_t;

static void on_disp_event(lv_event_t* e)
{
    lv_area_t* area = lv_event_get_param(e);
    LV_LOG_USER("ori_area: (%d,%d) - (%d,%d)", area->x1, area->y1, area->x2, area->y2);

    const scroll_ctx_t* ctx = lv_event_get_user_data(e);
    _lv_area_intersect(area, &ctx->area, area);
    LV_LOG_USER("new_area: (%d,%d) - (%d,%d)", area->x1, area->y1, area->x2, area->y2);
}

static void on_scroll_event(lv_event_t* e)
{
    lv_obj_t* obj = lv_event_get_target_obj(e);
    lv_obj_t* child = lv_obj_get_child(obj, 0);
    scroll_ctx_t* ctx = lv_event_get_user_data(e);

    lv_area_t new_area;
    lv_obj_get_coords(child, &new_area);

    if (new_area.y2 >= ctx->last_area.y2) {
        LV_LOG_USER("scroll_down");
        /** 
         * When pulled down, the new area will completely cover the old area,
         * select the new area as the redraw area
         */
        ctx->area = new_area;
    } else {
        LV_LOG_USER("scroll_up");
        /**
         * When pulling up, you need to overwrite the background redraw,
         * select the old area as the redraw area
         */
        ctx->area = ctx->last_area;
    }

    ctx->last_area = new_area;
}

#endif /* SCROLL_OPTIMIZE */

void scroll_test(void)
{
    lv_obj_t* par = lv_obj_create(lv_scr_act());
    {
        lv_obj_remove_style_all(par);
        lv_obj_remove_flag(par, LV_OBJ_FLAG_SCROLL_ELASTIC);
        lv_obj_set_scroll_snap_y(par, LV_SCROLL_SNAP_CENTER);
        lv_obj_set_size(par, 480, 480);
        lv_obj_set_style_bg_color(par, lv_color_white(), 0);
        lv_obj_set_style_bg_opa(par, LV_OPA_COVER, 0);
        lv_obj_set_flex_flow(par, LV_FLEX_FLOW_COLUMN);
#if SCROLL_OPTIMIZE
        static scroll_ctx_t ctx;
        lv_area_set(&ctx.area, 0, 0, 480, 480);
        lv_area_set(&ctx.last_area, 0, 0, 480, 480);
        lv_obj_add_event_cb(par, on_scroll_event, LV_EVENT_SCROLL, &ctx);
        lv_display_add_event_cb(lv_obj_get_display(par), on_disp_event, LV_EVENT_INVALIDATE_AREA, &ctx);
#endif
    }

    lv_obj_t* obj = lv_obj_create(par);
    {
        lv_obj_remove_style_all(obj);
        lv_obj_set_size(obj, lv_pct(100), lv_pct(100));
        lv_obj_set_style_bg_color(obj, lv_palette_main(LV_PALETTE_BLUE), 0);
        lv_obj_set_style_bg_opa(obj, LV_OPA_COVER, 0);

        lv_obj_t* label = lv_label_create(obj);
        lv_label_set_text(label, "Scrollable Menu");
        lv_obj_center(label);
    }

    {
        lv_obj_t* obj_pad = lv_obj_create(par);
        lv_obj_remove_style_all(obj_pad);
        lv_obj_set_size(obj_pad, lv_pct(100), lv_pct(100));
    }
}
```

Before optimization:

https://github.com/user-attachments/assets/8e55bac7-9075-438f-b00f-18dfb9a70570

After optimization:

https://github.com/user-attachments/assets/c37e0edc-944c-4244-9ece-b7b08270d8b0

kisvegabor, 2024-09-17:
I'm tying to wrap my head around it so I made some test. I modified the bottom square like this:
```c

    {
        lv_obj_t* obj_pad = lv_obj_create(par);
        lv_obj_remove_style_all(obj_pad);
        lv_obj_set_size(obj_pad, lv_pct(100), lv_pct(100));

       //Add a label here too
        lv_obj_t* label = lv_label_create(obj_pad);
        lv_label_set_text(label, "Bottom Menu");
        lv_obj_center(label);
    }
```

With `SCROLL_OPTIMIZE 1` `"Bottom Menu"` is not shown. I haven't investigated the reason.

----

Anyway, I see the problem that you are about to solve. Please confirm if it's related to a single specific case: do not update the area of a widget if it's fully transparent. If so a more generic solution could be to determine automatically if a object don't have any content and it's transparent. E.g. skip the invalidation in `lv_obj_invalidate` if `opa` or `opa_layered` are `0`. (One invalidation needs to be added when the opacity becomes 0)

FASTSHIFT, 2024-09-18:
> I'm tying to wrap my head around it so I made some test. I modified the bottom square like this:
> 
> ```c
>     {
>         lv_obj_t* obj_pad = lv_obj_create(par);
>         lv_obj_remove_style_all(obj_pad);
>         lv_obj_set_size(obj_pad, lv_pct(100), lv_pct(100));
> 
>        //Add a label here too
>         lv_obj_t* label = lv_label_create(obj_pad);
>         lv_label_set_text(label, "Bottom Menu");
>         lv_obj_center(label);
>     }
> ```
> 
> With is not shown. I haven't investigated the reason.`SCROLL_OPTIMIZE 1``"Bottom Menu"`

This is normal because `obj_pad` is a completely transparent object to occupy a space, and no child should be drawn.

> Anyway, I see the problem that you are about to solve. Please confirm if it's related to a single specific case: do not update the area of a widget if it's fully transparent. If so a more generic solution could be to determine automatically if a object don't have any content and it's transparent. E.g. skip the invalidation in if or are . (One invalidation needs to be added when the opacity becomes 0)`lv_obj_invalidate``opa``opa_layered``0`

This may check a lot of conditions (need to check background, border, scrollbar mode), and can also cause a lot of invlidates if the number of children is large.

It may be difficult to make a complete automatic judgment. Can we add a flag to allow the user to select which invlidate mode to use in scroll? The default is full obj invlidate. After setting the flag, if there is no need to draw the background, invlidate child by invlidate child can be selected.

kisvegabor, 2024-09-30:
>> With SCROLL_OPTIMIZE 1 "Bottom Menu" is not shown. I haven't investigated the reason.

> This is normal because obj_pad is a completely transparent object to occupy a space, and no child should be drawn.

Okay, I can't see what makes LVGL not to render the children.

> This may check a lot of conditions (need to check background, border, scrollbar mode), and can also cause a lot of invlidates if the number of children is large.

Actually it's only the `opa` or the `opa_layered` proeperties. If these are 0, the children will be transparent as well.

closed 2024-10-23 by lvgl-bot