lvgl/lvgl #6862: Slider Events for Knob

fatiihyildizz, 2024-09-16:
### LVGL version

v9.1.0

### What happened?

I cannot scroll in the areas of the knob that are outside the slider. It does not detect an event. Why might this be happening?

[image]

C47D, 2024-09-16:
> I cannot scroll in the areas of the knob that are outside the slider bar.

What do you mean with scroll?

1. The knob not going all the way to the slider edge OR
2. Not being able to detect a click there?

As far as I know the widget is designed so the knob center is the limit of movement for the knob. If you meant option 2, you can use [extended-click-area](https://docs.lvgl.io/latest/en/html/widgets/obj.html#extended-click-area)

fatiihyildizz, 2024-09-17:
Hi,
I was talking about option 2. And this function worked for me.

Thank you

kisvegabor, 2024-09-17:
> If you meant option 2, you can use [extended-click-area](https://docs.lvgl.io/latest/en/html/widgets/obj.html#extended-click-area)

Confirmed

kisvegabor, 2024-10-02:
As the question seems to be answered I close this issue.

closed 2024-10-02 by kisvegabor