lvgl/lvgl #6865: lv_obj_add_style () may crash appl

glory-man, 2024-09-17:
### LVGL version

v8.3, v8.4

### What happened?

There is no checking of limits for
[`obj->style_cnt++;](https://github.com/lvgl/lvgl/blob/03e01bdd3e552da632a9b5e853d2efa3540c02d9/src/core/lv_obj_style.c#L91) may overflow 63 limit and become zero value

Also if you add the same style multiple times it will be added many times. Unlike version 7 where the same style could only be added once because the function `lv_obj_add_style()` in v7 uses a `_lv_style_list_remove_style()`  call to prevent multiple insertion of the same style

glory-man, 2024-09-17:
also I interested what will return `lv_mem_realloc()` couldn't have enough memory to allocate increased styles buffer, or how will work `lv_obj_add_style()` if `lv_mem_realloc()` returned NULL

kisvegabor, 2024-09-17:
In v9 we already have `LV_ASSERT(obj->style_cnt < 63);` and `    lv_obj_remove_style(obj, style, selector);` to address these. I think the same can be backported to v8. 

Regarding `lv_mem_realloc`, it returns `NULL` on out of memory. It's not ideal for sure, however the approach small memory allocations like this is that if you are out of memory here it's hard to recover and the system will crash anyway.

glory-man, 2024-09-17:
> Regarding `lv_mem_realloc`, it returns `NULL` on out of memory. It's not ideal for sure, however the approach small memory allocations like this is that if you are out of memory here it's hard to recover and the system will crash anyway.

maybe it makes sense to at least add `LV_ASSERT_MALLOC(obj->styles);` as this realized for labels

kisvegabor, 2024-10-02:
https://github.com/lvgl/lvgl/pull/6873 is merged, closing. Thank you!

closed 2024-10-02 by kisvegabor