lvgl/lvgl #6867: LV_USE_DRAW_DMA2D and LV_USE_DRAW_DMA2D_INTERRUPT question

tdjastrzebski, 2024-09-17:
### Introduce the problem

I am trying to grasp the current concept behind `LV_USE_DRAW_DMA2D` and `LV_USE_DRAW_DMA2D_INTERRUPT`.
What is stated in [lvgl\docs\integration\chip\stm32.rst](https://github.com/lvgl/lvgl/blob/master/docs/integration/chip/stm32.rst) makes sense but I do not quite understand why `LV_USE_DRAW_DMA2D_INTERRUPT` requires some RTOS or, otherwise, `lv_draw_dma2d_transfer_complete_interrupt_handler()` function does nothing - see [here](https://github.com/lvgl/lvgl/blob/eecb3c774a1431394d4947abe498f7b0826bb7c7/src/draw/dma2d/lv_draw_dma2d.c#L115). `LV_DRAW_DMA2D_ASYNC` is defined only when `LV_USE_OS` is defined - see [here](https://github.com/lvgl/lvgl/blob/eecb3c774a1431394d4947abe498f7b0826bb7c7/src/draw/dma2d/lv_draw_dma2d_private.h#L27).
That is, why not to simply call `lv_draw_dma2d_transfer_complete_interrupt_handler()` function upon DMA2D complete IRQ to invoke the next step in the render process pipeline?
What is the design scenario here?
Nothing against RTOSes but, although very convenient to use, they introduce overhead which at times outweighs benefits and features which may be just not required.

### Proposal

Make `lv_draw_dma2d_transfer_complete_interrupt_handler()` function callable regardless of whether RTOS is used or not.

liamHowatt, 2024-09-18:
I understand your proposal. What you propose is actually how it was initially, but there was a good case made [here](https://github.com/lvgl/lvgl/pull/6691#pullrequestreview-2244124443) to reduce the number of "modes" it can be in. There were far more preprocessor cases (`#if`) before the simplification.

From the linked comment:

> **Interrupt, no OS**: We can't do much here, fallback to 1)

Maybe @kisvegabor misunderstood the power of supporting the interrupt with no OS, which you've described well.

kisvegabor, 2024-09-18:
My way of thinking was this in case of no OS but with IRQ:
1. DMA2D start e.g. a fill in `dispatch_cb` and exits.
2. Now 2 things can happen:
   1. The software render can start drawing something else. -> We can't do anything else while any of these are working s bot draw units are busy
   2. There is no task for SW rendering -> We can't do anything while DMA2D is working, except try dispatching because a new draw_task might be added in the meantime. 
3. In either cases, we need to wait for finish of SW or DMA2D and keep dispatching. And exactly this is what's happening. In case of `not LV_DRAW_DMA2D_ASYNC` we immediately call `lv_draw_dispatch_request` [here](https://github.com/lvgl/lvgl/blob/eecb3c774a1431394d4947abe498f7b0826bb7c7/src/draw/dma2d/lv_draw_dma2d.c#L115) to make LVGL call the dispatcher again to check if the DMA2D is ready [here](https://github.com/lvgl/lvgl/blob/eecb3c774a1431394d4947abe498f7b0826bb7c7/src/draw/dma2d/lv_draw_dma2d.c#L337). If not ready `LV_DRAW_UNIT_IDLE` will be [returned](https://github.com/lvgl/lvgl/blob/eecb3c774a1431394d4947abe498f7b0826bb7c7/src/draw/dma2d/lv_draw_dma2d.c#L338) which will trigger an other `lv_draw_dispatch_request` internally. 

I can easily happen the I missed something, so please be critical :slightly_smiling_face:

tdjastrzebski, 2024-09-18:
I think the above assumes that LVGL is the only or at least the main process running on MPU, which may not be the case.
However important, LVGL may only be a presentation layer while max MPU time is needed somewhere else entirely.
While DMA2D task is cranking, I think MPU should be able to handle another, probably non-LVGL process in the meantime, at least until DMA2D task is done.

tdjastrzebski, 2024-09-19:
One more thought: IF LVGL is supposed to nicely hide some complexity I would at least suggest making it RTOS-agnostic.
This would require additional `continue_cb()` invoked upon (i.e.) DMA2D_Complete where (i.e.) `tx_event_flags_set()` could be called to unblock whatever block was set in `wait_cb()`.
In case of no-RTOS setup, within `continue_cb()` user would have to call `lv_continue()` function. This way DMA2D_Complete() would be handled within LVGL. As a consequence LVGL would have to handle buffer flush as well because DMA2D would be unavailable to non-LVGL code.

liamHowatt, 2024-09-20:
After reviewing the logic that had existed when "no OS + interrupt" was possible, I see that it was pointless. It just spun in the dispatcher instead of the dma2d code.

To do what you suggest for no-OS, it would take an overhaul of the draw pipeline, which you touched upon. From past experience, turning sync code into no-OS async code is a big endeavor because state has to be saved at the yield points and returned to. Normal functions become state machines, and state machines take over everything. It would be good if it could be done in a way that doesn't require multiple parts of LVGL or user code to become state machines.

tdjastrzebski, 2024-09-20:
That is right, state has to be saved. Practically it means stored in shared structs, rather than kept in function chain of calls.

liamHowatt, 2024-10-01:
It could definitely work. I've used this pattern before. My hesitation is that this will only benefit bare metal users and slightly negatively impact OS users and add complexity. If it can be done in a way that will not increase the cost of maintaining code after it has been converted to a state machine and not add too much overhead when OS is used (state machine yield points not used) then it can be worth reviewing a possible implementation. Do you have interest in opening a PR?

closed 2024-10-27 by lvgl-bot

tdjastrzebski, 2024-10-27:
I think, eventually I will get back to it and open a PR - once I am certain it is needed. At this moment I am very busy designing hardware.