lvgl/lvgl #6870: v9.2.0 PXP incomplete types

faxe1008, 2024-09-18:
### LVGL version

v9.2.0

### What happened?

The move to private APIs has caused build breakage for NXP PXP engine:
```
modules/lib/gui/lvgl/src/draw/nxp/pxp/lv_draw_pxp_fill.c: In function 'lv_draw_pxp_fill':
modules/lib/gui/lvgl/src/draw/nxp/pxp/lv_draw_pxp_fill.c:55:35: error: invalid use of incomplete typedef 'lv_draw_unit_t'
   55 |     lv_layer_t * layer = draw_unit->target_layer;
      |                                   ^~
modules/lib/gui/lvgl/src/draw/nxp/pxp/lv_draw_pxp_fill.c:63:43: error: invalid use of incomplete typedef 'lv_draw_unit_t'
   63 |     lv_area_copy(&rel_clip_area, draw_unit->clip_area);
      |                                           ^~
modules/lib/gui/lvgl/src/draw/nxp/pxp/lv_draw_pxp_fill.c:67:9: warning: implicit declaration of function 'lv_area_intersect'; did you mean 'lv_area_increase'? [-Wimplicit-function-declaration]
   67 |     if(!lv_area_intersect(&blend_area, &rel_coords, &rel_clip_area))
      |         ^~~~~~~~~~~~~~~~~
      |         lv_area_increase
```

This happens despite declaring `LV_USE_PRIVATE_API` in `lv_conf.h`.

kisvegabor, 2024-09-18:
`LV_USE_PRIVATE_API` adds `lvgl_private.h` only to `lvgl.h`. Internally usually not `lvgl.h` is included but only the required headers. 

Does it help if you include `lv_draw_private.h` directly?

kschwan, 2024-09-18:
I have the same problem. I fixed it by including `lv_draw_sw_private.h` from `lv_draw_pxp.h`. Not sure if it's the "correct" way to fix it though

```
diff --git a/src/draw/nxp/pxp/lv_draw_pxp.h b/src/draw/nxp/pxp/lv_draw_pxp.h
index d5e98d791..eb0f69e17 100644
--- a/src/draw/nxp/pxp/lv_draw_pxp.h
+++ b/src/draw/nxp/pxp/lv_draw_pxp.h
@@ -24,6 +24,7 @@ extern "C" {

 #if LV_USE_DRAW_PXP
 #include "../../sw/lv_draw_sw.h"
+#include "../../sw/lv_draw_sw_private.h"

 /*********************
  *      DEFINES`
```

faxe1008, 2024-09-23:
@kisvegabor adding `LV_USE_PRIVATE_API` does not suffice, I suppose that the private header is not included in the right spots in the PXP engine.

kisvegabor, 2024-09-25:
Is this issue present in `master` too?

@nicusorcitu Is it already fixed on your side?

nicusorcitu, 2024-09-25:
We fixed the PXP build, the patch will be send till 1 oct together with others for 9.2.1.

nicusorcitu, 2024-09-25:
This should fix it if you really need it these days:
[image]

nicusorcitu, 2024-09-25:
@cosmindanielradu19

faxe1008, 2024-10-15:
I can confirm that PXP now compiles on rev `4a75a62fe208a6e52db8e8c3f7916d76aabbd688`.

closed 2024-10-15 by faxe1008

kisvegabor, 2024-10-15:
Thank you for the feedback!