lvgl/lvgl #6874: LVGL with PXP on doesn't draw (unless `LV_USE_PERF_MONITOR` is enabled)

kschwan, 2024-09-18:
### LVGL version

v9.2.0-58-g1dd63c7e5

### What happened?

I'm seeing a weird issue: LVGL is not refreshing if I disable LV_USE_PERF_MONITOR. With LV_USE_PERF_MONITOR enabled, it seems to be working correctly.

I am using PXP + FreeRTOS. The display updates correctly if PXP is disabled regardless of `LV_USE_PERF_MONITOR`.

`lv_timer_handler()` appears to be returning `0xffffffff`. If I work around this and manually compute the delay, things seem to be working correctly.

kschwan, 2024-09-18:
Note: if I use PXP on v9.2.0-58-g1dd63c7e5, I'm seeing the issue of #6793

nicusorcitu, 2024-09-26:
@cosmindanielradu19

nicusorcitu, 2024-09-26:
Note: PXP is not officially ready on LVGL 9. There is still work in progress, but not officially released as supported.
Next SDK 24.12 will have the PXP support for LVGL v9.

I have done many tests these days and I saw no such issues on my RT1170 with PXP enabled.

> Note: if I use PXP on v9.2.0-58-g1dd63c7e5, I'm seeing the issue of https://github.com/lvgl/lvgl/pull/6793
>
Please use the latest from master branch to include the freertos sync fixes as well.

kschwan, 2024-09-26:
> Note: PXP is not officially ready on LVGL 9. There is still work in progress, but not officially released as supported.
> Next SDK 24.12 will have the PXP support for LVGL v9.

Oh, I see. I figured it would be fine to use since it's there ^^
I suppose you mean the next version of MCUXpresso SDK? Can you say when it's supposed to be released. And also, how is VGLite support? Sorry for the off-topic questions

> Please use the latest from master branch to include the freertos sync fixes as well.
Thanks, I'll try with the latest updates. I have some other tasks currently, but I should be able to check at least early next week.

nicusorcitu, 2024-09-27:
The SDK 24.12 will be released on December 24 (here the name comes).
VGLite support will be there as well for all the platforms.

kisvegabor, 2024-09-30:
Nice Christmas gist :slightly_smiling_face:

kschwan, 2024-10-04:
I'm a bit. The samme thing appears to happen on v9.2.0-102-ga853a1289.

`lv_timer_handler()` runs two times after reset: the first time it returns `0`, the second time it returns `0xffffffff`.

[image]

nicusorcitu, 2024-10-04:
Probably you do not have the tick hook.
This does not look as a PXP issue.

kschwan, 2024-10-04:
> Probably you do not have the tick hook. This does not look as a PXP issue.

I do. And as soon as I enable the perf monitor (overlay), things are working as expected.

kschwan, 2024-10-04:
And it is working also working if I disable PXP :o

kschwan, 2024-10-10:
I'll see if I can produce a an example for you to test. I'm on a custom platform but might be able to make something for the MIMXRT1170-EVK. Would that work for you?

nicusorcitu, 2024-10-10:
Please try reproducing the problem with latest 9.2.1 and let me know if any issue.

nicusorcitu, 2024-10-10:
As Gabor suggested you could also try reverting this patch: https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136

kschwan, 2024-10-10:
> As Gabor suggested you could also try reverting this patch: [d824618](https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136)

Ah, didn't notice that comment, sorry. I'll check

kschwan, 2024-10-14:
> As Gabor suggested you could also try reverting this patch: [d824618](https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136)

`git revert d824618` on top of v9.2.0-133-g255c79171 also solves the issue

nicusorcitu, 2024-10-14:
So it is not the PXP but this patch is the rootcause: [d824618](https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136)

kschwan, 2024-10-14:
> So it is not the PXP but this patch is the rootcause: [d824618](https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136)

Correct :)

kschwan, 2024-10-14:
> > So it is not the PXP but this patch is the rootcause: [d824618](https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136)
> 
> Correct :)
> 
> I'll edit the title

There is a relation somehow because it also works as expected if I disable PXP. I suppose it has to do with how the tasks are interacting.

Orange-Murker, 2024-10-22:
I am facing the exact same issue with FreeRTOS running on ESP32

kisvegabor, 2024-11-06:
Hey,

Could you try these?
1. Try with the latest master
2. Try `git bisect` to find where it went wrong.

kschwan, 2024-11-06:
> Hey,
> 
> Could you try these?
> 
> 1. Try with the latest master
> 2. Try `git bisect` to find where it went wrong.

Thanks, I'll try to make time for it

Orange-Murker, 2024-11-06:
> Hey,
> 
> Could you try these?
> 
>     1. Try with the latest master
> 
>     2. Try `git bisect` to find where it went wrong.

1. Still does not work on the latest master
2. Reverting d824618366e474eef57081d647031965e6edc136 fixes the issue

kschwan, 2024-11-06:
> > Hey,
> > Could you try these?
> > ```
> > 1. Try with the latest master
> > 
> > 2. Try `git bisect` to find where it went wrong.
> > ```
> 
> 1. Still does not work on the latest master
> 2. Reverting [d824618](https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136) fixes the issue

Right, I found the same thing

kschwan, 2024-11-19:
Any news on this? Let me know if I can provide more info

kschwan, 2024-12-14:
Not abandoned

kisvegabor, 2024-12-18:
@uLipe Can you reproduce it with PXP on your i.MX RT 1170 board?

kisvegabor, 2024-12-22:
That's great, as at least we can reproduce it. We can also take a look but I wonder if the NXP team (@nicusorcitu @cosmindanielradu19 @anaGrad ) is also already working on it.

kisvegabor, 2025-01-13:
@uLipe I've assigned this issue to you to have a responsible on our side.

kschwan, 2025-01-14:
> Hi @kschwan @uLipe, can you share the config used for this test? Have you used PXP + VGLite or PXP + SW for testing? Thanks

Sure! This should be it. This was with PXP + SW.

[lv_conf.h.txt](https://github.com/user-attachments/files/18409270/lv_conf.h.txt)

cosmindanielradu19, 2025-01-14:
> > Hi @kschwan @uLipe, can you share the config used for this test? Have you used PXP + VGLite or PXP + SW for testing? Thanks
> 
> Sure! This should be it. This was with PXP + SW.
> 
> [lv_conf.h.txt](https://github.com/user-attachments/files/18409270/lv_conf.h.txt)

Thanks! I'll look into it

kschwan, 2025-02-11:
Not abandoned :o

liamHowatt, 2025-02-19:
I think I figured out what's going on here. It's not necessarily a bug. `lv_timer_handler` returns `LV_NO_TIMER_READY` (`0xFFFFFFFF`) if the perf monitor is disabled and the user has not created any indevs or timers. The way most people use LVGL, the driver they use creates indevs or at least one timer like SDL's `event_handler_timer`.

> LVGL is not refreshing

I believe it refreshes and flushes once. That's what I observe.

The reason this is not a bug is because LVGL knows nothing will change on the screen (no anims) and there's nothing to check (no indevs) until the user calls some LVGL functions, so it returns `LV_NO_TIMER_READY`.

[Someone commented here](https://github.com/lvgl/lvgl/issues/7724#issuecomment-2665151158) that callers of `lv_timer_handler` should handle `LV_NO_TIMER_READY` accordingly. 

The users most susceptible to an issue are ones porting LVGL incrementally.

I see two possible solutions.
1. Update all docs and driver uses of `lv_timer_handler` to show this handling
   ```c
   uint32_t ms_delay = lv_timer_handler();
   if(ms_delay == LV_NO_TIMER_READY) ms_delay = LV_DEF_REFR_PERIOD;
   ```
   to educate users about that possible return value.
2. Make lv_timer_handler return `LV_DEF_REFR_PERIOD` instead of `LV_NO_TIMER_READY` so that a more reasonable value will be passed to the delay function that usually follows the call. I think this would be an unfortunate solution because we'd take away the user's freedom to know "nothing will happen until I change an object, start an animation, indev, etc."

### How the issue happens

The global anim timer is paused during `lv_init`, and any display timers pause before doing a single flush. The perf mon is disabled.

https://github.com/lvgl/lvgl/blob/377cc5aa801dcb4ca757611238651cd8deed3b96/src/core/lv_refr.c#L363-L365

`lv_timer_handler` returns `LV_NO_TIMER_READY` because all timers are paused. No indevs, anims, other timers, etc. were created.

https://github.com/lvgl/lvgl/blob/377cc5aa801dcb4ca757611238651cd8deed3b96/src/misc/lv_timer.c#L119-L129

vwheeler63, 2025-02-19:
Brilliant analysis!

kschwan, 2025-02-20:
> I think I figured out what's going on here. It's not necessarily a bug. lv_timer_handler returns LV_NO_TIMER_READY (0xFFFFFFFF) if the perf monitor is disabled and the user has not created any indevs or timers.

Aha, it's as simple as that :)

Thanks for taking the time to investigate. I think updating the docs/examples to check for `LV_NO_TIMER_READY` would be good.

PR #7920 (2025-03-28): docs(timer): document LV_NO_TIMER_READY
Fixes #6874
Fixes #7724

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3. [... cut]
closed 2025-03-28 by uLipe