lvgl/lvgl #6875: `lv_draw_sw_rotate()` opposite of `lv_draw_pxp_rotate()`

kschwan, 2024-09-18:
### LVGL version

v9.2.0-58-g1dd63c7e5

### What happened?

Somewhere between release/v9.1 and master (1dd63c7e5) `lv_draw_sw_rotate()` started rotating the opposite direction of `lv_draw_pxp_rotate()`- Ie., if I rotate a buffer 90 deg, `lv_draw_sw_rotate()` is flipped compared to `lv_draw_pxp_rotate()`. It's not the case on release/v9.1.

kisvegabor, 2024-09-25:
We have really discovered some misalignment recently. Now `lv_draw_sw_rotate()` and `lv_display_rotate_area()` are thinking the same way. 

cc @nicusorcitu

nicusorcitu, 2024-09-25:
@cosmindanielradu19

cosmindanielradu19, 2024-09-25:
I've seen that in SW, all the rotate90_X functions are rotating at 270 degrees, and rotate270_X functions are rotating at 90 degrees.

nicusorcitu, 2024-09-25:
See also https://github.com/lvgl/lvgl/issues/6885

kisvegabor, 2024-09-26:
>  I've seen that in SW, all the rotate90_X functions are rotating at 270 degrees, and rotate270_X functions are rotating at 90 degrees.

How do we know which is the correct direction?

cosmindanielradu19, 2024-09-26:
> > I've seen that in SW, all the rotate90_X functions are rotating at 270 degrees, and rotate270_X functions are rotating at 90 degrees.
> 
> How do we know which is the correct direction?

I compared the output with PXP rotation and one is clockwise the other one is counterclockwise

kschwan, 2024-09-26:
> > I've seen that in SW, all the rotate90_X functions are rotating at 270 degrees, and rotate270_X functions are rotating at 90 degrees.
> 
> How do we know which is the correct direction?

Looks like `lv_draw_sw_rotate()` changed while `lv_draw_pxp_rotate()` stayed the same

liamHowatt, 2024-10-01:
`lv_draw_sw_rotate` was clockwise in v9.1 and now it is counterclockwise in master. It changed in this PR: #5703

Clockwise is more intuitive but counterclockwise is more correct in my opinion.

cosmindanielradu19, 2024-10-01:
> `lv_draw_sw_rotate` was clockwise in v9.1 and now it is counterclockwise in master. It changed in this PR: #5703
> 
> Clockwise is more intuitive but counterclockwise is more correct in my opinion.

I believe that all the draw units should rotate in the same direction because it might be confusing at some point if you are doing development with CPU, then you want to use a board with PXP, and suddenly your display is upside down.

kisvegabor, 2024-10-02:
Absolutely. As none of the directions are better let's use the SW render as a reference.

kisvegabor, 2024-10-11:
It is already fixed in PXP, right?

nicusorcitu, 2024-10-14:
No, not in this release. The PXP rotation have not changed in 9.2.1.

nicusorcitu, 2024-10-14:
@cosmindanielradu19 please prepare a patch to sync PXP with CPU and create a PR upstream to be integrated to 9.2.1 as well.
We need to keep the consistency.

cosmindanielradu19, 2024-10-14:
I've created a PR with the patch

#7063

liamHowatt, 2024-10-15:
Fixed by #7063

closed 2024-10-15 by liamHowatt

kschwan, 2025-05-16:
#7063 was merged into release/v9.2, but not master. So the issue persists on master

@liamHowatt, @cosmindanielradu19

cosmindanielradu19, 2025-05-16:
@kisvegabor do you mind if I'm adding this to #8156 ?

nicusorcitu, 2025-05-16:
Please do it - the change is harmless for any others except PXP users.

PR #7063 (2024-10-14): fix(pxp): sync rotation direction with SW render
Fixes #6875 

Sync the PXP rotation direction with the SW direction

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/ [... cut]