lvgl/lvgl #6876: lv_anim_t's completed_cb is not called anymore with no exec_cb is set and used with anim_timeline

Viatorus, 2024-09-18:
### LVGL version

9.2.0

### What happened?

The behavior changed from 8.x to 9.x.

In 8.x a virtual exec_cb was set in case of an absent exec_cb:
https://github.com/lvgl/lvgl/blob/4d96c27ce35dd2ea6b34926f24a647e7ea7c4b0c/src/misc/lv_anim_timeline.c#L84-L88

In 9.x not anymore: 
https://github.com/lvgl/lvgl/blame/a89cda10c103276d9efde6d5a3b3e68eaa159324/src/misc/lv_anim_timeline.c#L76C6-L76C26

So the animation is never started here:
https://github.com/lvgl/lvgl/blob/a89cda10c103276d9efde6d5a3b3e68eaa159324/src/misc/lv_anim_timeline.c#L209-L211

Viatorus, 2024-09-18:
A quick solution could be to reintroduce the `lv_anim_timeline_virtual_exec_cb`.

A better solution, but with major changes, might be to make `anim_timeline_set_act_time` work with `exec_cb == null`.

kisvegabor, 2024-09-30:
Hey,

I haven't tested it but it seems 
```c
        if(a->exec_cb == NULL && a->custom_exec_cb == NULL) {
            continue;
        }
```
can be removed as, below it's checked if `exec_cb` or `custom_exec_cb` are `NULL` before calling them.

PR #7043 (2024-10-12): fix(anim_timeline): run animations which do not have an exec cb
Fixes #6876
closed 2024-10-12 by FASTSHIFT