lvgl/lvgl #6885: Rotation differs v9.1 and v9.2

focussing, 2024-09-19:
### LVGL version

v9.1

### What happened?

I have 2 boards containing a Raspberry Pi RP2040 in combination with a 480x320 px tft controlled via SPI.
The 2 displays have the same resolution but are from different manufacturers. Both have their control lines on the left side of the board, however the flatcable is on the left (board #1) the other has it on the right (board #2).

Initially I made my application using Arduino IDE with LVGL v9.1 on board #1 and it works fine.

The same application runs also on board #2 but it is upside down. I cannot get it right using `lv_display_set_rotation(..,..)` function. Calling it with LV_DISPLAY_ROTATION_0, 90, 180 or 270 all giving the same result. All showing the application gui upside down. Looks that the `lv_display_set_orientation` is not working.
When I update LVGL to v9.2 the gui is shown perfectly when using `LV_DISPLAY_ROTATION_90` or `LV_DISPLAY_ROTATION_270`, and rotated over 90 degrees when using `LV_DISPLAY_ROTATION_0` or `LV_DISPLAY_ROTATION_180`.

Next I programmed the v9.2 application on board #1 and now the gui is shown upside down (!) when using `LV_DISPLAY_ROTATION_90` or`DISPLAY_ROTATION_270`, and rotated over 90 degrees (distorted) when using `LV_DISPLAY_ROTATION_0` or `LV_DISPLAY_ROTATION_180`.

Any idea what is going on and how I

focussing, 2024-09-20:
Never mind, my mistake

focussing, 2024-09-20:
Reopening

kisvegabor, 2024-09-26:
These can be really tricky. Please use the latest `master` as there were some related updates in it recently. 

To have a stable starting point, on one display you should see a not cropped normally working UI where the resolution in LVGL and the resolution of display are set the the same value. This can be achieved by setting the Memory Access Control (36h) register of the chip to the correct initial value. 

For the other display, I think all you need to do is setting the Memory Access Control register differently. If needed for the touch you can also set `LV_DISPLAY_ROTATION_180`.

kisvegabor, 2024-10-11:
Is it already fixed in PXP, right?

focussing, 2024-10-14:
@kisvegabor I already use the latest version that is shown in Arduino library manager, do you have an indication when to expect release 9.3? I prefer to only use released versions as our customers are using the source code as well.

kisvegabor, 2024-10-15:
We will release v9.2.1. in a few days. Just waiting for the last feedback.

v9.3 will be released early 2025.

kisvegabor, 2024-10-30:
As the original issue should be resolved I close this now.

focussing, 2024-11-07:
Hi @kisvegabor it is not solved, not by 9.2.1 nor 9.2.1. How do I reopen it?
It becomes even more strange when I implement touch to the dispaly... 

[image]

I still need to set rotation to 270 degrees via `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);`.
This picture shows the xy coordinates at all 4 corners when I touch using a stylus. And the orange rectangles show the active areas of the first 2 buttons, which are not aligned to where the button is drawn. The rightmost button cannot even be activated.
The initialization of the touch part uses `touchscreen.setRotation(0);` (using XPT2046_Touchscreen library v1.4).

kisvegabor, 2024-11-07:
It seems LVGL rotated the touch correctly. How do you rotate the image? It should done either 
- by  the display controller
- by using `lv_draw_sw_rotate`.

focussing, 2024-11-07:
I did not rotate the image itself, I rotate the display object via `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);`.
The image it is part of the active screen like labels, buttons, etc.

```
void gui_init() {
  // register print function for debugging
#if LV_USE_LOG != 0
  lv_log_register_print_cb(log_print);
#endif

  // create display object
  disp = lv_tft_espi_create(SCREEN_HEIGHT, SCREEN_WIDTH, draw_buf, sizeof(draw_buf));
  lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);

  // initialize touchscreen
  touchscreen.begin(SPI);
  touchscreen.setRotation(0);

  // initialize lvgl input device object
  lv_indev_t *indev = lv_indev_create();
  lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER); /*Touchpad should have POINTER type*/
  lv_indev_set_read_cb(indev, touchscreen_read);

  // display background color
  lv_obj_set_style_bg_color(lv_screen_active(), lv_color_white(), LV_PART_MAIN);
}
```
The rotation what is needed depends on the part. We ordered parts in China that seem equal but needed different rotation, eg 270 or 90 degrees.

focussing, 2024-11-08:
If I don't swap this is the result...

[image]

kisvegabor, 2024-11-14:
So, v9 doesn't do the actual rotation for you. `lv_display_set_rotation` just swaps the the resolution and transforms the touch points. 

You need to use `lv_draw_sw_rotate` for the image rotation part as  it give your more flexibility to control the buffers. This function just rotates the buffer to an other buffer. You should also use `lv_display_rotate_area` to get the rotated coordinates. 

@liamHowatt and I will an example about it next week.

focussing, 2024-12-10:
I am still curious about the example...

liamHowatt, 2024-12-10:
The PR was merged and the example is now visible in the online docs here: https://docs.lvgl.io/master/details/main-components/display.html#rotation

focussing, 2025-01-07:
This week I will test the updated docs with my setup, thanks!

focussing, 2025-01-21:
I tried to test your example but it does not work at all.

Updated the set rotation function with the callback hook:
```
void gui_set_rotation(int rot) {
  switch (rot) {
    case 0:
      disp = lv_tft_espi_create(SCREEN_WIDTH, SCREEN_HEIGHT, draw_buf, sizeof(draw_buf));
      lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_0);
      break;
    case 90:
      disp = lv_tft_espi_create(SCREEN_HEIGHT, SCREEN_WIDTH, draw_buf, sizeof(draw_buf));
      lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);
      break;
    case 180:
      disp = lv_tft_espi_create(SCREEN_WIDTH, SCREEN_HEIGHT, draw_buf, sizeof(draw_buf));
      lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);
      break;
    case 270:
      disp = lv_tft_espi_create(SCREEN_HEIGHT, SCREEN_WIDTH, draw_buf, sizeof(draw_buf));
      lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);
      break;
  }

  lv_display_set_flush_cb(disp, flush_cb);
}
```

From the example you provided in the doc I had to comment out `my_set_window` and `my_send_colors()`. 
After doing that the following error occurs:

```
C:/Users/..../AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe:C:\Users\....\AppData\Local\Temp\arduino\sketches\63AEB0FCFB6647F58F1A2E56AFB944A9/memmap_default.ld:235 cannot move location counter backwards (from 200a5624 to 20040000)
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

It is becoming quite critical as I still do not have been able to come with a working LCD display with touch.
Are you sure it has nothing to do with the TFT_eSPI that I need to use also?
Do you have any suggestion what to do, otherwise I might be forced to abandon the whole project....

liamHowatt, 2025-01-23:
`my_set_window` and `my_send_colors` are to show where you should add your own functions that do those things. Can you send your full flush callback?

focussing, 2025-02-13:
I will provide this infomation this weekend. I will remove all non-related stuff from my current project and test the basics setup first.
Thank you!!

focussing, 2025-02-21:
Hello @kisvegabor @liamHowatt 

Wanted to let you know that I fixed the issue.
The trick was that x and y are swapped because of the rotation of 90 resp 270 degrees.
So for x I had to use the vertical resolution and for y the horizontal resolution.

```
void touchscreen_read(lv_indev_t* indev, lv_indev_data_t* data) {
    int x, y, z;

    if (touchscreen.tirqTouched() && touchscreen.touched()) {
        TS_Point p = touchscreen.getPoint();

        x = map(p.x, 300, 3800, 1, LV_VER_RES);
        y = map(p.y, 150, 3900, 1, LV_HOR_RES);
        z = p.z;

        data->state = LV_INDEV_STATE_PRESSED;

        // set the coordinates
        data->point.x = x;
        data->point.y = y;
    } else {
        data->state = LV_INDEV_STATE_RELEASED;
    }
}
```
Thanks a lot for all your help and patience!

kisvegabor, 2025-02-25:
Happy to here it! There is something clearly unintuitive on our side, as it seem multiple people are struggling with it. I keep an on this topic.