lvgl/lvgl #6886: Lvgl deinit functions are not freeing resources

AmritpalSinghSarao, 2024-09-19:
### LVGL version

8.3.11

### What happened?

(00:00:00.177) Heap libero: 277420 bytes
W (00:00:00.183)  Heap minimo mai raggiunto: 277012 bytes
W (00:00:00.190) Heap libero per allocazioni 8-bit: 277420 bytes
I (00:00:00.200) Install panel IO
I (00:00:00.204) Install SSD1306 panel driver
I (00:00:00.302) `Initialize LVGL`
W (00:00:00.302) Heap libero: 276656 bytes
W (00:00:00.303) Heap minimo mai raggiunto: 276656 bytes
W (00:00:00.310) Heap libero per allocazioni 8-bit: 276656 bytes
I (00:00:00.318) `LVGL: Starting LVGL task`
W (00:00:00.329) lcd_panel.io.i2c: Callback on_color_trans_done was already set and now it was owerwritten!
I (00:00:00.344) Display LCD initiated
W (00:00:00.345) Heap libero: 255548 bytes
W (00:00:00.347) Heap minimo mai raggiunto: 255548 bytes
W (00:00:00.354) Heap libero per allocazioni 8-bit: 255548 bytes
I (00:00:00.369) Start deinit Display!
W (00:00:00.374) Heap libero: 252860 bytes
W (00:00:00.380) Heap minimo mai raggiunto: 252860 bytes
W (00:00:00.387) Heap libero per allocazioni 8-bit: 252860 bytes
I (00:00:00.462) Display turned on
I (00:00:00.502) Display turned off
I (00:00:00.506) esp_lcd_panel_del
W (00:00:00.506) Heap libero: 252860 bytes
W (00:00:00.508) Heap minimo mai raggiunto: 252860 bytes
W (00:00:00.515) Heap libero per allocazioni 8-bit: 252860 bytes
I (00:00:00.523) esp_lcd_panel_io_del
W (00:00:00.529) Heap libero: 252932 bytes
W (00:00:00.535) Heap minimo mai raggiunto: 252860 bytes
W (00:00:00.542) Heap libero per allocazioni 8-bit: 252932 bytes
I (00:00:00.550)` lvgl_port_deinit`
W (00:00:00.556) Heap libero: 253280 bytes
W (00:00:00.562) Heap minimo mai raggiunto: 252860 bytes
W (00:00:00.569) Heap libero per allocazioni 8-bit: 253280 bytes
I (00:00:00.577) i2c_driver_delete
W (00:00:00.582) Heap libero: 253412 bytes
W (00:00:00.588) Heap minimo mai raggiunto: 252860 bytes
W (00:00:00.596) Heap libero per allocazioni 8-bit: 253412 bytes
I (00:00:00.604) Done
W (00:00:03.622) Heap libero: 258196 bytes
W (00:00:03.622)  Heap minimo mai raggiunto: 252860 bytes
W (00:00:03.623)  Heap libero per allocazioni 8-bit: 258196 bytes

This can be seen from logs that `lvgl_port_deinit();` `lv_deinit();` don't deallocate the memory. 

I am missing something?

I am using Lvgl on esp32.

kisvegabor, 2024-09-25:
If I recall correctly it really wasn't perfect in v8. Is migrating to v9 an option for you?

kissa96, 2024-10-28:
Hi @AmritpalSinghSarao have you resolved this issuse?

kisvegabor, 2024-11-15:
Closing in lack of feedback.

closed 2024-11-15 by kisvegabor