lvgl/lvgl #6889: lodepng: decoder_open: error 83: memory allocation failed

mtojek, 2024-09-19:
### LVGL version

v9.2.0

### What happened?

Hey!

I'm having a hard time figuring out how should I modify `lv_conf.h` to display PNG file using lodepng:

```
[Error] (7.180, +8)      lv_draw_image_normal_helper: Failed to open image lv_draw_image.c:174
[Warn]  (7.242, +62)     decoder_open: error 83: memory allocation failed
```

I want to run the code on ESP32-S3 (Lolin32). I was able to easily display PNG images with v8.3.2, but API in v9 changed, and I can't determine which `#define` values should I use.

### How to reproduce?

I'm experimenting with:

```
// LVGL memory
#define LV_MEM_CUSTOM 1
// #define LV_MEM_SIZE (128U * 1024U)
#define LV_MEM_CUSTOM_INCLUDE <esp32-hal-psram.h>
#define LV_MEM_CUSTOM_ALLOC(size) ps_malloc
#define LV_MEM_CUSTOM_FREE(ptr) ps_free
```

kisvegabor, 2024-10-03:
So is the goal to change to new allocators? You can select `LV_STDLIB_CUSTOM` [here](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L43) and create a new file similar to [this](https://github.com/lvgl/lvgl/blob/master/src/stdlib/clib/lv_mem_core_clib.c).

Actually I think we should have a ready to use config for ESP.

cc @tore-espressif @espzav

kissa96, 2024-10-21:
@mtojek How is this goin' on?

kisvegabor, 2024-11-13:
Closing in lack of feedback.

closed 2024-11-13 by kisvegabor