lvgl/lvgl #6892: lv_calendar_chinese.h(47,23): error C2526: “lv_calendar_gregorian_to_chinese”: C 链接函数无法返回 C++ 类“_lv_calendar_chinese_t”

xinsuinizhuan, 2024-09-20:
### LVGL version

9.3

### What happened?

mvs compile：
lv_calendar_chinese.h(47,23): error C2526: “lv_calendar_gregorian_to_chinese”: C link can not reture  C++ class“_lv_calendar_chinese_t”

lhdjply, 2024-09-20:
#6894

xinsuinizhuan, 2024-09-23:
still error，please repaire it as:
void lv_calendar_gregorian_to_chinese(lv_calendar_date_t * gregorian, lv_calendar_chinese_t* chinese_calendar);

void lv_calendar_gregorian_to_chinese(lv_calendar_date_t * gregorian, lv_calendar_chinese_t* chinese_calendar)
{
    uint16_t year = gregorian->year;
    uint8_t month = gregorian->month;
    uint8_t day = gregorian->day;

    /*Record the number of days between the Spring Festival
    and the New Year's Day of that year.*/
    uint16_t by_spring;

    /*Record the number of days from the gregorian calendar
    to the New Year's Day of that year.*/
    uint16_t by_gregorian;

    /*Record the number of days in that month*/
    uint8_t days_per_month;

    /*Record from which month the calculation starts.*/
    uint8_t index;

    bool leep_month;
    lv_calendar_chinese_t chinese_calendar;

    if(year < 1901 || year > 2099) {
        chinese_calendar->leep_month = 0;
        chinese_calendar->today.year = 2000;
        chinese_calendar->today.month = 1;
        chinese_calendar->today.day = 1;
        return chinese_calendar;
    }

    if(((calendar_chinese_table[year - 1901] & 0x0060) >> 5) == 1)
        by_spring = (calendar_chinese_table[year - 1901] & 0x001F) - 1;
    else
        by_spring = (calendar_chinese_table[year - 1901] & 0x001F) - 1 + 31;

    by_gregorian = month_total_day[month - 1] + day - 1;

    if((!(year % 4)) && (month > 2))
        by_gregorian++;

    if(by_gregorian >= by_spring) {/*Gregorian calendar days after the Spring Festival*/
        by_gregorian -= by_spring;
        month = 1;
        index = 1;
        leep_month = false;

        if((calendar_chinese_table[year - 1901] & (0x80000 >> (index - 1))) == 0)
            days_per_month = 29;
        else
            days_per_month = 30;

        while(by_gregorian >= days_per_month) {
            by_gregorian -= days_per_month;
            index++;

            if(month == ((calendar_chinese_table[year - 1901] & 0xF00000) >> 20)) {
                leep_month = !leep_month;
                if(leep_month == false)
                    month++;
            }
            else
                month++;

            if((calendar_chinese_table[year - 1901] & (0x80000 >> (index - 1))) == 0)
                days_per_month = 29;
            else
                days_per_month = 30;
        }
        day = by_gregorian + 1;
    }
    else {/*Solar day before the Spring Festival*/
        by_spring -= by_gregorian;
        year--;
        month = 12;
        if(((calendar_chinese_table[year - 1901] & 0xF00000) >> 20) == 0)
            index = 12;
        else
            index = 13;
        leep_month = false;

        if((calendar_chinese_table[year - 1901] & (0x80000 >> (index - 1))) == 0)
            days_per_month = 29;
        else
            days_per_month = 30;

        while(by_spring > days_per_month) {
            by_spring -= days_per_month;
            index--;

            if(leep_month == false)
                month--;

            if(month == ((calendar_chinese_table[year - 1901] & 0xF00000) >> 20))
                leep_month = !leep_month;

            if((calendar_chinese_table[year - 1901] & (0x80000 >> (index - 1))) == 0)
                days_per_month = 29;
            else
                days_per_month = 30;
        }

        day = days_per_month - by_spring + 1;
    }
    chinese_calendar->today.day = day;
    chinese_calendar->today.month = month;
    chinese_calendar->today.year = year;
    chinese_calendar->leep_month = leep_month;
}

PR #6894 (2024-09-26): fix(calendar): fix lv_calendar_gregorian_to_chinese compile error
fix(calendar): fix lv_calendar_gregorian_to_chinese compile error

Fixes #6892

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://githu [... cut]
closed 2024-09-26 by liamHowatt