lvgl/lvgl #6896: when LV_USE_DRAW_SDL set,labels are in canvas and imgbtns didnot show

docterling, 2024-09-20:
### LVGL version

v9.2.0 and v9.3.0 dev

### What happened?

i use the lv_draw_line api implement a simple draw app that run on an arm64 soc with a touchpad,it's output res is 4k
when use the default sw render i can feel it some kind of slow,cause there is a mali gpu in the soc i want to use it to accelerate the drawing speed
i set the LV_USE_DRAW_SDL and LV_USE_SDL to 1,replace lv_display_create to lv_sdl_window_create,and change static uint8_t sdl_render_buf[2048 * 1024 * 4] to uint8_t sdl_render_buf[4096* 4096* 4] in file src/draw/sdl/lv_draw_sdl.c
in a thread i blend the sdl_render_buf to my drm picture plane's bo just like sw render i do,but it only show the backgroud picture and nothing else
1.sw render output
[image]
2.sdl output,there are no labels and two imgbtns,and when i touch the two imgbtn's position the event_cb called
[image]
what could be the problem?

docterling, 2024-09-24:
it probaboly like this
https://github.com/libsdl-org/SDL/issues/9438
so i change the init place,and the imgbtn show up,but anything else that in canvas didn't

kisvegabor, 2024-10-03:
First of all it's great the LVGL is being used on 4k displays! :rocket: 

---

`sdl_render_buf` is an internal buffer and it shouldn't be used like this. Can you draw to a normal `lv_canvas` manually and use SDL to blend that canvas automatically.

I've just tested it an this patch needs to be applied:
```diff
diff --git a/src/draw/sdl/lv_draw_sdl.c b/src/draw/sdl/lv_draw_sdl.c
index f06d79230..d9c63cfc4 100644
--- a/src/draw/sdl/lv_draw_sdl.c
+++ b/src/draw/sdl/lv_draw_sdl.c
@@ -167,6 +167,10 @@ static int32_t evaluate(lv_draw_unit_t * draw_unit, lv_draw_task_t * task)
 {
     LV_UNUSED(draw_unit);
 
+    /*If not refreshing the display probably it's a canvas rendering
+     *which his not support in SDL as it's not a texture.*/
+    if(lv_refr_get_disp_refreshing() == NULL) return 0;
+
     if(((lv_draw_dsc_base_t *)task->draw_dsc)->user_data == NULL) {
         task->preference_score = 0;
         task->preferred_draw_unit_id = DRAW_UNIT_ID_SDL;

```

docterling, 2024-10-08:
i've tested the patch,there's no effect,the labels in canvas didn't show
i malloc a g_ui_sdl_render_buf in app,and in draw_to_texture in lv_draw_sdl.c,after SDL_UpdateTexture
```
if(!getSdlBufFlag())
{
    memcpy(g_ui_sdl_render_buf,sdl_render_buf,texture_w * texture_h * 4);
    setSdlBufFlag(true);
}
```
now i blend the g_ui_sdl_render_buf to my drm picture plane's bo just like sw render i do
when i draw on the touchpad one time,i will occupy dozens of memory,and the line didn't show,finaly it will be killed by oom,why ? am i missing somthing?

kisvegabor, 2024-10-14:
Firs of all, you shouldn't edit LVGL itself. Instead you can create an `lv_canvas`, draw anything that you need there and let LVGL blend that canvas.

docterling, 2024-10-18:
so does the sdl_render_buf hold the lvgl finally blend result?
i just copy out the finally blend result and not to modify it,i don't think it would be a problem,right?

kisvegabor, 2024-10-19:
sdl_render_buf contains only a single primitive which was rendered (rectangle, border, text, etc) The final image is in an SDL texture Whis is not readable as a simple array.

docterling, 2024-10-19:
ok,i've knew the problem,in sw render,i malloc a buffer _**lvgl_disp_buf**_ and set it to lvgl disp driver via lv_disp_draw_buf_init api,in sw render mode,the _**lvgl_disp_buf**_ will contain the finally blend result,right? i use it to do drm display and it works fine.
but in sdl render i can't do it like sw render that i had done,in this mode i need to set LV_USE_LINUX_DRM to 1 to let sdl render do the display itself?

kisvegabor, 2024-10-28:
Correct. With SDL rendering we don't have access to the rendered image to read it. It's write only (by SDL).

closed 2024-11-26 by lvgl-bot