lvgl/lvgl #6907: LVGL 9.x.x significant increase in memory requirements for LV_COLOR_FORMAT_I4 canvas and image

egonbeermat, 2024-09-22:
### LVGL version

9.2.0

### What happened?

Trying to create a canvas of size 512x190 LV_COLOR_FORMAT_I4 with a pre-allocated data buffer causes the following error:

```
[Warn]  (0.000, +0)      lv_draw_buf_create_ex: No memory: 512x190, cf: 16, stride: 2048, 389120Byte,  lv_draw_buf.c:286
[Error] (0.000, +0)      decode_indexed: No memory for indexed image lv_bin_decoder.c:636
```
This works in my code base without issue LVGL 8.x.x. 

In LVGL 8.x.x, I have `LV_MEM_SIZE (85U * 1024U) ` but I think that is irrelevant as a factor for the canvas itself, as it used a pre-allocated buffer for data storage and it's expanded size to LV_COLOR_DEPTH 16 would be larger than that buffer, indicating to me that the canvas data buffer was converted and rendered to the LVGL flush buffer directly. In LVGL 9.x.x, it appears to be trying to allocate memory for an intermediate buffer sized at 389120 (4 bytes * 512 * 190) and failing, obviously. I have other canvas objects that are smaller but initially failed with the same error, but with `LV_MEM_SIZE (285U * 1024U) ` it was able to render, but that a) defeats the purpose of indexed color formats and b) isn't a feasible amount of memory on an MCU.

Using a 4 bit canvas was both efficient for both memory and processing, drawing a real-time chart that only requires 5 colors and can be drawn by directly setting the buffer memory and then flushed to the screen (by invalidating the canvas) when complete. Using the 4 bit indexed color canvas only required 48,640 bytes vs 194,560 bytes for the 16 bit color output, but under LVGL 9.x.x it seems to need 8x indexed buffer size to draw it!

This issue also occurs for images, can recreate with the code below if you create the image.

### How to reproduce?

//dsc_blank512x190 refers to a 512x90 image and convert using LV_COLOR_FORMAT_I4
LV_IMG_DECLARE(dsc_blank512x190);

lv_obj_t * image = lv_image_create(screen);
lv_image_set_src(image, &dsc_blank512x190);

egonbeermat, 2024-09-26:
@kisvegabor 
Can you spare a minute or two to share thoughts? It's a simple use case, but I've spent hours trying to understand, resolve or work around the issue but unfortunately I am not an expert on the innards of LVGL 9 to understand what is going on. I find it hard to imagine this is working as designed, given the amount of memory LVGL 9 grabs completely negates the purpose of using indexed canvases to save memory, so I must be doing something wrong!

I basically am creating an INDEXED_4BIT canvas in LVGL 8, adding my own buffer, manipulate the bytes in the buffer and watch LVGL render it to the screen in glorious full color. When I run it under LVGL 9, I get the memory errors above. If I increase LV_MEM_SIZE from 85*1024 that I had for v8, to 285*1024 for v9, then some of my smaller canvases will render. Any ideas? It is requesting 16x the memory used to store the 4 bit canvas, which seems to negate the point of using the 4 bit canvas if LV_MEM_SIZE has to be able to accomodate 4 * w * h bytes to render it!

egonbeermat, 2024-09-27:
Turns out I was doing something wrong. This setting in lv_conf.h was set to 1:

/*Decode bin images to RAM*/
#define LV_BIN_DECODER_RAM_LOAD 0

Slightly confusing comment and name, for it to also apply to the canvas object, even though under the hood it uses image. Maybe update the comment?

closed 2024-09-28 by egonbeermat

XuNeo, 2024-10-03:
The log saids `Legacy bin image detected` which means you are using lvgl v8 version of image. Please update to v9 format.

If the image is compressed, then `LV_BIN_DECODER_RAM_LOAD` should be enabled, in order to decompress the whole image to RAM, which is memory consuming.

egonbeermat, 2024-10-05:
Thanks for the replies. I did resolve this, once I realized that LV_BIN_DECODER_RAM_LOAD was the culprit, I just wasn't clear that I updated it to 0 :)

But this 

```
/Decode bin images to RAM/
#define LV_BIN_DECODER_RAM_LOAD 0
```

comment in lv_conf.h is misleading - it says decode **bin images** to RAM, but applies beyond that, to at least the canvas object. Perhaps update the comment, you can't expect a regular user to draw line from 'bin images' to 'canvas'

XuNeo, 2024-10-06:
The comment is indeed too short. Would you mind to open a PR to include your understanding?