lvgl/lvgl #6908: Function naming causing inconsistant naming in the MicroPython binding

kdschlosser, 2024-09-22:
### Introduce the problem

OK so in the MicroPython binding what it does is it looks at the first parameter of a function and if that first parameter is a pointer to a structure that function gets added as a method of the class that gets created for that structure.

Here is an example and I am fairly sure this is a more widespread issue then what is seen below.

we are going to use `lv_grad_dsc_t` as the example. 

In MicroPython that structure becomes the `grad_dsc_t` class. so in Python code that would look like this...

```python
class grad_dsc_t:

    def __init__(kwargs={}):
        for key, value in kwargs.items():
            setattr(self, key, value)
```

Here are the functions that deal with the gradients and are not guarded by the `LV_USE_DRAW_SW_COMPLEX_GRADIENTS` macro.
```
lv_gradient_color_calculate
lv_gradient_get
lv_gradient_cleanup
lv_gradient_init_stops
```

these functions are guarded by the `LV_USE_DRAW_SW_COMPLEX_GRADIENTS` macro.
```
lv_grad_linear_init

lv_grad_radial_init
lv_grad_radial_set_focal
lv_grad_conical_init

lv_gradient_linear_setup
lv_gradient_linear_cleanup

lv_gradient_linear_get_line

lv_gradient_radial_setup
lv_gradient_radial_cleanup

lv_gradient_radial_get_line

lv_gradient_conical_setup
lv_gradient_conical_cleanup

lv_gradient_conical_get_line
```

This is a little off topic but something I find strange. if the `LV_USE_DRAW_SW_COMPLEX_GRADIENTS` is not set then using gradients is not possible because there is no way to set even create a basic linear gradient. So why have functions that are not guarded by `LV_USE_DRAW_SW_COMPLEX_GRADIENTS`??? It doesn't make any sense why this is done. A "complex" gradient should only guard the radial and conical gradients and not the linear.

OK back on topic..
How the code generator works is it looks at that first parameters type name and removes from the start of the function anything that matches with the type name.

so this is what you end up with.

```python
class grad_dsc_t:

    def __init__(kwargs={}):
        for key, value in kwargs.items():
            setattr(self, key, value)

    def gradient_color_calculate(self, ...):
        ...

    def gradient_get(self, ...):
        ...

    def gradient_cleanup(self, ...):
        ...

    def gradient_init_stops(self, ...):
        ...

    def linear_init(self, ...):
        ...

    def radial_init(self, ...):
        ...

    def radial_set_focal(self, ...):
        ...

    def conical_init(self, ...):
        ...

    def gradient_linear_setup(self, ...):
        ...

    def gradient_linear_cleanup(self, ...):
        ...

    def gradient_linear_get_line(self, ...):
        ...

    def gradient_radial_setup(self, ...):
        ...

    def gradient_radial_cleanup(self, ...):
        ...

    def gradient_radial_get_line(self, ...):
        ...

    def gradient_conical_setup(self, ...):
        ...

    def gradient_conical_cleanup(self, ...):
        ...

    def gradient_conical_get_line(self, ...):
        ...
```

See the problem?

The whole purpose to the reboot of the API that was done for version 9.0 was to do away with the truncated naming. I think that the gradient portions of the code got missed because `grad` should have been changed to `gradient`. All functions, structures, unions, typedefs, enums and enumerations should have also been changed to align with this as well.

### Proposal

Correct the naming.

kdschlosser, 2024-09-22:
I don't quite understand the need for the `lv_gradient_color_calculate` function. This should be done using the `lv_gradient_init_stops` function as that is where all of the colors, opacities and fractions are supplied

in the `lv_grad_dsc_t` structure there is a field names `dir` and that field holds the type that the gradient is. 

```c
typedef enum {
    LV_GRAD_DIR_NONE,       /**< No gradient (the `grad_color` property is ignored)*/
    LV_GRAD_DIR_VER,        /**< Simple vertical (top to bottom) gradient*/
    LV_GRAD_DIR_HOR,        /**< Simple horizontal (left to right) gradient*/
    LV_GRAD_DIR_LINEAR,     /**< Linear gradient defined by start and end points. Can be at any angle.*/
    LV_GRAD_DIR_RADIAL,     /**< Radial gradient defined by start and end circles*/
    LV_GRAD_DIR_CONICAL,    /**< Conical gradient defined by center point, start and end angles*/
} lv_grad_dir_t;
```

because of this there is no need to have multiple functions with the exact same signature. 

identical signatures so these should be combined into a single function.
```
void lv_gradient_cleanup(lv_grad_t * grad);
void lv_gradient_linear_cleanup(lv_grad_dsc_t * dsc);
void lv_gradient_radial_cleanup(lv_grad_dsc_t * dsc);
void lv_gradient_conical_cleanup(lv_grad_dsc_t * dsc);
```

identical signatures so these should be combined into a single function.
```
void lv_gradient_linear_setup(lv_grad_dsc_t * dsc, const lv_area_t * coords);
void lv_gradient_radial_setup(lv_grad_dsc_t * dsc, const lv_area_t * coords);
void lv_gradient_conical_setup(lv_grad_dsc_t * dsc, const lv_area_t * coords);
```

identical signatures and should be combined into a single function. Either need to add a function to set the gradient type or add a parameter to the combined function to identify what the gradient type is that is being created.
```
void lv_grad_linear_init(lv_grad_dsc_t * dsc, int32_t from_x, int32_t from_y, int32_t to_x, int32_t to_y, lv_grad_extend_t extend);
void lv_grad_radial_init(lv_grad_dsc_t * dsc, int32_t center_x, int32_t center_y, int32_t to_x, int32_t to_y, lv_grad_extend_t extend);
void lv_grad_conical_init(lv_grad_dsc_t * dsc, int32_t center_x, int32_t center_y, int32_t start_angle, int32_t end_angle, lv_grad_extend_t extend);
'''

Identical signatures so these should be combine into a single function.
```
void /* LV_ATTRIBUTE_FAST_MEM */ lv_gradient_linear_get_line(lv_grad_dsc_t * dsc, int32_t xp, int32_t yp, int32_t width, lv_grad_t * result);
void /* LV_ATTRIBUTE_FAST_MEM */ lv_gradient_radial_get_line(lv_grad_dsc_t * dsc, int32_t xp, int32_t yp, int32_t width, lv_grad_t * result);
void /* LV_ATTRIBUTE_FAST_MEM */ lv_gradient_conical_get_line(lv_grad_dsc_t * dsc, int32_t xp, int32_t yp,  int32_t width, lv_grad_t * result);
```

Things like this bloat the API and also the running code. It has to make things a bit more complicated when dealing with gradients because of needing to test for the gradient type so you know what function to call. By changing the above and adding a function to get the gradient type and also add a function to get the stops and also a function to get the stop count you would also be able to place `lv_grad_dsc_t` into a private header.

kisvegabor, 2024-10-03:
> So why have functions that are not guarded by LV_USE_DRAW_SW_COMPLEX_GRADIENTS?

I thing it's the so called oversight. :slightly_smiling_face:  

```py
    def gradient_color_calculate(self, ...):
        ...

    def gradient_get(self, ...):
        ...
```

These should be internal. If they made their way to the public API than it's a mistake. 

> identical signatures so these should be combined into a single function.

We can do that for sure.

kdschlosser, 2024-10-03:
anything that is public and begins with `lv_grad_` should be renamed to `lv_gradient_`. That includes all structures, functions, typedefs and enums.

kisvegabor, 2024-10-09:
It's doable. We can add backward compatibility defines in `lv_api_map_v9.2.h` files.

kdschlosser, 2024-10-09:
simple?? HAHAHAHHAA Yeahhhhh OKKKKKK... NOT!

kdschlosser, 2024-10-09:
"grad" appears 2,733 times. can't just do a simple text replacement in the files... Gonna have to check each and every single one of them to see if it is something that needs to be changed or not.

kisvegabor, 2024-10-10:
On the API level I think grad is used only in less than 10 names. We should provide backward compatibility for those only and search and replace the rest.

kisvegabor, 2024-10-29:
Not stale

kisvegabor, 2024-11-20:
Not stale

kisvegabor, 2024-11-20:
cc @zjanosy?

kdschlosser, 2024-11-20:
well from a binding aspect it makes a complete mess of the API especially with the MicroPython binding which every single public function gets linked to.

The question becomes how much of a code size difference is there. I cannot imagine it being a whole lot. so the tradeoff is going to be possibly a couple hundred more bytes of flash use or a larger more confusing API.

The naming aspect if it is all over the place. there needs to be some kind of common ground with it.

zjanosy, 2024-11-20:
@kdschlosser 

As I wrote, since this is performance-critical code, on most modern microcontrollers it would be run from SRAM instead of flash. SRAM is scarce resource on microcontrollers, and even a couple hundred bytes makes a difference. Since dynamic linking is not supported on most architectures, we need to be careful with bloating the code with unused parts.

We cannot even rely on the cache, because LVGL already has way more code than what can fit into the cache. Refilling the cache is very slow on most cheap processors, which nowadays use SPI flash memory.

Certainly, we could add some more #defines to enable features with more granularity, but that's not very nice either.

kdschlosser, 2024-11-20:
It is stored in flash. that is where the code get loaded from and the code runs in ram. The code is not stored in ram. There are different sections of ram where the code is able to run from and depending on the ram there can be different speeds associated with the running code.

Now I do not believe that the type of SRAM used is going to make a sizeable difference in how fast the code is able to run. It will make a difference in how fast a function call occurs so you want to place functions that are called a lot into IRAM. The actual execution time of the code inside of the function I do not believe changes. If you have things allocated into PSRAM/SPIRAM that would cause the code to execute slower but if everything is done in internal memory I don't think there would be any difference in the code execution time.

kisvegabor, 2024-11-21:
`lv_gradient_linear/radial/conic_get_line` should be internal to the SW render so they shouldn't be visible by the binding. In light of that it doesn't really matter if it's 1 or 3 functions.

kdschlosser, 2024-11-22:
when you are dealing with  say the MicroPython binding this is what ends up happening.

These macros are not available in the binding.
```c
#define LV_GRAD_LEFT    LV_PCT(0)
#define LV_GRAD_RIGHT   LV_PCT(100)
#define LV_GRAD_TOP     LV_PCT(0)
#define LV_GRAD_BOTTOM  LV_PCT(100)
#define LV_GRAD_CENTER  LV_PCT(50)
```

and here is what the binding is going to create for the Python API

```python
class GRAD_DIR:
    NONE = 0
    VER = 1
    HOR = 2
    LINEAR = 3
    RADIAL = 4
    CONICAL = 5

class GRAD_EXTEND:
    PAD = 0
    REPEAT = 1
    REFLECT = 2

class grad_t:
    def gradient_cleanup(self):
        ...

class gradient_stop_t:
    
    @property
    def color(self) -> color_t:
        ...
    
    @color.setter
    def color(self, value: color_t) -> None:
        ...
        
    @property
    def opa(self) -> opa_t:
        ...
    
    @opa.setter
    def opa(self, value: opa_t) -> None:
        ...
    
    @property
    def frac(self) -> int:
        ...
    
    @frac.setter
    def frac(self, value: int) -> None:
        ...

class grad_dsc_t:

    @property
    def stops(self) -> list[gradient_stop_t]:
        ...
    
    @stops.setter
    def stops(self, value: list[gradient_stop_t]) -> None:
        ...
        
    @property
    def stops_count(self) -> int:
        ...
    
    @stops_count.setter
    def stops_count(self, value: int) -> None:
        ...
        
    @property
    def dir(self) -> int:
        ...
    
    @dir.setter
    def dir(self, value: int) -> None:
        ...

    @property
    def extend(self) -> int:
        ...
    
    @extend.setter
    def extend(self, value: int) -> None:
        ...

    class params: 
        class linear:
            @property
            def start(self) -> point_t:
                ...
            
            @start.setter
            def start(self, value: point_t) -> None:
                ...
                
            @property
            def end(self) -> point_t:
                ...
            
            @end.setter
            def end(self, value: point_t) -> None:
                ...
            
        class radial:
            @property
            def focal(self) -> point_t:
                ...
            
            @focal.setter
            def focal(self, value: point_t) -> None:
                ...
                
            @property
            def focal_extent(self) -> point_t:
                ...
            
            @focal_extent.setter
            def focal_extent(self, value: point_t) -> None:
                ...
                
            @property
            def end(self) -> point_t:
                ...
            
            @end.setter
            def end(self, value: point_t) -> None:
                ...
                
            @property
            def end_extent(self) -> point_t:
                ...
            
            @end_extent.setter
            def end_extent(self, value: point_t) -> None:
                ...
        
        class conical:
            
            @property
            def center(self) -> point_t:
                ...
            
            @center.setter
            def center(self, value: point_t) -> None:
                ...
                
            @property
            def start_angle(self) -> int:
                ...
            
            @start_angle.setter
            def start_angle(self, value: int) -> None:
                ...
                
            @property
            def end_angle(self) -> int:
                ...
            
            @end_angle.setter
            def end_angle(self, value: int) -> None:
                ...
                
    @property
    def state(self) -> Any:
        ...
            
    @state.setter
    def state(self, value: Any) -> None:
        ...
        
    def gradient_color_calculate(self, range: int, frac: int, color_out: grad_color_t, opa_out: opa_t) => None:
        ...
        
    def gradient_get(self, w: int, h: int) -> grad_t:
        ...
        
    def gradient_init_stops(self, colors: list[color_t], opa: list[opa_t], fracs: list[int], num_stops: int) -> None:
        ...
        
    def linear_init(self, from_x: int, from_y: int, to_x: int, to_y: int, extend: int) -> None:
        ...
                         
    def radial_init(self, center_x: int, center_y: int, to_x: int, to_y: int, extend: int) -> None:
        ...                   
                             
    def radial_set_focal(self, center_x: int, center_y: int, radius: int) -> None:
        ...
    
    def conical_init(self, center_x: int, center_y: int, start_angle: int, end_angle: int, extend: int) -> None:
        ...
                                                
    def gradient_linear_setup(self, coords: area_t) -> None:
        ...
                                                    
    def gradient_linear_cleanup(self) -> None:
        ...
    
    def gradient_linear_get_line(self, xp: int, yp: int, width: int, result: grad_t) -> None:
        ...
                                
    def gradient_radial_setup(self, coords: area_t) -> None:
        ...
    
    def gradient_radial_cleanup(self) -> None:
        ...
    
    def gradient_radial_get_line(self, xp: int, yp: int, width: int, result: grad_t) -> None:
        ...
    
    def gradient_conical_setup(self, coords: area_t) -> None:
        ...
    
    def gradient_conical_cleanup(self) -> None
        ...
    
    def gradient_conical_get_line(self, xp: int, yp: int, width: int, result: grad_t) -> None
        ...

```

That whole API isn't right.

kdschlosser, 2024-11-22:
@zjanosy 

Just to let you know that whole thing about the code being omitted when not in use. The ONLY functions that applies to are `lv_grad_linear_init`, `lv_grad_radial_init`, `lv_grad_conical_init`, `lv_grad_radial_set_focal` and `lv_gradient_init_stops`. Totals a whipping 56 lines of code. 

see the following locations

https://github.com/lvgl/lvgl/blob/bd2927736acbe24b89ce63fa996851e8e69c0d49/src/draw/sw/lv_draw_sw_fill.c#L170

https://github.com/lvgl/lvgl/blob/bd2927736acbe24b89ce63fa996851e8e69c0d49/src/draw/sw/lv_draw_sw_fill.c#L211

https://github.com/lvgl/lvgl/blob/bd2927736acbe24b89ce63fa996851e8e69c0d49/src/draw/sw/lv_draw_sw_fill.c#L290

You are basically doing what I suggested to do. As it turns out combining the functions is not only going to reduce the code footprint it is also going to make it less complex and easier to maintain. The compiled binary will be smaller as a result of combining like functions. 

You can also improve the performance by a decent amount of you store the coordinates that are passed to the setup function and also store the initial values of the state so they do not change. This would allow you to check to see if the passed in coordinates have changed and if they have then run the setup code again and if not set the state back to the original setup values and exit. This would come at the cost of a small amount of additional heap use but it would mean that complex math would not need to be done over and over again if an object has not moved at all. 

This is the amount of additional bytes per gradient that would be allocated on the heap. 

60 additional bytes for a radial gradient
28 additional bytes for a linear gradient 
36 additional bytes for a conical gradient

for the radial gradient the code below is what gets run in just the setup function each and every single time the object gets redrawn even if the objects coordinates did not change. I commented above the functions the number of time that specific function gets called each time the setup function is run. 

<details>

<summary>Click to see code</summary>

```c
// this is called 2 times
int32_t LV_ATTRIBUTE_FAST_MEM lv_sqrt32(uint32_t x)
{
    static const unsigned char sqq_table[] = {
        0,  16,  22,  27,  32,  35,  39,  42,  45,  48,  50,  53,  55,  57,
        59,  61,  64,  65,  67,  69,  71,  73,  75,  76,  78,  80,  81,  83,
        84,  86,  87,  89,  90,  91,  93,  94,  96,  97,  98,  99, 101, 102,
        103, 104, 106, 107, 108, 109, 110, 112, 113, 114, 115, 116, 117, 118,
        119, 120, 121, 122, 123, 124, 125, 126, 128, 128, 129, 130, 131, 132,
        133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 144, 145,
        146, 147, 148, 149, 150, 150, 151, 152, 153, 154, 155, 155, 156, 157,
        158, 159, 160, 160, 161, 162, 163, 163, 164, 165, 166, 167, 167, 168,
        169, 170, 170, 171, 172, 173, 173, 174, 175, 176, 176, 177, 178, 178,
        179, 180, 181, 181, 182, 183, 183, 184, 185, 185, 186, 187, 187, 188,
        189, 189, 190, 191, 192, 192, 193, 193, 194, 195, 195, 196, 197, 197,
        198, 199, 199, 200, 201, 201, 202, 203, 203, 204, 204, 205, 206, 206,
        207, 208, 208, 209, 209, 210, 211, 211, 212, 212, 213, 214, 214, 215,
        215, 216, 217, 217, 218, 218, 219, 219, 220, 221, 221, 222, 222, 223,
        224, 224, 225, 225, 226, 226, 227, 227, 228, 229, 229, 230, 230, 231,
        231, 232, 232, 233, 234, 234, 235, 235, 236, 236, 237, 237, 238, 238,
        239, 240, 240, 241, 241, 242, 242, 243, 243, 244, 244, 245, 245, 246,
        246, 247, 247, 248, 248, 249, 249, 250, 250, 251, 251, 252, 252, 253,
        253, 254, 254, 255
    };

    int32_t xn;

    if(x >= 0x10000)
        if(x >= 0x1000000)
            if(x >= 0x10000000)
                if(x >= 0x40000000) {
                    if(x >= 65535UL * 65535UL)
                        return 65535;
                    xn = sqq_table[x >> 24] << 8;
                }
                else
                    xn = sqq_table[x >> 22] << 7;
            else if(x >= 0x4000000)
                xn = sqq_table[x >> 20] << 6;
            else
                xn = sqq_table[x >> 18] << 5;
[... 284 lines cut ...]
        state->bpx = dx << 1;
        state->bpy = dy << 1;
        state->bc = (state->r0 * dr - state->x0 * dx - state->y0 * dy) << 1;
    }
    state->inv_a4 = state->a4 != 0 ? (1 << (13 + 16)) / state->a4 : 0;
    /* check for possible clipping */
    if(dsc->extend == LV_GRAD_EXTEND_PAD &&
       /* if extend mode is 'pad', then we can clip to the end circle's bounding box, if the start circle is entirely within the end circle */
       (lv_sqr(start.x - end.x) + lv_sqr(start.y - end.y) < lv_sqr(r_end - r_start))) {
        if(r_end > r_start) {
            lv_area_set(&state->clip_area, end.x - r_end, end.y - r_end, end.x  + r_end, end.y + r_end);
        }
        else {
            lv_area_set(&state->clip_area, start.x - r_start, start.y - r_start, start.x  + r_start, start.y + r_start);
        }
    }
    else {
        state->clip_area.x1 = -0x7fffffff;
    }
}
```

</details>

I am thinking that it would be worth the additional heap use to keep from having to run that code over and over again when it really doesn't need to. copying the data to reset the state would be far faster than having to do all of that math.

closed 2024-12-18 by lvgl-bot