lvgl/lvgl #6909: Remove BOM

Narukara, 2024-09-23:
### Introduce the problem

I encountered some issues while building the documentation. And I found that this was due to some files in the repository containing BOM. Maybe we should remove them. Here is a list:

```shell
$ git --no-pager grep --perl-regexp -I -l -e '^\xEF\xBB\xBF'
.devcontainer/__lv_conf.h__
demos/render/lv_demo_render.c
demos/render/lv_demo_render.h
env_support/cmsis-pack/lv_conf_cmsis.h
examples/styles/lv_example_style.h
examples/styles/lv_example_style_16.c
examples/styles/lv_example_style_17.c
examples/styles/lv_example_style_18.c
lv_conf_template.h
src/draw/sw/blend/lv_draw_sw_blend.c
src/draw/sw/blend/lv_draw_sw_blend_to_al88.c
src/draw/sw/blend/lv_draw_sw_blend_to_al88.h
src/draw/sw/blend/lv_draw_sw_blend_to_argb8888.c
src/draw/sw/blend/lv_draw_sw_blend_to_l8.c
src/draw/sw/blend/lv_draw_sw_blend_to_l8.h
src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c
src/draw/sw/blend/lv_draw_sw_blend_to_rgb888.c
src/draw/sw/lv_draw_sw.h
src/draw/sw/lv_draw_sw_fill.c
src/draw/sw/lv_draw_sw_gradient.c
src/draw/sw/lv_draw_sw_gradient.h
src/draw/sw/lv_draw_sw_img.c
src/draw/sw/lv_draw_sw_transform.c
src/drivers/windows/lv_windows_context.c
src/drivers/windows/lv_windows_context.h
src/drivers/windows/lv_windows_input.c
src/libs/thorvg/tvgSwRasterAvx.h
src/misc/lv_color.c
src/misc/lv_color.h
src/misc/lv_math.c
src/misc/lv_math.h
src/misc/lv_style.h
src/osal/lv_windows.c
src/osal/lv_windows.h
src/stdlib/builtin/lv_tlsf.h
src/stdlib/lv_mem.h
src/widgets/canvas/lv_canvas.c
tests/unity/unity_support.c
```

### Proposal

This problem has occurred before (#4684, #4685). Perhaps we can configure CI to automatically check for this to avoid having to deal with it repeatedly.

kisvegabor, 2024-09-26:
Thank you for reporting it. 

 I've opened a PR to fix the current issues however a CI to automatically remove the BOM would be great for sure. 

See #6942

Narukara, 2024-09-27:
I found this action and it seems suitable:

https://github.com/marketplace/actions/bom-check

Can I submit a PR with this? (to avoid CI failures, after #6942 is merged)

vwheeler63, 2024-09-27:
https://github.com/marketplace/actions/  is a little "gold mine"....

Out of curiosity, @Narukara , what kind of errors was it causing for the docs generation?  And what platform are you running it on?

Narukara, 2024-09-27:
> Out of curiosity, @Narukara , what kind of errors was it causing for the docs generation? And what platform are you running it on?

I was trying to run `docs/build.py` on **Windows** and got this:

```shell
PS C:\Users\naruk\workshop\LVGL\lvgl> python docs/build.py skip_latex
C:\Users\naruk\workshop\LVGL\lvgl\docs\build.py:83: SyntaxWarning: invalid escape sequence '\*'
  br = re.sub('\* ', '', br)

****************
Building
****************
Traceback (most recent call last):
  File "C:\Users\naruk\workshop\LVGL\lvgl\docs\build.py", line 122, in <module>
    config_builder.run()
  File "C:\Users\naruk\workshop\LVGL\lvgl\docs\config_builder.py", line 20, in run
    data = f.read()
           ^^^^^^^^
UnicodeDecodeError: 'gbk' codec can't decode byte 0xbf in position 2: illegal multibyte sequence
```

Here, `config_builder.py` tries to read `lv_conf_template.h`:

https://github.com/lvgl/lvgl/blob/0daebca18b2e7d4c51d5b53e7f2a60df634850f2/docs/config_builder.py#L19-L20

However, `open()` does not guarantee to use UTF-8 encoding by default. In fact, on my Chinese version of Windows, it uses GBK encoding by default. And `lv_conf_template.h` happens to have a UTF-8 BOM at the beginning, which leads to the above error.

The interesting thing is, `lv_conf_template.h` actually only contains ASCII characters, so it shouldn't matter which encoding is used. But the presence of the UTF-8 BOM breaks this compatibility.

vwheeler63, 2024-09-27:
> The interesting thing is, `lv_conf_template.h` actually only contains ASCII characters, so it shouldn't matter which encoding is used. But the presence of the UTF-8 BOM breaks this compatibility.

cc:  @kisvegabor 
cc:  @kdschlosser 

Ahhh!  Actually, there is an alternate handling that might be more elegant (since some people's editors from here on into the future, are going to keep putting that BOM in) — the Python f.read() command has a simple modifier to get it to use a UTF-8 input translator before the content arrives into the `data` variable.  I've seen Kevin use it and it seems to me like a no-brainer (obvious good choice) for the docs generation logic.

@kdschlosser @kisvegabor   Thoughts?

Kind regards,
Vic

kdschlosser, 2024-09-27:
The problems exist because of using mixed character sets.. For some reason characters are starting to be reintroduced into the documentation build that should not be in there. 

I am also not sure as to why your environment is using the 'gbk' codec. Windows uses the 'cp1252' coded so you have some kind of a special environment that you are using to build on. 

One thing that pops out is you are running in PowerShell. The build system has not been tested inside of power shell. Try using a `cmd` shell and see what happens.

kdschlosser, 2024-09-27:
People need to change their editors so it doesn't add any BOM's. There should be a check that will cause the build to fail and the person submitting the PR's that include BOM's should be responsible for changing their development environment so it doesn't add BOM's

Narukara, 2024-09-28:
> I am also not sure as to why your environment is using the 'gbk' codec. Windows uses the 'cp1252' coded so you have some kind of a special environment that you are using to build on.

🤔 I don't think I have some kind of special environment. The 'cp1252' seems to be the default encoding [only in some regions](https://en.m.wikipedia.org/wiki/Windows-1252). On the Chinese version of Windows, the default encoding seems to be GBK.

> One thing that pops out is you are running in PowerShell. The build system has not been tested inside of power shell. Try using a `cmd` shell and see what happens.

Thanks for your advice. I guess WSL would be better. I'm trying it out.

> People need to change their editors so it doesn't add any BOM's. There should be a check that will cause the build to fail and the person submitting the PR's that include BOM's should be responsible for changing their development environment so it doesn't add BOM's

As I mentioned above, [this action](https://github.com/marketplace/actions/bom-check) seems suitable.

kdschlosser, 2024-09-28:
I 100% agree about testing for BOM.... That being said the BOM should not be removed automatically. That should be the responsibility of the person that is submitting the PR. 

I am not sure why you are having an issue. I am able to open up the file it is having a problem with using the gbk encoder and it doesn't error at all. Is your lv_conf_template.h file modified in some way?

AJingshou, 2024-09-28:
I encountered the same issue. In Windows 11 Home Chinese version, the `lv_conf_template.h` file is displayed as `UTF-8 with BOM` encoding in Visual Studio Code.
[image]

kdschlosser, 2024-09-28:
commit #6793 is what added the BOM's

kdschlosser, 2024-09-28:
The date of the modification to the `lvgl_conf_template.h` on my system is 9/22/2024 and it doesn't have the BOM. the last time that file was touched was 4 days ago on the 24th.

Narukara, 2024-09-28:
> I am not sure why you are having an issue. I am able to open up the file it is having a problem with using the gbk encoder and it doesn't error at all. Is your lv_conf_template.h file modified in some way?

No. I asked my friend @AJingshou to try it and reproduce the error.

> The date of the modification to the `lvgl_conf_template.h` on my system is 9/22/2024 and it doesn't have the BOM. the last time that file was touched was 4 days ago on the 24th.

So I guess there should be no doubt now?

kdschlosser, 2024-09-28:
I can tell you that there is no BOM in the version I have of  the template file on my system, and that file is dated 2 days before that PR was merged. I have to download that file from the repo and see if there is a BOM in it. If there is then there are going to be several files that need to be fixed.

kdschlosser, 2024-09-28:
I was wrong the BOM is there back further. My bad..

kdschlosser, 2024-09-30:
It hasn't been an issue until now because someone has their computer set up with an encoding that doesn't play nice nice with the BOM. It's probably because the engine uses a form of UTF-16 and the BOM is in UTF-8

It is not a far stretch to have a requirement of the files not having a BOM. Most IDE's don't add a BOM by default it has to be turned on. Why someone did that was possibly for another project and they simply forgot to turn it off.. IDK when it ended up getting added...

Just to let you know there are other sections of the documentation build system that reads files as well as other scripts that do as well. You would have to go and make changes every single place and not just the one location.

kdschlosser, 2024-09-30:
Yes we can change the encoding to UTF-8 but then what happens is someone's IDE adds a UTF16 BOM. we are right back to square 1.

Also if I change it to UTF-8 encoding and read the file as binary data I would either have to change every single location that is dealing with the data so it deals with it in bytes instead of as a string. I won't be able to decode the data into a string because it will end up throwing the same error. 

Header files are not supposed to contain any binary data which is what a BOM is.

vwheeler63, 2024-10-01:
Just a thought...  and for sake of transparency, I am biased towards ASCII because my favorite (old) editor behaves strangely on a few things when the BOMs are present.  That said, in the `.py` files that generate the docs, 7 out of 10 occurrences of text file reads already look like this:
```
data = f.read().decode('utf-8')
```

The remaining 3 that still look like this:
```
data = f.read()
```

are for:

- adding the Chinese translation link step (in `add_translation.py`),
- the step that creates a temporary `lv_conf.h` file for the docs generation to use (reads `lv_conf_template.h`) (in `config_builder.py`), and
- the process-index step of generating the examples (in `example_list.py`).

Might it be worth a try for @Narukara to try changing those 3 to see if that solves the problem he's running into?

kdschlosser, 2024-10-01:
The problem is there is something there that shouldn't be there. fixing the issue by doing something other than removing the things that are not supposed to be there is just going to band-aide the problem. we all know what happens with band-aides right? They eventually fall off exposing the problem again. 

The problem is there is something there that is not supposed to be there and the symptom is the script throwing an error because of it. trying to fix it by stopping the symptom doesn't fix the problem.

Narukara, 2024-10-01:
> @Narukara, was the problem you ran into generated from one of these? If so, might it be worth a try to change those 3 temporarily to see if that solves the problem you are running into?

Yes. It is the second one. Changing that line of code to `data = f.read().decode('utf-8')` and **opening the file in binary mode** does solve the problem.

By the way, I think `open(filename, 'r', encoding='utf-8')` is more elegant.

kdschlosser, 2024-10-01:
adding the encoding symptom fix only handles UTF-8 BOM's as well. what happens if UTF16 BOM's get introduced? Now you have to add additional code to handle both UTF-8 BOM's and UTF-16 BOMS. and it get a tad bit more complex to filter out the UTF-16 BOM's.

kdschlosser, 2024-10-01:
It is not normal to have BOM's added to header and source files. This was a persons incorrectly configured IDE or editor that did this. The problem is their IDE and not anything in the project. You should not have to code around someone incorrectly configured IDE.

kisvegabor, 2024-10-02:
Okay, let's
1. Fix the `f.read()`s just to make them unified
2. Remove the current BOMs #6942
3. Add CI to check BOMs. The best would be to 
   - Show if there is an error
   - Make it commit the fixes when triggered by e.g. a special comment (such as "ci: fix BOM").

Do you agree?

kdschlosser, 2024-10-02:
However you guys want to go about it. I am just wanting you guys to look at it from all angles and to fix the problem and not to fix a symptom.

Also if changes are made to the read to use encoding you also need to make sure to change the writes too. where something is read in utf-8 it also needs to be written in utf-8.

The reason why I use the long form

```python
with open('some.file', 'rb') as f:
    data = f.read().decode('utf-8')
```

instead of doing this...

```
with open('some.file', 'r', encoding='utf-8') as f:
    data = f.read()
```

They do the exact same thing in the end except the latter has a higher probably of having issues.

It has long been known that the developers of Python favor nix OS's over others. They make changes without consideration of the impact that will happen on other OS's and they are not quick to correct those issues when they occur either. encoding has been one of those areas that python isn't the most graceful at handling. A fantastic example of this is using `sys.set_encoding` which used to be able to be used to set the python environment to use a specific encoding globally inside of python. Well that's what it used to do in Python 2.x. in Python 3 this doesn't work when opening files. The behavior has changed.

an example of a difference is os.set_blocking which is used to set if a read/write for a file is blocking or not. This is available for nix OS's but not for Windows. This causes a huge problem when wanting to read data from the stdout of a subprocess. 

This is the reason why I use the long form is I know it works and it works across all OS's and across all Python 3.x versions. 
You can also see exactly what is taking place... I am opening the file for reading and it is being opened in binary mode I am reading the data as bytes and then decoding that byte data into a string. You can see what is being done from the code. With the encoding being supplied to the open command... It's a big ? as to what it is doing... shorter is not always better...

this can also be done as well..

```python
data = open('some.file', 'r', encoding='utf-8').read()
```

yes it will work.. yes it is shorter. should it be done? No. because the file never gets closed. Does it cause any problems, yes it could. If the file gets opened anywhere else in the code it will fail due to the file never being closed. The error when this happens will leave you scratching your head.  

@vwheeler63 just ran across problems with file encoding in a different repo. Works on one OS but not on another. 

The way you have seen me do it is because of experience not because of lack of knowledge of an easier way. Trust me, I have had lots of hair pulling moments dealing with this stuff. PLUS... It is also good coding to spell things out and make them easier to read and to understand.

PR #6961 (2024-12-03): chore: removes BOM
Fixes #6909
closed 2024-10-26 by lvgl-bot