lvgl/lvgl #6912: [LVGL 9.2] Incorrect hover handling

zorro59, 2024-09-23:
### LVGL version

9.2

### What happened?

I have 2 imagebuttons, placed next to eachother, 0 distance between them, the first one is disabled, the second one in default state. Both can be hovered. 
When hovering over the default state button moving the cursor  to the right (outside all the buttons area), the hover is processed corectly. Moving the cursor to the left (over the disabled button) will send only the hover enter event, the hover leave event will not be sent to the previously hovered button.

zorro59, 2024-09-23:
Proposed fix in lv_indev.c
`        lv_obj_t ** last = &indev->pointer.last_hovered;
        lv_obj_t * hovered = pointer_search_obj(lv_display_get_default(), &indev->pointer.act_point);
        printf("last %lX hovered %lX\n", *last, hovered);
        if(*last != hovered) {
            printf("sending leave to %lX\n", *last);
            lv_obj_send_event(*last, LV_EVENT_HOVER_LEAVE, indev);
            if(indev_reset_check(indev)) return;
            lv_indev_send_event(indev, LV_EVENT_HOVER_LEAVE, *last);
            if(indev_reset_check(indev)) return;

            printf("sending enter to %lX\n", hovered);
            lv_obj_send_event(hovered, LV_EVENT_HOVER_OVER, indev);
            if(indev_reset_check(indev)) return;
            lv_indev_send_event(indev, LV_EVENT_HOVER_OVER, hovered);
            if(indev_reset_check(indev)) return;
            *last = hovered;
        }`
It seems that because the enter event was first sent to the button, the event was somehow called recursively and set the *last pointer was set to NULL, thus when the leave event triggered the original target was lost.

liamHowatt, 2024-10-01:
Hi, thanks! Please feel free to open a PR with your fix and also send code to reproduce the issue (two imagebuttons side-by-side, one disabled, etc.).

liamHowatt, 2024-10-01:
Please include a link to this issue in the PR description.

kissa96, 2024-10-29:
Hi @zorro59 can you please provide some code snippet, so I can reproduce the issue?

closed 2024-11-25 by lvgl-bot