lvgl/lvgl #6913: Layer memory allocation failures during lvgl 9.2.0 Benchmark

nicusorcitu, 2024-09-23:
### LVGL version

9.2.0

### What happened?

[Warn]  (187.545, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.557, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.567, +10)   lv_draw_buf_create_ex: No memory: 270x45, cf: 18, stride: 576, 25920Byte,  lv_draw_buf.c:296

[Warn]  (187.578, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.588, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.600, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.610, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.621, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.631, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.643, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.653, +10)   lv_draw_buf_create_ex: No memory: 270x45, cf: 18, stride: 576, 25920Byte,  lv_draw_buf.c:296

[Warn]  (187.664, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.674, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.686, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.696, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.707, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.717, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.729, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.739, +10)   lv_draw_buf_create_ex: No memory: 270x45, cf: 18, stride: 576, 25920Byte,  lv_draw_buf.c:296

[Warn]  (187.750, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.760, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.772, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.782, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.793, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.803, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.815, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.825, +10)   lv_draw_buf_create_ex: No memory: 270x45, cf: 18, stride: 576, 25920Byte,  lv_draw_buf.c:296

[Warn]  (187.836, +11)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.846, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

[Warn]  (187.858, +12)   lv_draw_layer_alloc_buf: Allocating layer buffer failed. Try later lv_draw.c:408

[Warn]  (187.868, +10)   lv_draw_buf_create_ex: No memory: 270x22, cf: 16, stride: 1088, 23936Byte,  lv_draw_buf.c:296

### How to reproduce?

Setup different than default from lv_conf_template.h from 9.2.0 branch:
```
#define LV_COLOR_DEPTH 16
#define LV_MEM_SIZE (128 * 1024U)          /*[bytes]*/
#define LV_DRAW_SW_DRAW_UNIT_CNT    1
#define LV_DRAW_BUF_STRIDE_ALIGN 64
#define LV_DRAW_BUF_ALIGN 64
#define LV_USE_LOG 1
#define LV_ATTRIBUTE_MEM_ALIGN_SIZE 64
#define LV_ATTRIBUTE_MEM_ALIGN __attribute__((aligned(LV_ATTRIBUTE_MEM_ALIGN_SIZE)))
#define LV_FONT_MONTSERRAT_24 1
#define LV_USE_DEMO_WIDGETS 1
#define LV_USE_DEMO_BENCHMARK 1
```

Running the benchmark application will hang on test "Containers with opa_layer".

nicusorcitu, 2024-09-23:
@cosmindanielradu19

kisvegabor, 2024-09-25:
The problem is the LVGL tries to allocate a lot of small layers a fully fill up the memory. If it fits to well there won't be place for any other small allocations. 

 We should really bring back the `LV_DRAW_LAYER_MAX_MEMORY` define. It should prevent allocating too much layer memory [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw.c#L451). I can add it next week, but if you have time earlier a PR would be very welcome. `lv_draw_layer_alloc_buf` can just return `NULL` if `_draw_info.used_memory_for_layers_kb + new_layer_size > LV_DRAW_LAYER_MAX_MEMORY`.

kisvegabor, 2024-09-26:
Alternatively, you can use a custom simple allocator in `lv_draw_buf_create`. By using a separate memory pool the normal LVGL can be cleaner. 

However if we limit the max layer memory usage e,g. the 48kB the fragmentation will be lower too.

kisvegabor, 2024-10-29:
No stale.

nicusorcitu, 2024-11-20:
Which PR?

nicusorcitu, 2024-11-20:
@cristian-stoica

closed 2024-12-16 by kisvegabor