lvgl/lvgl #6915: Potential API breaks?

kisvegabor, 2024-09-23:
### Introduce the problem

We are developing LVGL's UI editor and in order to generate code in an elegant way it's important to have a consistent API in LVGL. 

### Proposal

For most of the things I can just create new functions and retire the old ones later, but there are two cases where slightly changing the functions' signature would work better. 

API break in a minor version would be very odd so I'd like to discuss these in the hope that you will have good idea/workaround. So 
- `lv_slider_set_value` now has an `anim_en` parameter too. Having a `set_value` and an `set_value_anim` function would be cleaner. An alternative could be to add `set_value_now` and `set_value_anim` as wrappers around set_value
- `lv_roller_set_options` has an infinite/ non infinite parameter too. There should be a separate `set_mode` function instead. Here the workaround can be to have a `set_option_normal` and `set_options_infinite`. 

I'm still exploring how to add custom code to process non standard cases by a hand written parser but in the meantime it'd be great to hear your thoughts on these. 

Cc @XuNeo

XuNeo, 2024-09-24:
Can we separate those properties? 
For slider, `value` and `anim`: `enable` or `disable`.
For roller:, `mode`: `normal` or `infinite` and `options`.

kisvegabor, 2024-09-25:
I've just found a way to describe it in XML. I'll share my concept about XML next week, but in short it should support multiple parameters for a property. E.g.
```xml
<roller options="'a\nb\nc' infinite" />
```
The same could work for slider:
```xml
<slider value="30 anim_on"/>
```

This way we can leave LVGL's API as it is. I realized that LVGL's API is
- mature and battle-tested
- shouldn't be broken

So we should align the XML the C API and not the opposite.

XuNeo, 2024-09-25:
XML is not for human to write, but for automation and parser.
I have to say I cannot agree with this approach. The whole idea of property API is to make it easy for this scenario.

`<roller options="'a\nb\nc' infinite" />` This is absolutely two properties: `mode` and `options`.

> This way we can leave LVGL's API as it is. I realized that LVGL's API is
> 
> mature and battle-tested
> shouldn't be broken

It shouldn't conflict with each other. We should keep two APIs both available.
Instead of current `roller_set_options`. We should also add `set_mode`.

W-Mai, 2024-09-25:
```html
<roller options="'a'\n'b'\n" mode="infinite" />
<slider value="20" min="0" max="100" with-animation="on" />
```

I think this way will be more clearer.

And it will be translated to like

```cpp
lv_roller_set_options(xxxx);
lv_roller_set_mode(xxxx);

lv_anim_enable_t with_animation = lv_xxxx_get_params();
lv_slider_set_xxxx(xxxxx);
lv_slider_set_xxxx(xxxxx);
lv_slider_set_value(xxxx, with_animation);
```

kisvegabor, 2024-09-25:
> XML is not for human to write, but for automation and parser.

Could you elaborate on that? Why do you think that? What can be a good alternative in your opinion?

---
In case of slider you should be able to decide case by case how to set the value (e.g. no animation when the slider is created, and with animation  when the value was changed externally). If you want to disable animation in general you can do it with a style. So the modified API could be:
```c
lv_slider_set_value(slider, 25);
lv_slider_set_value_with_anim(slider, 25);
```

In case of the roller I agree that 
```c
lv_roller_set_options(roller "a\nb\nc");
lv_roller_set_mode(roller, LV_ROLLER_MODE_INFINITE);
```
would be better.

However, both are API breaks. I'm not sure how shall we we handle this kind of API breaks, when the function is not removed, just its signature changes.

XuNeo, 2024-09-26:
> Could you elaborate on that? Why do you think that? What can be a good alternative in your opinion?

I believe the XML file generated by the editor serves a different purpose compared to the C API, which is designed for developers. The API should be user-friendly, like the current `lv_slider_set_value(obj, value, anim)`, which is straightforward and easy to use.

On the other hand, the XML file should follow a simple, consistent rule to pass parameters from the UI editor to the device, allowing for easy parsing and seamless re-creation of the UI. There should be no special cases, so a common parser can handle it universally.  I imagine the unified property API can serve for this purpose.

> What can be a good alternative in your opinion?

I don't have experience in defining such protocols, I’d like to refer to existing standards like HTML for guidance. It would also be helpful to consult with the community on this. Let’s reach out to the folks in [this discussion](https://github.com/lvgl/lvgl/issues/2428) to gather input and insights.

> However, both are API breaks. I'm not sure how shall we we handle this kind of API breaks, when the function is not removed, just its signature changes.

Let's establish a clear process for introducing breaking changes. In the NuttX community, for example, they tag PRs with a "breaking change" label and ensure that the release notes and documentation are prepared accordingly. The frustration users are experiencing stems from losing functionality without guidance on how to implement the changes with the new API, and no documentation detailing the break.

For this particular `slider` case, a non-breaking solution is to add property `animation_enable` and `value`. No API change.

```
static void _lv_slider_set_value(obj, value);
static void _lv_slider_set_anim_enable(obj, enable);

static const lv_property_ops_t properties[] = {
    {
        .id = LV_PROPERTY_SLIDER_VALUE,
        .setter = _lv_slider_set_value,
        .getter = lv_label_get_value,
    },
    {
        .id = LV_PROPERTY_SLIDER_ANIM,
        .setter = _lv_slider_set_anim_enable,
        .getter = _lv_slider_get_anim_enable,
    },
```

kisvegabor, 2024-09-30:
> There should be no special cases, so a common parser can handle it universally. I imagine the unified property API can serve for this purpose.

But there are special cases :slightly_frowning_face: I can't see how can you manage `span`s, `chart_series`, target the list of the drop-down with styles,  add a new tab to a tabview, set the current year + month + day on a calendar, pass a string array to button matrix as a map, select a cell row and column to set its value, add a style to a slider's indicator in pressed state, etc. These were some of the complex cases I've encountered with recently and found that a simple one `set(prop_id`, value) couldn't work well.  

> I don't have experience in defining such protocols, I’d like to refer to existing standards like HTML for guidance. It would also be helpful to consult with the community on this. Let’s reach out to the folks in https://github.com/lvgl/lvgl/issues/2428 to gather input and insights.

I also took HTML as a reference and I feel confident about XML mainly because of HTML too. It's battle tested, shows that it can be effectively used to describe small and large UIs, and people are familiar with it. 

I'm planning to share my XML design description today. 

> Let's establish a clear process for introducing breaking changes. In the NuttX community, for example, they tag PRs with a "breaking change" label and ensure that the release notes and documentation [...]

:+1: