lvgl/lvgl #6918: When I compiled lvgl_demos, the memory quickly reached 27G and the computer crashed

xinsuinizhuan, 2024-09-24:
### LVGL version

9.3

### What happened?

newest code,compile in widows with cmake and vs2019, compile the lvgl first, then when compile the  lvgl_demos,the memory quickly reached 27G and the computer crashed

xinsuinizhuan, 2024-09-25:
I find the error, when compile the svg tiger.c file,the memery will quickly reached 27G

liamHowatt, 2024-09-27:
The Visual Studio compiler uses over 27 GB of RAM to compile tiger.c from #6845?

xinsuinizhuan, 2024-10-12:
> The Visual Studio compiler uses over 27 GB of RAM to compile tiger.c from #6845?

yes.

xinsuinizhuan, 2024-10-29:
have solved this issue?

liamHowatt, 2024-10-29:
That file is gone now for unrelated reasons so try again.

For reference, here was the file before it was removed: https://github.com/lvgl/lvgl/blob/b7b2c9b7e348f538dd667e80795cff5d3ee9e0a0/demos/svg/assets/tiger.c

closed 2024-11-29 by lvgl-bot