lvgl/lvgl #6926: Dropdown scrollbar never set as LV_STATE_SCROLLED when using encoders

zalexzperez, 2024-09-24:
### LVGL version

v9.2.0

### What happened?

When a dropdown list is scrolled using an encoder, the scrollbar always has the same color: 
[image]
(Encoder simulated with mouse wheel)

I specifically set the color for the LV_PART_SCROLLBAR | LV_STATE_SCROLLED with no effect. Here's the code for the list customization:

```
lv_obj_t *obj = lv_dropdown_get_list(parent_obj);
lv_obj_set_style_bg_color(obj, lv_color_hex(0xffd0d0d0), LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_text_font(obj, &lv_font_montserrat_12, LV_PART_MAIN | LV_STATE_DEFAULT);
lv_obj_set_style_bg_color(obj, lv_color_hex(0xff0000ff), LV_PART_SCROLLBAR | LV_STATE_DEFAULT);
lv_obj_set_style_bg_color(obj, lv_color_hex(0xffff0000), LV_PART_SCROLLBAR | LV_STATE_SCROLLED);
```
Then, I looked into the source code, and found this statement in `lv_dropdown.c` :
`lv_obj_scroll_to_y(dropdown->list, line_y1, LV_ANIM_OFF);`

If `LV_ANIM_OFF` is changed to `LV_ANIM_ON`, then it works as expected:
[image]

Fix found with the help of mvladic

liamHowatt, 2024-10-01:
Nice, I think this is a good fix. I think the scrollbar should have all the same styling capabilities in the dropdown as anywhere else. I wonder if there's any reason to not animate encoder scrolling in the dropdown.

cc @kisvegabor

kisvegabor, 2024-10-02:
I'm ok with making it animated. However `position_to_selected()` (calls `lv_obj_scroll_to_y`) should have an `anim_en` parameter because no animation should happen when e.g. the drop-down opens.

kissa96, 2024-10-17:
@zalexzperez Do you want to make a PR out of this, or should I handle it?

zalexzperez, 2024-10-17:
> @zalexzperez Do you want to make a PR out of this, or should I handle it?

I don't have experience with that, please, handle it yourself

liamHowatt, 2024-11-15:
Closing because it was fixed by #7271 courtesy of @AndreCostaaa. As mentioned there, `lv_dropdown_set_selected` could get an `anim_en` parameter in another PR if the breaking change is ok.

closed 2024-11-15 by liamHowatt