lvgl/lvgl #6931: cmake Linking lv_blend_helium.S.o and lv_blend_neon.S.o resulted in errors during compilation.

1696774082, 2024-09-25:
### LVGL version

v9.2

### What happened?

[ 46%] Building CXX object ../lvgl-9.2.0/out/CMakeFiles/lvgl_thorvg.dir/src/libs/thorvg/tvgLottieExpressions.cpp.o
ld.lld: error: lib/liblvgl.a(lv_blend_helium.S.o) is incompatible with armelf_linux_eabi
ld.lld: error: lib/liblvgl.a(lv_blend_neon.S.o) is incompatible with armelf_linux_eabi
clang-15: error: linker command failed with exit code 1 (use -v to see invocation)

Cross-compiling resulted in errors during linking. Upon checking with the file command, only these two .o files were found to be of x86 architecture, while all the other files are ARM. What’s going on?

kisvegabor, 2024-10-03:
Thanks for reporting. It smells like an error in our CMake file.

How can we reproduce this issue?

kissa96, 2024-10-21:
Hi @1696774082 how is this issue going on?

JoshHeitzman, 2024-11-05:
@kisvegabor I can reproduce the error on Windows x64 (AMD) using Visual Studio 2022:

```
C:\<...>\lvgl\src\widgets\spinbox\lv_spinbox.c(427,56): warning C4267: 'initializing': conversion from 'size_t' to 'const int', possible loss of data 
    lv_spinner.c
    lv_switch.c
    lv_table.c
    lv_tabview.c
    lv_textarea.c
    lv_tileview.c
    lv_win.c
    Generating Code...
    Auto build dll exports
    Couldn't open file 'C:/<...>/lvgl/build/windows-base/lvgl.dir/Debug/lv_blend_helium.obj' with CreateFile()

Build All failed.
```

The repro is to clone the lvgl repo, open the directory using Visual Studio 2022 (Version 17.8.3), and then doing Build->Build All.  Searching I see there is a lv_blend_helium.S with arm assembly in it but there is no lv_blend_helium.c (i.e. the object file appears to for arm only).  So this does indeed look like a cmake issue in that it shouldn't be trying to link in an arm only object file when building for x64.

ChivenZhang, 2024-11-13:
I have met the bug today using the method from offical website : https://docs.lvgl.io/master/details/integration/building/cmake.html

[image]

kisvegabor, 2024-11-20:
Hi,

@MouriNaruto is our Visual Studio expert and the maintainer of the [lv_port_pc_visual_studio](https://github.com/lvgl/lv_port_pc_visual_studio) repo. In this repo do we use CMake to build? 

These `S` files are protected by an [`#if` guard](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/blend/helium/lv_blend_helium.S#L12) but it can happen that the preprocessor is not running on these files for some reason. (According to the standard `*.s` is plain assembly, `*.S` (upper cased) should be preprocessed first).

MouriNaruto, 2024-11-20:
@kisvegabor 

We don't use CMake in the lv_port_pc_visual_studio repo. We use the MSBuild, the Visual Studio builtin build system.

Kenji Mouri

JoshHeitzman, 2024-11-26:
One additional piece of information on this.  When I run the following command to generate the Visual Studio solution and project files:

`cmake -DLV_CONF_SKIP=true -S . -B . -G "Visual Studio 17 2022" -A x64`

I'm able build just fine using the generated VS solution.

The issue seems to be specific to using the CMake presets by opening the directory as CMake project in VS.

liamHowatt, 2024-12-02:
Related issue with potential fix: #7371

PR #7401 (2024-12-04): fix(cmake): fix msvc building error
fix #6931 #7371 
(similar issue exists when using msvc as compiler, and ninja or nmake as generator, they are also being fixed. )

further investigation of the issue:

CMake generates a PreLinkEvent in the "lvgl.vcxproj" when using visual studio as the generator. 

```
    <PreLinkEvent>
      <UseUtf8Encoding>Always</UseUtf8Encoding>
      <Message>Auto build dll exports</Message>
      <Command>setlocal
cd D:\lvgl_windows_9.3\build
if %errorlevel% neq 0 goto :cmEnd
D:
if %errorlevel% neq 0 goto :cmEnd
"C:\Program Files\CMake\bin\cmake.exe" -E __create_def  [... cut]
closed 2024-12-04 by liamHowatt