lvgl/lvgl #6937: Drawing a radial or conical gradient to the canvas widget

kdschlosser, 2024-09-25:
### Introduce the problem

I have tried and tried to render gradients to a canvas widget and I have not had any success. The program just sits there spinning it's wheels and never renders anything. It gets stuck on the radial init function.

Here is the code I am using. 

```c

lv_obj_t *scrn = lv.screen_active();
lv_obj_set_style_bg_color(scrn, lv_color_hex(0x000000), 0);

lv_area_t area1 = {
    .x1=0,
    .y1=0,
    .x2=240,
    .y2=160
};

lv_grad_dsc_t grad1 = {
    .stops= {
        (lv_gradient_stop_t){
            .color=lv_color_hex(0x0000FF),
            .opa=255,
            .frac=10
        },
        (lv_gradient_stop_t){
            .color=lv_color_hex(0xFF0000),
            .opa=255,
            .frac=100
        }
    },
    .stops_count=2
};

lv_grad_radial_init(&grad1, 120, 80, 120, 80, lv.GRAD_EXTEND.PAD);
lv_gradient_radial_setup(&grad, &area);

lv_draw_rect_dsc_t rect1 = {
    .bg_grad=&grad1
};

lv_obj_t *canvas1 = lv_canvas_create(scrn);
lv_obj_set_pos(canvas1, 0, 0);
lv_obj_set_size(canvas1, 240, 160);

uint8_t *buf1 = (uint8_t *)malloc(240 * 160 * 3);
lv_canvas_set_buffer(canvas1, buf1, 240, 160, lv.COLOR_FORMAT.RGB888);
lv_canvas_fill_bg(canvas1, lv_color_hex(0x000000))

lv_layer_t layer1 = {};

lv_canvas_init_layer(canvas1, &layer1);
lv_draw_rect(&layer1, &rect1, &area1);
lv_canvas_finish_layer(canvas1, &layer1);
lv_gradient_radial_cleanup(&grad1);

lv_area_t area2 = {
    .x1=0,
    .y1=0,
    .x2=240,
    .y2=160
};

lv_grad_dsc_t grad2 = {
    .stops= {
        (lv_gradient_stop_t){
            .color=lv_color_hex(0x0000FF),
            .opa=255,
            .frac=10
        },
        (lv_gradient_stop_t){
            .color=lv_color_hex(0xFF0000),
            .opa=255,
            .frac=100
        }
    },
    .stops_count=2
};

lv_grad_conical_init(&grad2, 120, 80, 0, 3600, lv.GRAD_EXTEND.PAD);
lv_gradient_conical_setup(&grad2, &area2);

lv_draw_rect_dsc_t rect2 = {
    .bg_grad=&grad2
};

lv_obj_t *canvas2 = lv_canvas_create(scrn);
lv_obj_set_pos(canvas2, 240, 0);
lv_obj_set_size(canvas2, 240, 160);

uint8_t *buf2 = (uint8_t *)malloc(240 * 160 * 3);
lv_canvas_set_buffer(canvas2, buf2, 240, 160, lv.COLOR_FORMAT.RGB888);
lv_canvas_fill_bg(canvas2, lv_color_hex(0x000000));

lv_layer_t layer2 = {};

lv_canvas_init_layer(canvas2, &layer2);
lv_draw_rect(&layer2, &rect2, &area2);
lv_canvas_finish_layer(canvas2, &layer2);

lv_gradient_conical_cleanup(&grad2);

```

Maybe someone can shed some light as to why this is not working. It is getting stuck on this line...

```c
lv_grad_radial_init(&grad1, 120, 80, 120, 80, lv.GRAD_EXTEND.PAD);
```

I put this under a development discussion mostly because there is zero documentation on how to use the gradients and also because I am not sure if there is a bug or there is something that I am doing wrong.

TridentTD, 2024-09-26:
[image]

```
#define CANVAS_WIDTH  480
#define CANVAS_HEIGHT 320

void lv_example_canvas_radius_conical_gradient(void)
{
  /*Create a buffer for the canvas*/
  LV_DRAW_BUF_DEFINE_STATIC(draw_buf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_ARGB8888);
  LV_DRAW_BUF_INIT_STATIC(draw_buf);

  /*Create a canvas and initialize its palette*/
  lv_obj_t * canvas = lv_canvas_create(lv_screen_active());
  lv_canvas_set_draw_buf(canvas, &draw_buf);

  lv_canvas_fill_bg(canvas, lv_color_hex3(0xccc), LV_OPA_COVER);
  lv_obj_center(canvas);

  lv_layer_t layer;
  lv_canvas_init_layer(canvas, &layer);

  /* conical gradient */
  lv_draw_rect_dsc_t dsc;
  lv_draw_rect_dsc_init(&dsc);
  dsc.bg_color      = lv_palette_main(LV_PALETTE_RED);
  dsc.border_width  = 0;
  dsc.radius        = 0;
  const lv_color_t grad_color[2] = {
    lv_palette_main(LV_PALETTE_CYAN),
    lv_palette_main(LV_PALETTE_RED),
  };
  lv_gradient_init_stops(&dsc.bg_grad, grad_color, NULL, NULL, 2);
  dsc.bg_grad.dir = LV_GRAD_DIR_CONICAL;
  dsc.bg_grad.params.conical.center.x = lv_pct(50);       /*center x position*/
  dsc.bg_grad.params.conical.center.y = lv_pct(50);       /*center y position*/
  dsc.bg_grad.params.conical.start_angle = 135;           /*start angle*/
  dsc.bg_grad.params.conical.end_angle = 135+270;         /*end angle*/
  dsc.bg_grad.extend = LV_GRAD_EXTEND_PAD;                /*color pattern outside the vector*/

  lv_area_t coords = {10, 10, 10+200, 10+120};
  lv_draw_rect(&layer, &dsc, &coords);

  /* radial gradient */
  lv_draw_rect_dsc_t dsc2;
  lv_draw_rect_dsc_init(&dsc2);
  dsc2.bg_color      = lv_color_black();
  dsc2.border_width  = 0;
  dsc2.radius        = 0;

  const lv_color_t grad_color2[2] = {
    lv_palette_main(LV_PALETTE_PURPLE),
    lv_color_black(),
  };
  lv_gradient_init_stops(&dsc2.bg_grad, grad_color2, NULL, NULL, 2);
  dsc2.bg_grad.dir = LV_GRAD_DIR_RADIAL;
  dsc2.bg_grad.params.radial.focal.x = lv_pct(50);                                  /*start circle center x position*/
  dsc2.bg_grad.params.radial.focal.y = lv_pct(50);                                  /*start circle center y position*/
  dsc2.bg_grad.params.radial.focal_extent.x = dsc2.bg_grad.params.radial.focal.x;   /*start circle point x coordinate*/
  dsc2.bg_grad.params.radial.focal_extent.y = dsc2.bg_grad.params.radial.focal.y;   /*start circle point y coordinate*/
  dsc2.bg_grad.params.radial.end.x = dsc2.bg_grad.params.radial.focal.x + 2;        /*end circle center x position*/
  dsc2.bg_grad.params.radial.end.y = dsc2.bg_grad.params.radial.focal.y + 2;        /*end circle center y position*/
  dsc2.bg_grad.params.radial.end_extent.x = lv_pct(95);                             /*end circle point x coordinate*/
  dsc2.bg_grad.params.radial.end_extent.y = lv_pct(95);                             /*end circle point y coordinate*/
  dsc2.bg_grad.extend = LV_GRAD_EXTEND_PAD;                                         /*color pattern outside the border circles*/

  lv_area_t coords2 = {250, 170, 250+200, 170+120};
  lv_draw_rect(&layer, &dsc2, &coords2);

  lv_canvas_finish_layer(canvas, &layer);
}
```

Note
 - Radial or conical gradient while still supporting the drawing of rectangles. 
   The drawing of arcs is not yet functional.

kdschlosser, 2024-09-26:
you shouldn't have to use 1/2 of that code that is there. that is why there are the init functions. so perhaps it is the init functions that are broken?

kdschlosser, 2024-09-26:
and those gradients look terrible. the banding is pretty bad.

kisvegabor, 2024-09-26:
> and those gradients look terrible. the banding is pretty bad.

No dithering is added yet.

kdschlosser, 2024-09-26:
dithering should be something that is part of the gradient process by default. LV_COLOR_FORMAT_ARGB8888 was used and not RGB565 so you should NOT being seeing those steps. LVGL is not properly calculating the colors that are to be used.

In the case of a radial gradient the radius of the gradient is the number of color transitions that should be occurring. so if I want to go from red to blue and the radius is 255 pixels there should be 255 different colors used to get from one color to the next. 

with the radial gradient what in the world is this??

```c
  dsc2.bg_grad.params.radial.focal.x = lv_pct(50);                                  /*start circle center x position*/
  dsc2.bg_grad.params.radial.focal.y = lv_pct(50);                                  /*start circle center y position*/
  dsc2.bg_grad.params.radial.focal_extent.x = dsc2.bg_grad.params.radial.focal.x;   /*start circle point x coordinate*/
  dsc2.bg_grad.params.radial.focal_extent.y = dsc2.bg_grad.params.radial.focal.y;   /*start circle point y coordinate*/
  dsc2.bg_grad.params.radial.end.x = dsc2.bg_grad.params.radial.focal.x + 2;        /*end circle center x position*/
  dsc2.bg_grad.params.radial.end.y = dsc2.bg_grad.params.radial.focal.y + 2;        /*end circle center y position*/
  dsc2.bg_grad.params.radial.end_extent.x = lv_pct(95);                             /*end circle point x coordinate*/
  dsc2.bg_grad.params.radial.end_extent.y = lv_pct(95);                             /*end circle point y coordinate*/
```

it looks like 
```c
  dsc2.bg_grad.params.radial.focal.x = lv_pct(50);                                  /*start circle center x position*/
  dsc2.bg_grad.params.radial.focal.y = lv_pct(50);                                  /*start circle center y position*/
```

and 
```c
  dsc2.bg_grad.params.radial.focal_extent.x = dsc2.bg_grad.params.radial.focal.x;   /*start circle point x coordinate*/
  dsc2.bg_grad.params.radial.focal_extent.y = dsc2.bg_grad.params.radial.focal.y;   /*start circle point y coordinate*/
```

are the exact same thing. they control the exact same thing...

why does the start circle and the end circle have to be shifted from each other? can they be the same coordinates. they should be able to be set to the same coordinates.

the end focal is set to 50% of what? 

also just to let you know because of the use of nested anonymous unions and structures in `lv_grad_desc_t` they are not available in the MicroPython binding. You cannot have a nested named union or structure while providing a field name. this is not correct C syntax. I am not even sure why there is the `params` field name to begin with. You are only going to be using only one of the nested structures inside of the union depending on what type of gradient you are using. having "params" and then "radial", "linear" or "conical" after that serves no purpose, it's just extra key strokes that serve no purpose.

kdschlosser, 2024-09-26:
That banding that is being seen is something wrong in the rendering algorithm. 

[image]

zoom into that and you will see what I am talking about. for each "step" there are 2 colors being using and they don't form a linear color change.

kdschlosser, 2024-09-26:
why does this need to be done??

```c
dsc.bg_color      = lv_palette_main(LV_PALETTE_RED);
```

and why does the start angle and the end angle of the conical gradient deviate from the API? it should have the decimal shifted to the right one place. 135° should be 1350 That is how all of the rotation angles are done. This allows for a higher precision. 

to calculate the number of steps used for a conical gradient you would use this math..

```
R = min(width, height) / 2
C = 2 × π × R

angle_range = stop_angle - start_angle
assert(angle_range < 3600 and angle_range > 0)

num_steps = angle_range / 3600 * C

```

radial and conical gradients are linear gradients. it's simply how the colors in the gradient are applied when rendering. the number of colors that are using in the gradient array should be the radius for a radial gradient and for the conical it should be calculated using the math from above. 

In the radial gradient here is what the gradient is when it is converted into a linear gradient

[image]
It's not correct that's for sure...

I am not sure how it determined the colors and only 35 of them at that. it should have been 60 colors but with how the bounding rectangle is being calculated when doing the rendering it should have been 100 colors. I am not sure how the radius of the gradient is being determine because the gradient goes out of the bounding rectangle vertically and horizontally it is smaller than the bounding area. The entire gradient should fit inside of the bounding area. 

Here is an example of how the gradient should be done to make the colors correct. This handles everything in a single go. 
```c
void linear_interpolation(lv_color_t *in_color, int16_t red_dif, int16_t green_dif, int16_t blue_dif, float frac, lv_color_t *ret_color) {
    ret_color->red = (uint8_t)(in_color->red + red_dif * frac);
    ret_color->green = (uint8_t)(in_color->green + green_diff * frac);
    ret_color->blue = (uint8_t)(in_color->blue + blue_diff * frac);
}

lv_color_t c1 = lv_color_black();
lv_color_t c2 = lv_palette_main(LV_PALETTE_PURPLE);

int16_t red_dif = (int16_t)c2.red - (int16_t)c1.red;
int16_t green_diff = (int16_t)c2.green - (int16_t)c1.green;
int16_t blue_diff = (int16_t)c2.blue - (int16_t)c1.blue;

lv_color_t *colors = malloc(sizeof(lv_color_t) * num_steps);
float num_steps_f = (float)num_steps;

for (int i=0;i<num_steps;i++) {
    linear_interpolation(&c1, red_dif, green_diff, blue_diff, i / num_steps_f, &colors[i]);
```

and this is what it comes out looking like..

[image]

kisvegabor, 2024-09-30:
Oh, great findings! Could you send a PR with the suggested changes? If you can please also add some tests.

kdschlosser, 2024-09-30:
That's going to be best suited for the person that added the gradients to LVGL.

kisvegabor, 2024-10-02:
I'm afraid @zjanosy is busy with an other project now. Correct me if I'm wrong :slightly_smiling_face: 

@kdschlosser if you have a working code I'd be happy to test it, but it's easier to see what changed in a PR.

zjanosy, 2024-10-02:
@kdschlosser Currenly the algorithm always generates a 256 point color lookup table for the color gradient. The gradient algorithm returns a value between 0 and 255 for any (x, y) coordinate. This value is used to index into the color lookup table. 

There are two sources of error: the limited resolution of the color lookup table, and the calculation error in the gradient algorithm. Certainly, we can improve either of them (or both), but it would likely require more memory/CPU. But feel free to try!

To be able to integrate the gradient algorithm into the rendering it is necessary that it can be calculated line by line. The current algorithm uses an incremental (recursive) calculation to speed up the calculation a little bit, which may also introduce some errors. Moreover, since we only use integer arithmetics, the trigonometric and square root functions are only approximations.

> why does the start circle and the end circle have to be shifted from each other? can they be the same coordinates. they should be able to be set to the same coordinates.

It is possible to set a focal point which is different from the center for additional effects. For the simplest case they are the same, but the algorithm can handle the more general case as well.

> the focal is set to 50% of what?

> end_extent is set to 95% of what?

The percent values are relative to the object's width and height. I.e., 50% is at the center of the object. Using the percent values it is easy to create the most common cases.

The init functions were provided only for convenience for the most common cases. However, it is possible to specify the parameters by hand for the more sophisticated cases.

zjanosy, 2024-10-02:
> why does this need to be done??
> dsc.bg_color      = lv_palette_main(LV_PALETTE_RED);

That's not necessary, I think. Probably it was a leftover from some experiments.

> and why does the start angle and the end angle of the conical gradient deviate from the API? it should have the decimal shifted to the right one place. 135° should be 1350 That is how all of the rotation angles are done. This allows for a higher precision.

The trigonometric functions needed degrees. But you are right, we should not use different units at different places. Can you fix this? Right now I don't have time to get back to this.

kdschlosser, 2024-10-02:
I have to comb over the code to see if I am able to grasp what is being done without it having to be fully explained to me. The inner working of LVGL are quite intricate and it's not easy to follow the data path through the code to understand how things are being done.

I will take some time to look at what is going on and if I have questions I will ask them and hopefully someone will be able to answer them. I know that the gradients are a very tough nut to crack so only integer math is used and to keep the rendering time up where it should be.

So there is a hard limit set for the array stored in `_lv_gradient_cache_t.lv_color_t` of 256 colors? Is that consistent across all gradient types or only conical and radial?

closed 2024-10-26 by lvgl-bot