lvgl/lvgl #6944: Add back label recoloring

kisvegabor, 2024-09-26:
### Introduce the problem

As [requested by multiple people](https://github.com/lvgl/lvgl/issues/6019) the label recoloring feature of v7 and v8 should be added back.

### Proposal

The same code that we had in v8 should work in v9 too. To re-implement this feature mainly these parts needs to be added again:
- https://github.com/lvgl/lvgl/blob/release/v8.4/src/widgets/lv_label.c#L484-L489
- https://github.com/lvgl/lvgl/blob/release/v8.4/src/draw/lv_draw_label.c#L244-L282
- https://github.com/lvgl/lvgl/blob/release/v8.4/src/misc/lv_txt.c#L196-L205

In general searching for `LV_TEXT_FLAG_RECOLOR` should be a good starting point.

C47D, 2024-09-29:
I can take over this if you still need the help @kisvegabor

kisvegabor, 2024-09-30:
Fantastic! Thank you Carlos! :heart: 

> I guess also we need to add:

All seems correct.

kisvegabor, 2024-10-15:
Done in https://github.com/lvgl/lvgl/pull/6966

Thank you Carlos! :pray:

closed 2024-10-15 by kisvegabor