lvgl/lvgl #6945: Spinner animation overflows after a while

KlemenDEV, 2024-09-26:
### LVGL version

v9.x.x

### What happened?

In LVGL 9, it appears in some cases, the spinner animation overflows. See example below:

[image]

We use code from https://docs.lvgl.io/master/widgets/spinner.html to define the spinner, a completely default implementation.

This does not happen on e.g. Windows emulator but happens on our final device (STM32-based solution, LV_OS_CMSIS_RTOS2).

It takes around 2 minutes when the animation arc gets longer and longer and then it seems like the start goes before the end of the arc stretching.

Looking at https://github.com/lvgl/lvgl/blob/40577fe685ee0592010438bdb65fe07c892f364d/src/widgets/spinner/lv_spinner.c#L56, it seems the same animation structure is used to call lv_anim_set_exec_cb and lv_anim_start twice, is this ok?

C47D, 2024-09-28:
Maybe it has to do with integers width, as in the Windows machine they are 64bit and in the STM they are 32bits, I'll take a look at the animation code, but I'm not sure I'll be able to catch the bug, thanks for the GIF, it is helpful to notice the unexpected behavior.

KlemenDEV, 2024-09-29:
I have also noticed this and tried "fixing" it with this code:

```c
void lv_spinner_set_anim_params(lv_obj_t * obj, uint32_t t, uint32_t angle)
{
    /*Delete the current animation*/
    lv_anim_delete(obj, NULL);

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, obj);
    lv_anim_set_exec_cb(&a, arc_anim_end_angle);
    lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
    lv_anim_set_duration(&a, t);
    lv_anim_set_values(&a, angle, 360 + angle);
    lv_anim_start(&a);

    lv_anim_set_path_cb(&a, lv_anim_path_custom_bezier3);
    lv_anim_set_bezier3_param(&a, LV_BEZIER_VAL_FLOAT(0.42), LV_BEZIER_VAL_FLOAT(0.58),
                              LV_BEZIER_VAL_FLOAT(0), LV_BEZIER_VAL_FLOAT(1));

    lv_anim_t a2;
    lv_anim_init(&a2);
    lv_anim_set_var(&a2, obj);
    lv_anim_set_repeat_count(&a2, LV_ANIM_REPEAT_INFINITE);
    lv_anim_set_duration(&a2, t);

    lv_anim_set_path_cb(&a2, lv_anim_path_custom_bezier3);
    lv_anim_set_bezier3_param(&a2, LV_BEZIER_VAL_FLOAT(0.42), LV_BEZIER_VAL_FLOAT(0.58),
                              LV_BEZIER_VAL_FLOAT(0), LV_BEZIER_VAL_FLOAT(1));

    lv_anim_set_values(&a2, 0, 360);
    lv_anim_set_exec_cb(&a2, arc_anim_start_angle);
    lv_anim_start(&a2);

    lv_arc_set_bg_angles(obj, 0, 360);
    lv_arc_set_rotation(obj, 270);
}
```

However this did not fix the reported bug

C47D, 2024-09-29:
Now that I see the animation it looks like the end angle reaches the start angle without the start angle getting bigger (let's call it gets stopped), then the end angle 'restarts' its value, e.g. 20, so it looks like the animation overflows.

kisvegabor, 2024-09-30:
I have any idea of what the issue could be. While the animation is running an absolute time is used. That is LVGL calculates where the animation should at a given time, no matter if the animation handler is executed every milliseconds or only every 100 ms.

But if the animation is called rarely, e.g every 100ms, and the animation is repeated the residual time at the end is not handled correctly. The following is happening e.g. with a duration of 270ms and repeat:
- 0ms: start value
- 100ms: some intermediate value
- 200ms: some later intermediate value
- 300ms: animation restarts with the start value, but it should be already at 30ms.

I think these offsets from the residual times are slowly accumulating. 

It seems it should be handled [here](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_anim.c#L595). 

Do you have time to take look?

KlemenDEV, 2024-09-30:
This could be indeed the case, because it happens sooner the less often lv_timer_handler is called (due to e.g. CPU load by other tasks).

We can not log on our end-device due to absence of logging ports.

kisvegabor, 2024-10-02:
It's still hard to reproduce on my end, but I opened a PR with the supposed solution. Could you try it out?
https://github.com/lvgl/lvgl/pull/6989

KlemenDEV, 2024-10-02:
I have tested the PR and it indeed fixes the bug reported in this ticket.

closed 2024-10-12 by XuNeo

KlemenDEV, 2024-11-11:
@kisvegabor, would it be possible to reopen this, or should I open a new ticket?

This is happening again using LV_USE_NEMA_GFX without OS enabled.

KlemenDEV, 2024-11-11:
https://github.com/lvgl/lvgl/issues/7276

PR #6989 (2024-10-12): fix(anim): compensate over time
Important with repeating animatins when lv_timer_handler is called with larger period time.

Fixes #6945

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts [... cut]