lvgl/lvgl #6950: #6622 Causes Compilation Errors in ESP-IDF v5.x.x

pjensen-addco, 2024-09-26:
### LVGL version

v9.2.0

### What happened?

PR #6622 added several formatting strings to dump debugging information.

[Per this comment](https://github.com/espressif/esp-idf/issues/6906#issuecomment-1207373706) in IDF version 5 and above uint32s are now considered longs, which is triggering format string exceptions upon compile.

```
In file included from components\lvgl\src\misc\cache\../../draw/../misc/lv_assert.h:17,
                 from components\lvgl\src\misc\cache\../../draw/../misc/lv_color.h:17,
                 from components\lvgl\src\misc\cache\../../draw/lv_draw_buf.h:18,
                 from components\lvgl\src\misc\cache\../../draw/lv_image_decoder.h:18,
                 from components\lvgl\src\misc\cache\../../draw/lv_image_decoder_private.h:17,
                 from components\lvgl\src\misc\cache\lv_image_cache.c:10:
components\lvgl\src\misc\cache\lv_image_cache.c: In function 'iter_inspect_cb':
components\lvgl\src\misc\cache\lv_image_cache.c:176:33: error: format '%d' expects argument of type 'int', but argument 8 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  176 | #define IMAGE_CACHE_DUMP_FORMAT "       %4dx%-4d        %9d     %d      %d      "
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
components\lvgl\src\misc\cache\../../draw/../misc/lv_log.h:131:96: note: in definition of macro 'LV_LOG_USER'
  131 | #    define LV_LOG_USER(...) lv_log_add(LV_LOG_LEVEL_USER, LV_LOG_FILE, LV_LOG_LINE, __func__, __VA_ARGS__)
      |                                                                                                ^~~~~~~~~~~
components\lvgl\src\misc\cache\lv_image_cache.c:179:25: note: in expansion of macro 'IMAGE_CACHE_DUMP_FORMAT'
  179 |             LV_LOG_USER(IMAGE_CACHE_DUMP_FORMAT "file\t%-12p\t%s", header->w, header->h, decoded->data_size, header->cf,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~
components\lvgl\src\misc\cache\lv_image_cache.c:176:59: note: format string is defined here
  176 | #define IMAGE_CACHE_DUMP_FORMAT "       %4dx%-4d        %9d     %d      %d      "
      |                                                         ~~^
      |                                                           |
      |                                                           int
      |                                                         %9ld
```

I looked at the unit tests for 6622, and it appears that the unit tests are testing against IDF version 4.4.0 as a baseline, and not against the most recent version.

Is there a good workaround other than suppressing these warnings or modifying LVGL manually? Should LVGL be testing against the latest IDF in the test pipelines?

### How to reproduce?

Our project is using PlatformIO as the build platform, but I believe this should be able to be replicated using the IDF.

1. Install PlatformIO.
2. Install Espressif Platform 6.9.0, which uses ESP-IDF 5.3.1
3. Add LVGL 9.3 as a component.
4. Attempt to compile.

jakubstepniewicz, 2024-09-27:
Sorry it's not a Pull Request, but here's a quick fix for the formatting issues (also a new one introduced in #6761)
`
[patch.txt](https://github.com/user-attachments/files/17164364/patch.txt)
`

pjensen-addco, 2024-09-27:
Thank you! Worked a charm. For those of you attempting to replicate for v9.2, remove the first git diff command, as the 6761 change is not in that release

kisvegabor, 2024-10-03:
Some similar warnings were already fixed. Could you check with the latest master? If something is still missing could you send a PR?

kissa96, 2024-10-21:
How are we doing with this issue?

pjensen-addco, 2024-10-22:
Apologies, got pulled off on another project for a while.

Coming back to this and updating to the most recent main appears to have fixed this.

closed 2024-10-22 by pjensen-addco