lvgl/lvgl #6951: line widget not drawn correctly

Mair, 2024-09-26:
### LVGL version

v9.2.0-52-gdc9e42360

### What happened?

1. using the VS simulator, I cloned the lates
2. replaced lv_demo_widgets(); call in main() with the code in the sample as per https://docs.lvgl.io/master/widgets/line.html (see below)

# Expected result
[image]

# Actual result
[image]

### How to reproduce?

```c
 /*Create an array for the points of the line*/
    static lv_point_precise_t line_points[] = { {5, 5}, {70, 70}, {120, 10}, {180, 60}, {240, 10} };

    /*Create style*/
    static lv_style_t style_line;
    lv_style_init(&style_line);
    lv_style_set_line_width(&style_line, 8);
    lv_style_set_line_color(&style_line, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_rounded(&style_line, true);

    /*Create a line and apply the new style*/
    lv_obj_t * line1;
    line1 = lv_line_create(lv_screen_active());
    lv_line_set_points(line1, line_points, 5);     /*Set the points*/
    lv_obj_add_style(line1, &style_line, 0);
    lv_obj_center(line1);
```

cosmindanielradu19, 2024-09-27:
I also have a similar problem, I'm trying to create a dashed line, but the SW Draw unit creates a normal line.

[image]

### How to reproduce?
```c
static void line_create(lv_style_t * style)
{
    lv_obj_t * obj = lv_line_create(lv_screen_active());
    static lv_point_precise_t line_points[] = { {0, 0}, {OBJ_WIDTH, OBJ_HEIGHT}};
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_obj_set_style_line_color(obj, lv_palette_main(LV_PALETTE_ORANGE), 0);
    lv_obj_set_size(obj, OBJ_WIDTH, OBJ_HEIGHT);
    lv_line_set_points(obj, line_points, 2);
    lv_obj_center(obj);
}

static void line_dashed(void)
{
    lv_style_set_line_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_line_width(&style_common, LINE_WIDTH);
    lv_style_set_line_color(&style_common, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_dash_width(&style_common, LINE_DASH);
    lv_style_set_line_dash_gap(&style_common, LINE_GAP);
    line_create(&style_common);
}

/* function call from main */
line_dashed();
```

TridentTD, 2024-09-27:
@cosmindanielradu19
Dashed lines are supported only for horizontal and vertical lines. Diagonal lines are not supported.

Mair, 2024-09-30:
@kisvegabor possibly, Shall I log this as an issue on the VS repo?

kisvegabor, 2024-10-02:
> I can confirm that while the chart looks good in lv_demo_widgets():

Actually it doesn't look good either. The triangles are missing:
[image]

Can you check if it enters [here](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw.c#L527)?

vwheeler63, 2024-10-02:
> Can you check if it enters [here](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw.c#L527)?

Yes, it does enter the `LV_DRAW_TASK_TYPE_TRIANGLE` case branch immediately after I click the "Analytics" tab.

kisvegabor, 2024-10-09:
Duplicated of https://github.com/lvgl/lvgl/issues/6811. Let's continue there