lvgl/lvgl #6954: Software drawing wrong styles

cosmindanielradu19, 2024-09-27:
### LVGL version

v9.2

### What happened?

I was running some unit tests, and I've found a few issues on SW drawing

1. Rectangle with rounder corners and a background image doesn't have the corners rounded.
[image]

2. Rectangle styled is a rectangle with background image, but the image is not fitted into the rectangle.
[image]

3. Circle styled with a background image, same issue as before, the image is not fitted into the circle
[image]

### How to reproduce?

1.
```c
static void base_rect_create(lv_style_t * style, bool isCircle)
{
    lv_obj_t * obj = lv_obj_create(lv_screen_active());
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_obj_set_size(obj, OBJ_WIDTH, isCircle ? OBJ_WIDTH : OBJ_HEIGHT);
    lv_obj_center(obj);
}

static void rect_create(lv_style_t * style, bool isCircle)
{
    lv_style_set_bg_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_bg_color(style, lv_palette_main(LV_PALETTE_ORANGE));
    base_rect_create(style, isCircle);
}

static void rectangle_rounded(void)
{
    lv_style_reset(&style_common);
    lv_style_set_radius(&style_common, RADIUS);
    rect_create(&style_common, false);
}

rectangle_rounded();
```

2.
```c
static void base_rect_create(lv_style_t * style, bool isCircle)
{
    lv_obj_t * obj = lv_obj_create(lv_screen_active());
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_obj_set_size(obj, OBJ_WIDTH, isCircle ? OBJ_WIDTH : OBJ_HEIGHT);
    lv_obj_center(obj);
}

static void styled_rect_create(lv_style_t * style, bool isCircle)
{
    /* Solid Fill */
    lv_style_set_bg_opa(style, scene_with_opa ? LV_OPA_80 : LV_OPA_COVER);

    /* Gradient fill */
    lv_style_set_bg_grad_color(style, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_bg_grad_dir(style, LV_GRAD_DIR_HOR);

    /* Background image */
    lv_style_set_bg_image_src(style, &img_cogwheel_rgb565_128);
    lv_style_set_bg_image_opa(style, scene_with_opa ? LV_OPA_10 : LV_OPA_COVER);
    lv_style_set_bg_image_tiled(style, true);

    /* Border */
    lv_style_set_border_width(style, BORDER_WIDTH);

    lv_style_set_border_opa(style, scene_with_opa ? LV_OPA_80 : LV_OPA_COVER);
    lv_style_set_border_side(style, LV_BORDER_SIDE_FULL);

    /* Outline */
    lv_style_set_outline_width(style, OUTLINE_WIDTH);
    lv_style_set_outline_color(style, lv_palette_main(LV_PALETTE_RED));
    lv_style_set_outline_opa(style, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_outline_pad(style, OUTLINE_PAD);

    /* Rounded shadow takes too much memory */
    if(!isCircle) {
        /* Shadow */
        lv_style_set_shadow_width(style, SHADOW_WIDTH);
        lv_style_set_shadow_spread(style, SHADOW_SPREAD);
        lv_style_set_shadow_ofs_x(style, SHADOW_OFFSET_X);
        lv_style_set_shadow_ofs_y(style, SHADOW_OFFSET_Y);
        lv_style_set_shadow_opa(style, scene_with_opa ? LV_OPA_80 : LV_OPA_COVER);
    }

    base_rect_create(style, isCircle);
}

static void rectangle_styled(void)
{
    lv_style_reset(&style_common);
    lv_style_set_radius(&style_common, 0);
    styled_rect_create(&style_common, false);
}

rectangle_styled();
```

3.
```c
static void base_rect_create(lv_style_t * style, bool isCircle)
{
    lv_obj_t * obj = lv_obj_create(lv_screen_active());
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_obj_set_size(obj, OBJ_WIDTH, isCircle ? OBJ_WIDTH : OBJ_HEIGHT);
    lv_obj_center(obj);
}

static void styled_rect_create(lv_style_t * style, bool isCircle)
{
    /* Solid Fill */
    lv_style_set_bg_opa(style, scene_with_opa ? LV_OPA_80 : LV_OPA_COVER);

    /* Gradient fill */
    lv_style_set_bg_grad_color(style, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_bg_grad_dir(style, LV_GRAD_DIR_HOR);

    /* Background image */
    lv_style_set_bg_image_src(style, &img_cogwheel_rgb565_128);
    lv_style_set_bg_image_opa(style, scene_with_opa ? LV_OPA_10 : LV_OPA_COVER);
    lv_style_set_bg_image_tiled(style, true);

    /* Border */
    lv_style_set_border_width(style, BORDER_WIDTH);

    lv_style_set_border_opa(style, scene_with_opa ? LV_OPA_80 : LV_OPA_COVER);
    lv_style_set_border_side(style, LV_BORDER_SIDE_FULL);

    /* Outline */
    lv_style_set_outline_width(style, OUTLINE_WIDTH);
    lv_style_set_outline_color(style, lv_palette_main(LV_PALETTE_RED));
    lv_style_set_outline_opa(style, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_outline_pad(style, OUTLINE_PAD);

    /* Rounded shadow takes too much memory */
    if(!isCircle) {
        /* Shadow */
        lv_style_set_shadow_width(style, SHADOW_WIDTH);
        lv_style_set_shadow_spread(style, SHADOW_SPREAD);
        lv_style_set_shadow_ofs_x(style, SHADOW_OFFSET_X);
        lv_style_set_shadow_ofs_y(style, SHADOW_OFFSET_Y);
        lv_style_set_shadow_opa(style, scene_with_opa ? LV_OPA_80 : LV_OPA_COVER);
    }

    base_rect_create(style, isCircle);
}

static void circle_styled(void)
{
    lv_style_reset(&style_common);
    lv_style_set_radius(&style_common, LV_RADIUS_CIRCLE);
    lv_style_set_bg_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    styled_rect_create(&style_common, true);
}

circle_styled();
```

kisvegabor, 2024-09-30:
True! We have already added a radius parameter (`img_draw_dsc->radius)` but it's not used in SW render yet. I mark this issue as important to be sure that it will be added soon.

closed 2025-04-10 by kisvegabor

cosmindanielradu19, 2025-04-10:
Great! Thanks