lvgl/lvgl #6957: lvgl via CMake FetchContent + SDL does not allow for linking SDL to lvgl

glebm, 2024-09-28:
### LVGL version

0daebca18b2e7d4c51d5b53e7f2a60df634850f2

### What happened?

I'm trying to set up a project using lvgl via FetchContent and SDL.

The problem is that lvgl's CMake doesn't seem to provide a way to link SDL to `lvgl::lvgl` target when using it via `FetchContent`.

If I don't link SDL, I get a build error:

```
FAILED: _deps/lvgl-build/CMakeFiles/lvgl.dir/src/drivers/sdl/lv_sdl_keyboard.c.o 
/usr/bin/cc -DLV_CONF_INCLUDE_SIMPLE -DLV_CONF_PATH=platforms/sdl3/lv_conf.h -DLV_LVGL_H_INCLUDE_SIMPLE -isystem build/_deps/lvgl-src -isystem platforms/sdl3 -isystem build/_deps/lvgl-build -g -MD -MT _deps/lvgl-build/CMakeFiles/lvgl.dir/src/drivers/sdl/lv_sdl_keyboard.c.o -MF _deps/lvgl-build/CMakeFiles/lvgl.dir/src/drivers/sdl/lv_sdl_keyboard.c.o.d -o _deps/lvgl-build/CMakeFiles/lvgl.dir/src/drivers/sdl/lv_sdl_keyboard.c.o -c build/_deps/lvgl-src/src/drivers/sdl/lv_sdl_keyboard.c
In file included from build/_deps/lvgl-src/src/drivers/sdl/../../display/../misc/../lv_conf_internal.h:54,
                 from build/_deps/lvgl-src/src/drivers/sdl/../../display/../misc/lv_types.h:16,
                 from build/_deps/lvgl-src/src/drivers/sdl/../../display/lv_display.h:16,
                 from build/_deps/lvgl-src/src/drivers/sdl/lv_sdl_window.h:17,
                 from build/_deps/lvgl-src/src/drivers/sdl/lv_sdl_keyboard.h:16,
                 from build/_deps/lvgl-src/src/drivers/sdl/lv_sdl_keyboard.c:9:
platforms/sdl3/lv_conf.h:1041:37: fatal error: SDL3/SDL.h: No such file or directory
 1041 |     #define LV_SDL_INCLUDE_PATH     <SDL3/SDL.h>
```

If I try to link SDL to lvgl like this:

```cmake
target_link_libraries(lvgl::lvgl PRIVATE SDL3::SDL3)
```

Then I get the following error:

```cmake
CMake Error at platforms/sdl3/CMakeLists.txt:5 (target_link_libraries):
  Cannot specify link libraries for target "lvgl::lvgl" which is not built by
  this project.
```

This error is expected, one can't modify targets imported via FetchContent after all.
lvgl currently provides no ways to customize the libraries linked to it.

Adding a variable for this to https://github.com/lvgl/lvgl/blob/master/env_support/cmake/custom.cmake is probably the way to fix this.
Is this the right place to report this issue?

### How to reproduce?

Create a CMake project following this guide: https://docs.lvgl.io/8.2/get-started/cmake.html

Then, in `lv_conf.h`, `#define LV_USE_SDL 1`

kisvegabor, 2024-10-16:
Let's try to see cleaner in https://github.com/lvgl/lvgl/pull/6973 first, and get back to this after that.

kisvegabor, 2024-11-05:
Let's continue in #7223

ChivenZhang, 2024-12-01:
Does it support SDL3 ?
[image]

glebm, 2024-12-01:
No, lvgl doesn't support SDL3 yet (regardless of this PR).

ChivenZhang, 2024-12-01:
By the way, I would like to wonder whether it is possible to display gpu texture directly ?  For example, show OpenGL fbo in a widget.

kisvegabor, 2024-12-02:
> By the way, I would like to wonder whether it is possible to display gpu texture directly. For example, show OpenGL fbo in a widget. I want to use lvgl in 3D game.

See [here](https://docs.lvgl.io/master/details/integration/driver/opengles.html)