lvgl/lvgl #6962: Creating the same output as in version 5.3

SalahuddinAsh, 2024-09-28:
### LVGL version

5.3

### What happened?

I used to generate C arrays for my fonts using the online font converter version 5.3 using this link https://lvgl.io/tools/font_conv_v5_3

The generated arrays used to have the character as comments beside the code itself as in this screenshot
[image]
 
 The glyph descriptors used to be simple with w_px and glyph_index only
 
[image]

This used to work fine with my code

Currently as the link to font converter 5,3 is not valid anymore, when I try to create a C array with the current font converter I get completely different output with a much bigger glyph descriptor and there is no symbol for the character as comments beside the code itself.

Is there a way that I can get the same output as before?

kisvegabor, 2024-10-03:
Hey,

When we have created our new webpage the v5.3 font converter wasn't added. I thought it's not really used anymore :sweat_smile: 

You can still access it here: https://dev.lvgl.io/tools/font_conv_v5_3

kissa96, 2024-10-21:
@SalahuddinAsh Hi, is this still a valid issue?

kissa96, 2024-10-22:
Cool. Looking forward to it

kisvegabor, 2024-10-23:
it would be really great to add it, however the the old tool is written in PHP and the current site doesn't use PHP :slightly_frowning_face:

kisvegabor, 2024-10-30:
In lack of a good solution I close this issue for now. 🙁

closed 2024-10-30 by kisvegabor