lvgl/lvgl #6964: LVGL 9 Lottie build error on Teensy 4.1 using arm-none-eabi toolchain

egonbeermat, 2024-09-29:
### LVGL version

9.2.0

### What happened?

I'm trying to implement the simple Lottie example [lv_example_lottie_1.c](https://github.com/lvgl/lvgl/blob/2acd6ff7e338f6258a12888a6bfd217db5885592/examples/widgets/lottie/lv_example_lottie_1.c) on a Teensy 4.1, building using platformIO on VSCode and I get this linker error:
```
/Users/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /Users/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc.a(libc_a-openr.o): in function `_open_r':
(.text._open_r+0x14): undefined reference to `_open'
```
If I remove this one line:

`lv_lottie_set_src_data(lottie, lv_example_lottie_approve, lv_example_lottie_approve_size);`

it will build, so even though I'm using the method to load data, not a file, it's pulling that into the linker and I believe this error is caused (from Google search) due to arm-none-eabi toolchain not implementing STDIO. Not really sure what steps I can do at this point, this is a little outside my expertise.

plueschpruem, 2024-10-02:
OK, I thought it was just me…
Trying to build the example for ESP32 smartdisplay / cyd.
lv_lottie_set_src_data() and also lv_lottie_set_buffer() give me "undefined reference to" errors.

egonbeermat, 2024-10-03:
> OK, I thought it was just me… Trying to build the example for ESP32 smartdisplay / cyd. lv_lottie_set_src_data() and also lv_lottie_set_buffer() give me "undefined reference to" errors. But I can CMD-click on them in VSC and it takes me to the function.

The code is there in ThorVG, and when you compile that, it creates references to use the STDIO open, read, etc. functions in the STDIO lib, which doesn't exist in the Teensy baremetal world, hence the linker can't resolve those symbols to the library that actually implements them. Editing the ThorVG source code to remove all file system calls takes away those references allows it to build and link fine, now, although I have to use lv_lottie_set_src_data only, which is fine - I use the LVGL filetohex.py script to create hex from a lottie.json file, and use a bash script to wrap that output of the python script to create a c header file with the correct syntax and structure to create an includable array, so it's only a little more of step to get a Lottie into my codebase.

egonbeermat, 2024-10-03:
After a bit more research, I find I can solve this without commenting out the ThorVG source code, which is good, I don't want to have to keep editing it each LVGL release :) I can actually create stubs myself in my own code and the linker will use those. So, for this undefined reference error:

```
/Users/ianwall/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /Users/ianwall/.platformio/packages/toolchain-gccarmnoneeabi-teensy/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc.a(libc_a-openr.o): in function `_open_r':
(.text._open_r+0x14): undefined reference to `_open'
```
I add to my code:

`extern "C" int _open(){return -1;}`

and it links in. Obviously, file routines won't work, but this seems the cleanest hack.

kisvegabor, 2024-10-14:
You can open an issue in https://github.com/thorvg/thorvg to the have flag to disable file system calls, or replace them with a custom function.

closed 2024-10-25 by kisvegabor