lvgl/lvgl #6965: QR_Code not working under 9.2

Gisbert12843, 2024-09-29:
### LVGL version

v9.2

### What happened?

After migrating from v9.1 to 9.2 battling the increased usage of IRAM (had to put freertos into flash on my esp32), I am now experiencing this wierd behaviour where everything works but the QR Code just is not being shown.

No adjustments were made to the code that perfectly works under 9.1 and the update function even returns LV_RESULT_OK.

The function simply does not produce a visible QR Code in this version.

Of course QR Code features are enabled in menuconfig same as under 9.1.
The only change in settings that had to be done were the removal of the 	"-D LV_COLOR_16_SWAP=1" build Flag (in 9.2 this now leads to distorted colors.

### How to reproduce?

lv_obj_t *qr = lv_qrcode_create(content);
    lv_qrcode_set_size(qr, QR_CODE_SIZE);
    lv_qrcode_set_dark_color(qr, lv_color_hex(COLOR_LIGHTGREY));
    lv_qrcode_set_light_color(qr, lv_color_hex(COLOR_BLACK));
    lv_obj_set_style_border_width(qr, 2, 0);
    
    lv_qrcode_update(qr, "google.de", strlen("google.de"));

    lv_obj_center(qr);

    qr_code_timer_label = lv_label_create(content);
    lv_label_set_text(qr_code_timer_label, (std::to_string(qr_code_seconds_left) + " seconds remaining").c_str());
    lv_obj_set_size(qr_code_timer_label, 250, 20);

kisvegabor, 2024-10-14:
Hi,

Sorry for the delay. Please enable `LV_USE_LOG` as described [here ](https://docs.lvgl.io/master/porting/log.html) to see what can be the problem.

stephencox, 2024-10-19:
I ran into a similar issue. I had to enable LV_DRAW_SW_SUPPORT_ARGB8888 for it to work. Not 100% sure why. In my project I use RGB565 and RGB565A8. It looks like canvas is using LV_COLOR_FORMAT_I1, but I'm not 100% sure how that ends up as  ARGB8888.

kisvegabor, 2024-10-19:
Ahhh, I1 is converted to ARGB888 during rendering.

closed 2024-11-14 by Gisbert12843