lvgl/lvgl #6968: Calling lv_timer_handler too often causes nothing to be rendered

KlemenDEV, 2024-09-30:
### LVGL version

v9.0.x

### What happened?

If I call lv_timer_handler too often (e.g. every tick), nothing is rendered. Nothing gets stuck as lv_timer_handler is still correctly called (verified with breakpoints). The display flush method is called and properly runs, but it seems nothing is rendered since the screen just shows black.

Simply adding a delay of e.g. 5 ticks to the main loop fixes this problem.

What could be causing this / how could I debug this?

kissa96, 2024-10-01:
Hi!

I assume you are using a single core cpu,with RTOS enabled?
Could you give me some general details so i could take a deeper look into it?

Thanks in advance!

KlemenDEV, 2024-10-02:
Yes, we use STM32 with LV_OS_CMSIS_RTOS2 and one draw unit alongside the task calling lv_timer_handler. MCU does not have GPU so we use SW rendering.

This custom task calls lv_timer_handler and uses its return value to determine the delay before the next call (so the task is "nice" to other tasks). This is how I observed this bug since if the return value of lv_timer_handler is e.g. below ~5ms/ticks, nothing is rendered. So I need to limit the minimum value returned by lv_timer_handler to 5 in order for the system to operate.

Setup details:

* LV_OS_CMSIS_RTOS2, 1 SW render unit
* We use lv_tick_set_cb to provide OS tick value
* Driver: lv_lcd_generic_mipi_create, STM32 FMC with DMA
* Double full buffer
* We use lv_display_set_flush_wait_cb to use OS event flags to release custom task once DMA transfer of data to LCD is complete
* All lv_* calls are made within this custom task

kisvegabor, 2024-10-02:
Can it happen that the task with `lv_timer_handler` keeps spinning and the other OS related tasks are blocked by it?

KlemenDEV, 2024-10-02:
All tasks have normal priority and were verified to be running. We also use round-robin scheduling so all tasks get their chance to run.

kissa96, 2024-10-02:
Does the delay work as intended? Have you set up lvgl prior to calling the handler? Are the timers running? Does this task yield after done? These would be my next questions.

KlemenDEV, 2024-10-02:
> Does the delay work as intended?

Yes. It is a well-tested firmware that was previously running custom GFX monochrome toolkit but we are now switching to LVGL and color screen.

> Have you set up lvgl prior to calling the handler?

Yes, lv_init and other related functions are called in this same task before infinite loop is entered where sleep and timer handler calls happen

> Are the timers running?

In normal operation yes. When lv_timer_handler is called often? I will verify and come back.

> Does this task yield after done?

osDelay causes the task to yield. Manual yielding without delay results in nothing being rendered too.

kissa96, 2024-10-07:
Does the timer handler get even called?
So if I get it right, you are using RTOS and in the infinite loop you are handling the LVGL timer?

KlemenDEV, 2024-10-07:
Yes. The timer is called. Also animation timers are being called correctly.

However, nothing is rendered if lv_timer_handler is called too frequently.

kisvegabor, 2024-10-23:
Sorry for the delay. If this bug is still present I suggest simplifying the driver and adding bak features one-by-one to see where the problem is. The simplest case should be 
- no DMA
- no double buffer
- no OS related things in the driver
- no wait_cb just `lv_display_flush_ready`

KlemenDEV, 2024-10-23:
Yes, the issue is still present, but mitigated with not calling timer handler too often.

I will experiment with going more and more basic to see where this issue stops.

The only concern I have with that is that if I remove DMA and OS related things, flushing will be so slow delay will be bigger anyway and timer won't even be possible to be called more often

kisvegabor, 2024-10-23:
Anyway, hopefully it will give some hints. :crossed_fingers:

kissa96, 2024-10-25:
Apologies for not responding earlier. Have you verified all execution? Do you have semaphores inside the flush callback which are not set properly in ISRs or elsewhere?

If you could support us with some code or flowchart regarding the architecture, we might be able to help

KlemenDEV, 2024-10-25:
I will try to provide more details when I get some time to work on this project again (hopefully soon)

KlemenDEV, 2024-11-14:
> or that there are enough timers running that it doesn't have a chance to return with time to spare (or drawing time is long enough)

KlemenDEV, 2024-11-14:
> or that there are enough timers running that it doesn't have a chance to return with time to spare (or drawing time is long enough)

But in this case, calling it more often should make things better, not worse (in other words, not working).

We have got it working by limiting rate of calls, so unfortunately right now it is not the highest priority I have to investigate this further. But if I get more findings, I will keep you updated.

> unless different threads are involved

It is OS environment, but all graphics stuff is in single thread

> or that there are enough timers running

This happens with basic UI without any custom timers too

closed 2024-12-14 by lvgl-bot