lvgl/lvgl #6972: Text not updating properly

bojankiprudhvi, 2024-10-01:
### LVGL version

v9.2.0

### What happened?

I just create a label and created a timer and i want to update the numbers starting from 1 for that label.
the numbers were updating but some numbers were not updated properly
example: the  sequence i can observe is 
1,2,3,4,2,6,7,8,6,10...20,12,22...
and the sequence changes when i use different font.

### How to reproduce?

int count =1;

void update(lv_timer_t *timer) {
    printf("inisde update");
    char buf[32];
    sprintf(buf, "num: %d\n", count);
    lv_label_set_text(ui_TextArea4, buf);
    count++;
}

void update_time_alternative() {
    lv_obj_t * screen = lv_scr_act();
    ui_Screen1 = lv_obj_create(screen);
    lv_obj_set_size(ui_Screen1, lv_disp_get_hor_res(NULL), lv_disp_get_ver_res(NULL));
    ui_TextArea4 = lv_label_create(ui_Screen1);
    lv_obj_set_x(ui_TextArea4, -101);
    lv_obj_set_y(ui_TextArea4, -311);
    lv_obj_set_align(ui_TextArea4, LV_ALIGN_CENTER);
    lv_label_set_text(ui_TextArea4, "2");
    lv_obj_set_style_text_color(ui_TextArea4, lv_color_hex(0x000000), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_text_opa(ui_TextArea4, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_text_align(ui_TextArea4, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_text_font(ui_TextArea4, &lv_font_montserrat_48, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_color(ui_TextArea4, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_TextArea4, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_grad_color(ui_TextArea4, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_timer_t * timer = lv_timer_create(update, 1000, NULL);
}

bojankiprudhvi, 2024-10-01:
if i don't specify any font style then number '2' and '3' were treated same and were not updated on the screen properly.

kisvegabor, 2024-10-02:
It seems like a driver issue. Are using a display controller? If so some models have constraints on the x1 and x2 coordinate of the update window. 

Do you see the issue in a simulator too?

bojankiprudhvi, 2024-10-03:
I faced issue on Ubuntu using SDL
/* Draw using cached SDL textures*/
#define LV_USE_DRAW_SDL 0
By changing this the issue is resolved.

closed 2024-10-03 by bojankiprudhvi

kisvegabor, 2024-10-03:
Ah, it was a tricky bug in the SDL renderer. I've just fixed it here: https://github.com/lvgl/lvgl/pull/6994

commit 78080702 (2024-10-09): fix(sdl): add static_text flag to to lv_draw_label_dsc_t
... and cache only static texts in the SDL renderer. Otherwise the cached draw_dsc can contain wild pointers if teh texts are reallocated/freed.

Fixes #6972