lvgl/lvgl #6983: Bug: Compiler error when including lv_lcd_generic_mipi.h

mverch67, 2024-10-01:
### LVGL version

9.2.0

### What happened?

The compiler guards at the end of the file do not match the order used at the beginning which results in unterminated extern "C" declaration:
https://github.com/lvgl/lvgl/blob/0ff875423a9b7dad164324b658b044a566128489/src/drivers/display/lcd/lv_lcd_generic_mipi.h#L235

Start:
```
#ifndef LV_LCD_GENERIC_MIPI_H
#define LV_LCD_GENERIC_MIPI_H

#ifdef __cplusplus
extern "C" {
#endif

#if LV_USE_GENERIC_MIPI
```

End:
```
#ifdef __cplusplus
} /*extern "C"*/
#endif

#endif /*LV_USE_GENERIC_MIPI*/

#endif /* LV_LCD_GENERIC_MIPI_H */
```

It should be like this:
[image]

### How to reproduce?

Define one of these LV_USE_ST77xx as 1 in lv_conf.h

kissa96, 2024-10-14:
Nice catch! Fixed in #7062

kisvegabor, 2024-10-14:
FYI, it was fixed a few days ago in master: https://github.com/lvgl/lvgl/pull/6905

kisvegabor, 2024-10-29:
As https://github.com/lvgl/lvgl/pull/6905 is merged I close it.

closed 2024-10-29 by kisvegabor