lvgl/lvgl #6988: lv_bin_decoder decode_alpha only bit mask inversion on IMAGEA1

RamDav, 2024-10-02:
### LVGL version

v9.2.0

### What happened?

I'm trying to add Images to my application the goal is to have the same images recolered with different colors to indicate states.
The idea is to use a format which allows to recolor the Image. So i came upon the AX formats i tried this first with the A8 format, wrote a program to ouput files into this format and it worked great. Next step was to use the A1 format as this was all i needed.
After some trial and error with the bit stuffing, rows of the image have to be stuffed to complete bytes. My Images have still been wrong. An observation i made was that it looked like every each column of the Image appeared in reverse order.

As far as i understand this the pixels are defined in the same order as the pixels in the Image. 
So in an Image with 8 by 8 pixels, a row has 1 byte and the image consists of 2 bytes.
The first bit of the image data has to be the first pixel of the Image the second bit is the second pixel and so on.
The first Pixel of the second row has to be in byte 2 bit 0.

The problem is see is that the function 
```c
static lv_result_t decode_alpha_only(lv_image_decoder_t * decoder, lv_image_decoder_dsc_t * dsc)
```

in the 

> lv_bin_decoder.c 

File expands the pixels bit array starting from the last byte position to the last byte in the decoded A8 buffer, to allow using one buffer for decoding.

The problem is that by expanding the last A8 1 pixel first and starting at the last byte of the source A1 buffer at bitpos 0
```c
    if(dsc->header.cf != LV_COLOR_FORMAT_A8) {
        /*Convert A1/2/4 to A8 from last pixel to first pixel*/
        uint8_t * in = img_data + file_len - 1;
        uint8_t * out = img_data + buf_len - 1;
        uint8_t mask = (1 << bpp) - 1;
        uint8_t shift = 0;
        for(uint32_t i = 0; i < buf_len; i++) {
            /**
             * Rounding error:
             * Take bpp = 4 as example, alpha value of 0x0 to 0x0F should be
             * mapped to 0x00 to 0xFF. Using below equation will give us 0x00 to 0xF0
             * thus causes error. We can simply interpolate the value to fix it.
             *
             * Equation: *out = ((*in >> shift) & mask) << (8 - bpp);
             * Ideal: *out = ((*in >> shift) & mask) * 255 / ((1L << bpp) - 1)
             */
            uint8_t value = ((*in >> shift) & mask);
            *out = bit_extend(value, bpp);
            shift += bpp;
            if(shift >= 8) {
                shift = 0;
                in--;
            }
            out--;
        }
```
leads to expanding each 8 pixel block in reverse order. E.g. the first bit in the last byte in a 8 by 8 pixel image represents the pixel
x = 0; y = 7 but is expanded to pixel position x = 7; y = 7 in the A8 buffer.
I changed my A1 pixel format generation tool to fill the bits in reverse order and this produced a correct image.

As the pixels are expanded in reverse order so should the bits of the input be processed in reverse order.

byt changing the shifting operation
from 
```c
uint8_t value = ((*in >> shift) & mask);
```
to 
```c
uint8_t value = ((*in >> (7 - shift)) & mask);
```
```c
    if(dsc->header.cf != LV_COLOR_FORMAT_A8) {
        /*Convert A1/2/4 to A8 from last pixel to first pixel*/
        uint8_t * in = img_data + file_len - 1;
        uint8_t * out = img_data + buf_len - 1;
        uint8_t mask = (1 << bpp) - 1;
        uint8_t shift = 0;
        for(uint32_t i = 0; i < buf_len; i++) {
            /**
             * Rounding error:
             * Take bpp = 4 as example, alpha value of 0x0 to 0x0F should be
             * mapped to 0x00 to 0xFF. Using below equation will give us 0x00 to 0xF0
             * thus causes error. We can simply interpolate the value to fix it.
             *
             * Equation: *out = ((*in >> shift) & mask) << (8 - bpp);
             * Ideal: *out = ((*in >> shift) & mask) * 255 / ((1L << bpp) - 1)
             */
            uint8_t value = ((*in >> (7 - shift)) & mask);
            *out = bit_extend(value, bpp);
            shift += bpp;
            if(shift >= 8) {
                shift = 0;
                in--;
            }
            out--;
        }
    }
```

The decoder should behave in the "assumed" correct way.

### How to reproduce?

Test image rotary symmetrical 4 pixel long line on the border.
Test image:
[image]
Scaled up for viewing pleasure
[image]

"Correct" bit order
```c
...

const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_LARGE_CONST uint8_t test_image_map[] = {
  0x8f,
  0x80,
  0x80,
  0x80,
  0x01,
  0x01,
  0x01,
  0xf1,
  };

const lv_image_dsc_t test_image = {
  .header.cf = LV_COLOR_FORMAT_A1,
  .header.magic = LV_IMAGE_HEADER_MAGIC,
  .header.w = 8,
  .header.h = 8,
  .data_size = 8,
  .data = test_image_map,
};

...
```

reversed bit order
```c
...
const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_LARGE_CONST uint8_t test_image_map_rev[] = {
  0xf1,
  0x01,
  0x01,
  0x01,
  0x80,
  0x80,
  0x80,
  0x8f,
  };

const lv_image_dsc_t test_image_rev = {
  .header.cf = LV_COLOR_FORMAT_A1,
  .header.magic = LV_IMAGE_HEADER_MAGIC,
  .header.w = 8,
  .header.h = 8,
  .data_size = 8,
  .data = test_image_map_rev,
};
...
```

Load and add these to a screen

```c
LV_DECLARE_IMAGE(test_image);
LV_DECLARE_IMAGE(test_image_rev);
...
lv_obj_t* img = lv_image_create(lv_screen_active());
lv_image_set_src(img, &test_image);
lv_obj_set_style_image_recolor(img, lv_color_black(), 0);

lv_obj_t* img = lv_image_create(lv_screen_active());
lv_image_set_src(img, &test_image_rev);
lv_obj_set_style_image_recolor(img, lv_color_black(), 0);
```

RamDav, 2024-10-02:
Hello,
i wrote my own tool at one part for practice and to have something i can integrate into my build process.
A image created with the LVGLImage.py script outputs correctly to  the screen in the A1 format.
So is this A1 format a format specific to lvgl or is it an implementation of some standart file format?
As this seems kind of counterintuitive to order the pixel bits in this way.

The python scipts also seems to pack the bits into the byte in reverse order.

```python
    def _png_to_alpha_only(self, cf: ColorFormat, filename: str):
        reader = png.Reader(str(filename))
        w, h, rows, info = reader.asRGBA8()
        if not info['alpha']:
            raise FormatError(f"{filename} has no alpha channel")

        rawdata = bytearray()
        if cf == ColorFormat.A8:
            for row in rows:
                A = row[3::4]
                for e in A:
                    rawdata += uint8_t(e)
        else:
            shift = 8 - cf.bpp
            mask = 2**cf.bpp - 1
            rows = [[(a >> shift) & mask for a in row[3::4]] for row in rows]
            for row in png.pack_rows(rows, cf.bpp):
                rawdata += row

        self.set_data(cf, w, h, rawdata)
```

XuNeo, 2024-10-02:
There's no standard. 
LVGL previously uses this format, thus I wrote LVGLImage.py following the same way.
Your may tweak your tool to follow it too.

RamDav, 2024-10-02:
Ok thanks a lot.

closed 2024-10-02 by RamDav