lvgl/lvgl #6996: Documentation / Help request: Simulating an OLED

mado89, 2024-10-03:
Hi,
I am on the lookout for input on how to simulate a monochrome OLED display (128x64px). Can this be done with SDL while the simulated code uses lv_ functions?
Thanks

kisvegabor, 2024-10-03:
I moved this issue to the main LVGL repo as the SDL driver is locate here. 

It's really not supported now, but it's possible to add it to the SDL driver [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c). If the display's color format is `I1` we should use `ARGB8888` for SDL [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c#L384-L393) and convert the indexed data to ARGB8888 in the `flush_cb`

kisvegabor, 2024-10-23:
PR opened, see https://github.com/lvgl/lvgl/pull/7036

kisvegabor, 2024-11-13:
https://github.com/lvgl/lvgl/pull/7036 is ready and being reviewed.

mado89, 2024-11-13:
Thanks for your efforts. I will check when I have time again (currently working on a different project, sorry)

closed 2024-12-03 by liamHowatt