lvgl/lvgl #6997: Support for 8bpp indexed color mode

GW8RDI, 2024-10-03:
I saw a post in March, 2024 saying that you'd be adding support for indexed color, 8bpp with CLUT.
Is there a status update available? This is quite important because a lot of MCUs don't have the memory for higher BPPs.

kisvegabor, 2024-10-03:
I've transferred this issue to the main lvgl repo.

> I'm not sure, I don't know your code enough to say where it would best be done. 

No worries.

>  Isn't I8 grey-scale?

Almost, but no. grey-scale can be blended with opacity and it's easy to have anti-aliasing there. With I8 we should just ignore all the opacity. (< 50% opa should be transparent, the others are fully covering.) 

---

Let's see if someone picks up this issue.

GW8RDI, 2024-10-03:
Yes, so indexed color 8-bit, C8, fonts can antialiased but it will need to allow people to set the number of colors to use, say with a minimum of 4 color aliasing would be ideal, up to say 32.
Opacity and color-keying could be the same, disabled, 0 colors assigned, or 1-32 perhaps.
There are quite a lot of MCUs which would really benefit from this mode, with memory limitations/costs its ideal for IOT and embedded systems.

kisvegabor, 2024-10-14:
For the sake of simplicity I would rather ignore anti-aliasing for now.

closed 2024-11-09 by lvgl-bot

BitLoose, 2025-01-03:
So has 8-bit color mode been added to LVGL?

kisvegabor, 2025-01-03:
It's not implemented yet. The feature is not on our roadmap now, but if someone is interested in working on it, I/we'd be happy to review it.