lvgl/lvgl #7000: Released event for objects

fatiihyildizz, 2024-10-04:
### Introduce the problem

Hello, 
I want to use released event but there is a problem. Events such as pressed and pressing are only generated when the areas of the relevant objects are touched. However, when I move my finger to different areas while touching the object (exiting the object area), the released event is not generated. It is generated whenever I move my finger away. However, I want the released event to be generated when I move out of the relevant object area. Otherwise, a bad impression is created. How can I do this?

egonbeermat, 2024-10-05:
The released event should only be generated on a release. Events are very specific like that, and would cause issues if generated when not actually happening. Imagine trying to scroll a roller by moving your finger a long distance away from the roller to keep it scrolling, but it generated a release as soon as you left the bounds of the object? It'd be very hard to use.

You could do this, on the object you want this behavior:

`lv_obj_remove_flag(object, LV_OBJ_FLAG_PRESS_LOCK);`

[LV_OBJ_FLAG_PRESS_LOCK](https://docs.lvgl.io/master/widgets/obj.html): Keep the object pressed even if the press slid from the object

This would then generate `LV_EVENT_PRESS_LOST` once you moved outside the bounds, and you could handle that and 'pretend' it was released. From the docs:

[LV_EVENT_PRESS_LOST](https://docs.lvgl.io/master/overview/event.html): The object is still being pressed but slid cursor/finger off of the object

fatiihyildizz, 2024-10-07:
Hi,
Thanks for your answer. I have tried this method and it works. I was wondering if there was a more stable method.

egonbeermat, 2024-10-08:
I only suggested that after reading the docs and trying it in the sim, so I don't have any further answers :) Hopefully, others may share thoughts!

closed 2024-11-08 by lvgl-bot