lvgl/lvgl #7003: KEYPAD indev LV_EVENT_DOUBLE_CLICKED

jgauchia, 2024-10-06:
### Introduce the problem

Hi,
It's possible to have double click event, LV_EVENT_DOUBLE_CLICKED, on KEYPAD indev?

kissa96, 2024-10-14:
The keypad has a special mindset behind it, as you wouldn't normally double or triple press/hold, etc. it while entering the code. Only LV_EVENT_SHORT_CLICKED makes sense from a use case standpoint.
We could implement all events for the keypad as well, but as a normal use case does not require it, we decided not to.
If we receive enough feedback on this feature, then we might implement it in a future release.

jgauchia, 2024-10-14:
Ok @kissa96 thank you!

kisvegabor, 2024-10-16:
I think in the near future we should add it to our todo list as more an more people are asking for these features. 🙂 
We already have a lot of tings for v9.3, but it can fit in v9.4.

jgauchia, 2024-10-16:
No problem @kisvegabor ,It is not a priority.

All this was to be able to add, in the case of an ESP32 the GPIO0, by double clicking it enters sleep mode, perhaps it is not the most ideal option to use KEYPAD for this,

Currently I have it implemented with, single press sleep mode, long press shutdown.

Thank you so much!

kisvegabor, 2024-11-04:
I'm closing this for now.

closed 2024-11-04 by kisvegabor