lvgl/lvgl #7004: Odd Rendering behavior

kdschlosser, 2024-10-07:
### LVGL version

master

### What happened?

I am using an I8080 display, double buffering and the buffers are set to 1/10th the size of the display. the display is 800 x 480 x 16.  I am using DMA memory for the buffers. 

I have a JPG image that is 800 x 480 in size and I am using the LVGL file system to load the JPG file and I am passing the filename to an `lv_image` widget. The JPG does load properly so that aspect all works as it should. What is not working as it should is the rendering of the JPG.

I have timed how fast the flush function takes to run which is 0 milliseconds. This is what it should be because of using DMA memory. The time it takes for the actual buffer data to get flushed to the display is ~2 milliseconds. Which is a correct amount of time for the buffer size and 8 data lanes with a clock of 40mhz. All of that checks out. LVGL on the other hand is taking 40 milliseconds to render to only a single frame buffer The flush ready function is getting called when it should be so that's not causing an issue.

LVGL in general even when not dealing with images seems to have a horribly long render time. It should not take 20 or 30 milliseconds to render data to the buffer on 240mhz processor when it only takes 2 milliseconds to transfer the same amount of data at 40mhz. I am not sure what it is that is causing that kind of a latency or if there is even a way of figuring out what is causing it.

kdschlosser, 2024-10-07:
I did want to also note that I tested this on my PC as well and it took 15 milliseconds for LVGL to render the same image. That doesn't seem right running on a computer that has 32 logical processors running at 4.2 GHZ. There is definitely something amiss with the rendering.

zzzlll111, 2024-10-09:
I ran a full screen buffer of 1024x800x32 on a 1.2Gmz CPU and calculated that it would be around 40ms if the global rendering had jpg. If it is a local refresh, the minimum rendering time is about 10ms, such as the refresh of a label. I also encountered a problem that if it is a local refresh, the time of calling flush_cb once may be delayed by several renderings. If there are other callbacks later, such as touch callbacks, the time of calling flush_CB will be delayed for a long time. The interval between touch callbacks may be too long, causing the touch to stall.

kisvegabor, 2024-10-16:
Well, there are multiple things here (assuming that your using TJPGD):
1. You are reading the image from file, so the file system's speed is also slows things down. Try opening the image from C array as comparison. ([Converter](https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py))
2. You can also enable File System caching as an experiment. E.g. `LV_FS_STDIO_CACHE_SIZE 512`
2. TJPGD decodes the images in tiles to save RAM. With 8x8 tiles it's 6000 tiles, meaning LVGL needs to blend 6000 small 8x8 images, which head a small overhead 6000 times. If you have a lot of RAM consider rendering the image into a canvas first and use it as an image. 

cc @bitbank2

kdschlosser, 2024-10-16:
I have loaded the JPG and decoded it using LVGL, the entire thing to a canvas and then taken the buffer from a canvas and kept it held in memory and then timed how long it would take passing the buffer to an image and it is only slightly faster. No decoding needs to be done at this point because it was done beforehand and the decoded RGB is now what is being used instead. There is not a huge improvement in the rendering time. I don't think the issue is in the decoder. I think it is else where. I have run into the same issue rendering 2 large sliders that take up about 1/2 of the display and it is taking 20 milliseconds to render them. So that actually fits because the sliders take up 1/2 the display where as the JPG takes up the whole display and it takes 40 milliseconds. So there is something that is going on that effects everything and not just an image decoder. It doesn't matter if I use a direct buffer or a partial an image or sliders there is something that is causing a pretty consistent slow down.

kisvegabor, 2024-10-17:
What kind of MCU do you use?

kdschlosser, 2024-10-17:
I have used an ESP32 but also running on Linux it is taking >2ms to render which is a long time when running on a 4.2Ghz AMD ThreadRipper Pro. 

iterating over a buffer of the same size and setting byte data in Python takes 3 milliseconds and Python code is 200 times slower than C code is. That is on the same machine that I ran the Linux port on.

kissa96, 2024-10-18:
A ThreadRipper is not better for single core jobs than any other CPU is. Most tasks use a single core anyways.
40ms is 25fps, which considering a JPG and using a file system + OS.
As Gábor mentioned, small tiles can be an issue as well. Moreover if you use DMA for these, then it might as well be just a lot slower than moving the data with the CPU, as DMAs usually run on slower clocks and requires setup for each command.
Try it without DMA and see if its any better

kisvegabor, 2024-10-28:
> LVGL in general even when not dealing with images seems to have a horribly long render time. It should not take 20 or 30 milliseconds to render data to the buffer on 240mhz processor when it only [...]

We have compared the performance of v8 and  v9 on many devices and  in general there was no significant difference in our benchmarks. However there is one thing that we have noticed: with devices running the code from external flash v9 could be really slower, especially when using images from flash too. 

Please run `lv_demo_benchmark` to better see which the slow parts are.

closed 2024-11-24 by lvgl-bot