lvgl/lvgl #7005: Rotate the bitmap by a given angle in loop without using animation

brightproject, 2024-10-07:
### Introduce the problem

> name=lvgl
> version=8.3.11

The rotation is performed once.

### Proposal

Hello @kisvegabor 
Thank you for your work!

I use the `SLS` program to convert images to `C code`.
My device displays graphics perfectly.

[image]

I also set it up so that when you touch an object in the center of the display, the scale starts to rotate.
For this, I use an event and a function:

> void rotatescale_Animation(lv_obj_t * TargetObject, int delay)
> {
>     ui_anim_user_data_t * PropertyAnimation_0_user_data = lv_mem_alloc(sizeof(ui_anim_user_data_t));
>     PropertyAnimation_0_user_data->target = TargetObject;
>     PropertyAnimation_0_user_data->val = -1;
>     lv_anim_t PropertyAnimation_0;
>     lv_anim_init(&PropertyAnimation_0);
>     lv_anim_set_time(&PropertyAnimation_0, 5000);
>     lv_anim_set_user_data(&PropertyAnimation_0, PropertyAnimation_0_user_data);
>     lv_anim_set_custom_exec_cb(&PropertyAnimation_0, _ui_anim_callback_set_image_angle);
>     // lv_anim_set_values(&PropertyAnimation_0, 0, 1800); // from N to S
>     lv_anim_set_values(&PropertyAnimation_0, 0, 3600); // from N to N
>     lv_anim_set_path_cb(&PropertyAnimation_0, lv_anim_path_ease_in_out);
>     lv_anim_set_delay(&PropertyAnimation_0, delay + 0);
>     lv_anim_set_deleted_cb(&PropertyAnimation_0, _ui_anim_callback_free_user_data);
>     lv_anim_set_playback_time(&PropertyAnimation_0, 0);
>     lv_anim_set_playback_delay(&PropertyAnimation_0, 50000);
>     lv_anim_set_repeat_count(&PropertyAnimation_0, 1000);
>     lv_anim_set_repeat_delay(&PropertyAnimation_0, 10000);
>     lv_anim_set_early_apply(&PropertyAnimation_0, false);
>     lv_anim_start(&PropertyAnimation_0);
> 
> }

It works fine, but this is an animation:

https://github.com/user-attachments/assets/22763e49-1ab3-43a3-9d88-a58a03709d30

To rotate the image by a certain angle, I wrote the following function:

> void rotate_scale(lv_obj_t * TargetObject, int angleRotate)
> {
>     ui_anim_user_data_t * PropertyAnimation_0_user_data = lv_mem_alloc(sizeof(ui_anim_user_data_t));
>     PropertyAnimation_0_user_data->target = TargetObject;
>     PropertyAnimation_0_user_data->val = -1;
>     lv_anim_t PropertyAnimation_0;
>     lv_anim_init(&PropertyAnimation_0);
>     lv_anim_set_time(&PropertyAnimation_0, 5000);
>     lv_anim_set_user_data(&PropertyAnimation_0, PropertyAnimation_0_user_data);
>     lv_img_set_angle(TargetObject, angleRotate);
> 
> }

Perhaps there is a lot of unnecessary stuff in the function, it seems to me that the rotation is performed even if there is:

> void rotate_scale(lv_obj_t * TargetObject, int angleRotate)
> {
>  lv_img_set_angle(TargetObject, angleRotate);
> }

>     ui_roundscale = lv_img_create(ui_Screen);
>     // ui_roundscale = lv_img_create(lv_scr_act());
>     lv_img_set_src(ui_roundscale, &ui_img_compas360_png);
>     lv_obj_set_width(ui_roundscale, LV_SIZE_CONTENT);   /// 360
>     lv_obj_set_height(ui_roundscale, LV_SIZE_CONTENT);    /// 360
>     lv_obj_set_align(ui_roundscale, LV_ALIGN_CENTER);
>     lv_obj_add_flag(ui_roundscale, LV_OBJ_FLAG_ADV_HITTEST);     /// Flags
>     lv_obj_clear_flag(ui_roundscale, LV_OBJ_FLAG_SCROLLABLE);      /// Flags

For some reason using `lv_scr_act()` didn't work for me in version `8.3.11`

In the main loop of the `main.c` file I wrote this:

> void loop()
> {
>     lv_timer_handler(); /* Let the GUI do its work */
>     delay(5);
> 
>     int ang = 0;
> 
> // rotate_scale(ui_roundscale, ang);
>     lv_img_set_angle(ui_roundscale, ang);
> 
>     ang += 100;
> 
>     // delay(30);
> 
> }

The rotation is performed once, angle simply rotates from 0 degrees:

[image]

by `minus 30` degrees:

[image]

But in the end the image just rotates to the desired angle and does not change.
I don't want to change the image by commands from the outside.
I haven't managed to do this except through animation yet.
What is the way to rotate the image so that I can give angle values, for example from an encoder, and the angle changes in a `loop` cycle and the image rotates?
P.S. It would be great if you opened a discussion tab so as not to post such questions in the source.

lorddevereux, 2024-10-13:
The forum is probably a better place for this question/discussions.

[](https://forum.lvgl.io/)

brightproject, 2024-10-13:
> The forum is probably a better place for this question/discussions. [forum.lvgl.io/](https://forum.lvgl.io/)

Hello @lorddevereux 🙂
Thank you for your reply.
Unfortunately, I ended up in spam on the forum
I've been on GitHub for a long time and many communities have opened a discussion tab and it replaces the forum.
I was wondering how to increase the speed of my widget display?

https://github.com/user-attachments/assets/9be90e06-9a9e-4840-a4df-2350a664020a

By optimizing images or by using flags in the code?

kisvegabor, 2024-10-16:
> Unfortunately, I ended up in spam on the forum

I reviewed the flagged posts. You should be all good now. 🙂 

> What is the way to rotate the image so that I can give angle values, for example from an encoder, and the angle changes in a loop cycle and the image rotates?

I think you need this: https://docs.lvgl.io/master/examples.html#rotate-and-zoom

>  P.S. It would be great if you opened a discussion tab so as not to post such questions as issue.

I'm afraid that it would make the community fragmented. :slightly_frowning_face:

closed 2024-11-13 by lvgl-bot