lvgl/lvgl #7006: LVGL 9 lv_chart_add_series behavior opposite to LVGL 8 behavior

egonbeermat, 2024-10-08:
### LVGL version

9.2.0

### What happened?

In the process of migrating from LVGL 8.x.x to 9.x.x, I was testing one of my line chart components with two series with similar data ranges, and seeing unexpected results in the output when switching from LVGL 8.x.x to LVGL 9.x.x. I'll spare my debugging sob story, but ultimately the issue here is I was referencing the first added series by `lv_chart_get_series_next(chart, NULL)` and it appears LVGL now inserts series in the reverse order in LVGL 9.x.x, compared to LVGL 8.x.x. so when my code expected to be retrieving the series added first, it was in fact retrieving the series added second.

```
//LVGL 8 lv_chart_add_series:
lv_chart_series_t * ser = _lv_ll_ins_head(&chart->series_ll);

//LVGL 9 lv_chart_add_series:
lv_chart_series_t * ser = lv_ll_ins_tail(&chart->series_ll);
```

Once I switched over the references to the series, everything is fine. It's heads or tails (pun intended) if this is a bug or a documentation issue, but it's not desirable behavior to leave unspoken!

kisvegabor, 2024-10-11:
Thank you for opening this issue. If we would change it now in v9, it'd be a breaking change. :slightly_frowning_face: 
As none of the orders is better I'd leave it as it as, just document it correctly. 

However, there is something what bothers me. In case of line charts the first series is on the top. It's the opposite of how widgets are normally layered. (older widget is on the bottom). Maybe we just need to change draw order? I'm nor sure. 

What do you think?

egonbeermat, 2024-10-12:
Toss a coin, let it decide heads or tails for insert :) 

It's currently a breaking change from 8, but to change it would be a breaking change to 9, too, as you said, so it's really a question of picking the lower of two numbers: how many people have line charts with more than series, and reference those series by `lv_chart_get_series_next`, that will be migrating from v8 to v9, vs the ones already implemented in v9. Whichever number you feel is the lower, do that....

If you just reverse the drawing order in v9, then the behavior will be corrected, and maybe resolve the issue, but it may eat away at you that you have implemented a fix that way, and for those cases where 'code is documentation' it may cause a little head scratching, wondering why this widget is drawing in reverse and inserting at tail? Probably not a big deal, and may be your best path to fix this way.

kisvegabor, 2024-10-16:
Well, we can swap the insertions order which will
- fix `lv_chart_get_series_next` 
- fix Line chart drawing order
- but Column and Scatter chart drawing order will be swapped. 

I'm ok with that, if we swap the drawing order in Column and Scatter charts.

We can document it as breaking change which probably won't affect too many people. 

Would you be open to send a PR?

closed 2024-11-13 by lvgl-bot