lvgl/lvgl #7007: break! img_draw_core

xinsuinizhuan, 2024-10-08:
### LVGL version

v9.3.0

### What happened?

run lv_demo_widgets()，mouse roller in Analyses page:
the buf_stride is zero:
[image]

xinsuinizhuan, 2024-10-08:
still break:
[image]

kisvegabor, 2024-10-11:
Please enable LVGL's logging in `lv_conf.h`. Do you see an out of memory error?

xinsuinizhuan, 2024-10-17:
to make Division, juedge buffer_stride whether is zero is normal， it is hard to add?

xinsuinizhuan, 2024-11-01:
have add the judget?

liamHowatt, 2024-11-05:
> still break:

The fix you attempted in that screenshot leaves `tmp_buf` uninitialized if stride is zero.

The question is how stride is becoming zero? Is `blend_w` zero or is `lv_color_format_get_size(cf_final)` returning zero?

closed 2024-12-04 by lvgl-bot